merbanan/rtl_433 #977: Any possibility to add R900 and SCM meters from RTLAMR project?

TechnoSwiss, 2019-02-06:
I noticed there are some additions to the project in the 900MHz ISM band, is there any chance some of the SCM and R900 decoders from the bemasher/rtlamr project could get added in?

merbanan, 2019-02-06:
The answer to this can mostly be answer by the question if you can supply sample signals and decoding descriptions? If yes then yes, if no then most likely no.

TechnoSwiss, 2019-02-06:
I was trying to get sample signals earlier today, but even though I'm able to capture and decode packets with rtlamr I wasn't getting anything with rtl_433. Looking over the R900 decoder in the rtlamr project though it looks like maybe the issue was I had the frequency wrong. 

I think all the decoding descriptions are already in the github project for rtlamr.

merbanan, 2019-02-06:
The decoding descriptions must be in there some where but every bit of research is time I or some one else can spend on implementation rather on digging through code.

TechnoSwiss, 2019-02-06:
I'll see if I can dig those out then and check if the sample issue was just getting the freq. wrong. Thanks.

merbanan, 2019-02-06:
https://en.wikipedia.org/wiki/Encoder_receiver_transmitter gave the basics beside the bit rate.

dadler, 2019-09-29:
The rtlamr wiki has this page on their [Signal Processing](https://github.com/bemasher/rtlamr/wiki/Signal-Processing).  

Having rtl_433 support this would be great since then I could (presumably) have rtl_433 read alternately from 433 Mhz temperature sensors and a 900 Mhz water meter using the same dongle, without hacks like periodically stopping rtl_433 to run rtl_tcp and rtlamr.  Thanks.

merbanan, 2019-10-02:
@dadler, have you been able to record some samples? I suggest you use rtlcap (github.com/bemasher/rtlcap) to create proper signal samples. Then we can look into what is needed to maybe add support for rtlamr supported protocols.

TechnoSwiss, 2019-10-02:
I think I have actually gotten this working on my system, I need to verify, then figure out how to create a merge request I guess.

merbanan, 2019-10-31:
@TechnoSwiss have you been able to get some signal samples?

dadler, 2019-11-01:
Sorry, my water meter seems to have stopped transmitting so I can’t capture
data now.

TechnoSwiss, 2019-11-01:
@merbanan Yes I was able to get samples I have them checked into my fork of the test repo https://github.com/TechnoSwiss/rtl_433_tests/tree/master/tests/scm/01 I have partial working decoder for this meter I believe, but then my time got preempted by other tasks before I finished going through the documentation on how to get this decoder finished and checked back into the main project line.

decoder {
    name=SCM,
    modulation=OOK_MC_ZEROBIT,
    short=30,
    long=0,
    reset=64,
    bits=96,
    get=@32:{24}:consumption,
    get=@56:{24}:ID
}

dabeckham, 2019-12-03:
@merbanan Did you delete rtlcap?  

I too am trying to use rtl_433 for an R900 water meter. I would like to hop between frequencies to monitor everything with only one SDR dongle. Thanks for all your work making this available for everyone!!

merbanan, 2020-01-07:
@TechnoSwiss do you have the code for the decoder somewhere?

TechnoSwiss, 2020-01-07:
@merbanan I believe I'm just using one of the built-in decoders? I'll have to check when I get home. I believe I had rtl_433 spitting out decoded information, so I'll look and see how I got there.

merbanan, 2020-01-07:
@TechnoSwiss ok I see, you used a config file.

merbanan, 2020-02-10:
Implemented SCM support, send samples for other protocols.

closed 2020-02-10 by merbanan