merbanan/rtl_433 #978: Remote device number specified is invalid - but RTL-SDR works under all other apps

Abraxas3d, 2019-02-06:
I have an RTL-SDR on an Ubuntu machine (16.04 LTS)

rtl_433 installed without incident.

The rtl-sdr works under MATLAB, GNU Radio, and gqrx. It shows up under lsusb. 

Running rtl_433, whether plain, with various config files, with device ID set manually, with any of the examples - in every case the result is "Remote device number specified is invalid".

Verbose mode doesn't add anything to this particular error message. 

What am I doing wrong?

merbanan, 2019-02-06:
Can you provide command line output without any config file active or any command line option?

zuckschwerdt, 2019-02-06:
Yeah, that first line should give a hint. It should be
```
rtl_433 version 18.12-86-g931c9ce branch master at 201902061752 inputs file rtl_tcp RTL-SDR SoapySDR
```
But it sounds like you aare missing the `RTL-SDR` input. Install librtlsdr and rebuild.

Abraxas3d, 2019-02-07:
Thank you. 

Here's the command line output without any config file active or any command line option:

abraxas3d@marble:~$ rtl_433
rtl_433 version 18.12-84-g3e5157d branch master at 201902061559 inputs file rtl_tcp
Trying conf file at "rtl_433.conf"...
Trying conf file at "/home/abraxas3d/.config/rtl_433/rtl_433.conf"...
Trying conf file at "/usr/local/etc/rtl_433/rtl_433.conf"...
Trying conf file at "/etc/rtl_433/rtl_433.conf"...
Registered 96 out of 120 device decoding protocols [ 1-4 8 11-12 15-17 19-21 23 25-26 29-36 38-60 62-64 67-71 73-100 102-103 108-116 119 ]

Abraxas3d, 2019-02-07:
librtlsdr insisted it was already installed. Rebuilding gave the same results. 

So I backed up and removed everything rtl-sdr related that I could find, reinstalled everything in order, and then got the right version:

rtl_433 version 18.12-86-g931c9ce branch master at 201902061752 inputs file rtl_tcp RTL-SDR

And it appears to be working! 

Thank you for the encouraging words and the expected version. The mismatch was the right clue, I reckon.

zuckschwerdt, 2019-02-07:
We already feared that this kind of error might creep up. The CMake script autoselects the currently available backends. We should maybe demand the full  set of requirements and prompt the user to explicitly say something like DISABLE_RTL_SDR=1 (or here in your case DISABLE_SOAPYSDR=1).

Perhaps a first safety catch would be to have a better error message if the device string looks like RTL-SDR but it's not compiled in. On it!

zuckschwerdt, 2019-02-07:
Addressed with 19877ad now.

closed 2019-02-07 by zuckschwerdt