merbanan/rtl_433 #984: NC-7415 (InFactory) identical/compatible to Kedsum Temperature & Humidity Sensor

devZer0, 2019-02-16:
i tried NC-7415 from Pearl ( https://www.pearl.de/a-NC7427-3041.shtml ) with sdr 433 today and to my suprise, it works without great tuning effort.

the only thing i needed to set was set "-C si" as it would print fahrenheit by default, otherwise.

could you add this to the list?

furthermore,  here is some in depth thread about protocol decoding and i'm curious about sensor ID, 

as you can see here, there should be some id:

```
https://forum.fhem.de/index.php/topic,58110.15.html

[font=courier]if ($readedModel eq "NC_7427" || (hex($a[0]) == 0x5 && $readedModel eq "Unknown")) {
 # PEARL NC7427-675,
 #				/----------------------------------------------------- Leading-Sequence always 00								
 #				/  /--------------------------------------------------- ID, changes after every battery change
 #			   /  /		   /-------------------------------------------	Battery state 10 == Ok, 01 == not bad, 00 == bad
 #			  /	/		  /	/----------------------------------------- Channel 00 == Ch1, 01 == Ch2, 10 == Ch3
 #			/	/		/	/  /--------------------------------------- Temperature
 #			/  /		/  /  /			  /----------------------------	Humidity
 #		   /  /		   /  /	/			/		   /------------------- CRC
 #        /  /        /  /  /           /         /
 #       00 01101100 00 01 001101000110 01000010 00010100000000
 # Bit   01 23456789 01 23 456789012345 67890123 45678901234567
 #                   1           2          3          4
 # Byte  |  1 | 2 |  3  |  4 | 5 | 6 | 7  | 8 |  9 |10 | 11| 12
```

without printing sensor id, this device is meaningless to me because limited to three devices.

this is the output i get:
```
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2019-02-16 11:42:02
model     : Kedsum Temperature & Humidity Sensor   Channel   : 1             Temperature: 22.17 C      Humidity  : 48 %
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2019-02-16 11:42:52
model     : Kedsum Temperature & Humidity Sensor   Channel   : 1             Temperature: 22.39 C      Humidity  : 47 %
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2019-02-16 11:43:42
model     : Kedsum Temperature & Humidity Sensor   Channel   : 1             Temperature: 22.56 C      Humidity  : 46 %
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2019-02-16 11:44:32
model     : Kedsum Temperature & Humidity Sensor   Channel   : 1             Temperature: 22.67 C      Humidity  : 46 %
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2019-02-16 11:45:22
model     : Kedsum Temperature & Humidity Sensor   Channel   : 1             Temperature: 22.78 C      Humidity  : 45 %

_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2019-02-16 11:46:12
model     : Kedsum Temperature & Humidity Sensor   Channel   : 1             Temperature: 22.89 C      Humidity  : 45 %

```

merbanan, 2019-02-16:
Can you supply 2 sample signals from your device? Change the battery between the 2 recordings. That way we can investigate if the protocol specification is correct. It should be easy to add if we get a sample signal.

zuckschwerdt, 2019-02-16:
The ID is marked in the Kedsum source but strangle not added to the output. I'll fix that right away. Sample to confirm the protocol are always great to have.

devZer0, 2019-02-16:
yes, will submit, but give me some time.

zuckschwerdt, 2019-02-16:
The decoder got a major make-over with 70222da. ID and Battery are added as well as a MIC check (non-digest LFSR-4 comparable to a CRC-4).
The model has changed and is now simply "Kedsum-TH".

devZer0, 2019-02-17:
i tried the latest commit and for my curiousity my "kedsum" turned into a "Conrad-S3318P".

"rtl_433 -R57 -C si" does not decode anything  - so does rtl_433:master (it works when specifying -R47 manually)

"house code" DOES change on battery swap - but, btw - why is that called "house code" instead of "device id" ?

_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2019-02-17 13:19:59
model     : Conrad-S3318P House Code: 69
Channel   : 3            Battery   : OK            Button    : 0             Temperature: 19.00 C      Humidity  : 50 %          Integrity : CRC
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _

_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2019-02-17 13:18:08
model     : Conrad-S3318P House Code: 2
Channel   : 3            Battery   : OK            Button    : 1             Temperature: 18.72 C      Humidity  : 49 %          Integrity : CRC
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2019-02-17 13:18:12

zuckschwerdt, 2019-02-17:
Kedsum, Conrad S3318P, and Esperanza EWS are the exact same protocol, just with slightly different timing/sync. Will be interesting to see how your NC-7415 samples compare on timing and sync.

"House Code" is a misnomer, it's not a set code but really a random id. I'll change that name.

devZer0, 2019-02-17:
could you please provide a solution for easier upload of test-data? cloning the big repo is a major pain in the ass (as i have slow/flaky internet connection today)

zuckschwerdt, 2019-02-17:
It's okay to just zip the files and drop them here.

devZer0, 2019-02-17:
apparently there is some other sensors around...hope you can filter it appropriately. for my curiousity , it does not decode with -r

[NC-7415_samples.tar.gz](https://github.com/merbanan/rtl_433/files/2873013/NC-7415_samples.tar.gz)

devZer0, 2019-02-17:
mhh, think there are timing issues as for now they don't decode at all...

zuckschwerdt, 2019-02-17:
Looks like there is faint but constant transmission on 433.980 MHz. Drop a file onto http://triq.net/iqs to see what I mean. Could interesting to locate that. It could be a harmonic but I'd guess it's something like a beacon.
Your device is on 433.845 MHz but I'll need to investigate further to see why it's not decoded.

devZer0, 2019-02-17:
make sure you get the right file, i changed it because the first one was not good. 
```
NC-7415_samples roland$ ls -la
total 3584
drwx------@   12 roland  staff     384 17 Feb 15:25 .
drwxr-xr-x+ 1213 roland  staff   38816 17 Feb 15:25 ..
-rw-r--r--@    1 roland  staff  524288 17 Feb 15:23 g001_433.92M_250k.cu8
-rw-r--r--@    1 roland  staff  262144 17 Feb 15:23 g002_433.92M_250k.cu8
-rw-r--r--@    1 roland  staff  131072 17 Feb 15:23 g003_433.92M_250k.cu8
-rw-r--r--@    1 roland  staff  131072 17 Feb 15:23 g004_433.92M_250k.cu8
-rw-r--r--@    1 roland  staff  131072 17 Feb 15:23 g005_433.92M_250k.cu8
-rw-r--r--@    1 roland  staff  131072 17 Feb 15:24 g006_433.92M_250k.cu8
-rw-r--r--@    1 roland  staff  131072 17 Feb 15:24 g007_433.92M_250k.cu8
-rw-r--r--@    1 roland  staff  131072 17 Feb 15:24 g008_433.92M_250k.cu8
-rw-r--r--@    1 roland  staff  131072 17 Feb 15:24 g009_433.92M_250k.cu8
-rw-r--r--@    1 roland  staff  131072 17 Feb 15:24 g010_433.92M_250k.cu8
```

devZer0, 2019-02-17:
btw, the frequency shown at the mouse-pointer looks wrong to me

[image]

zuckschwerdt, 2019-02-17:
In your files the data decodes (ID: 25  Channel: 3  Battery: LOW  Temperature: 68.70 F Humidity: 51 %), but the captures are all incomplete, truncated at the beginning, not showning a complete transmission. Thus the integrity checks (minimum number of sync pulses, number of repeats) fail.
This might be different when reading live data.
Get this file and visualize it as comparison. https://github.com/merbanan/rtl_433_tests/blob/master/tests/kedsum/kedsum-60.0f-66h.cu8
I.e. try to grab a file that has: some silence, then a number of sync pulses (15 in the example), then a number of repeated pattern (6 in the example).

devZer0, 2019-02-17:
receiving range of that device really sucks, at least i don't get something received when it's more then 3 meters away...

and there must be timing issue, see below. batteries are fresh and at 1.6V

```
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2019-02-17 15:05:15
model     : Conrad-S3318P House Code: 16
Channel   : 1            Battery   : OK            Button    : 0             Temperature: 69.70 F      Humidity  : 48 %          Integrity : CRC
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2019-02-17 15:05:30
model     : Kedsum-TH    ID        : 16
Channel   : 1            Battery   : LOW           Temperature: 69.60 F      Humidity  : 48 %          Integrity : CRC
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
```

zuckschwerdt, 2019-02-17:
Btw. Kedsum is identified by 15 sync pulses, S3318P by 0 syncs, and Esperanza EWS by 2 syncs. That's why it switch from Kedsum to S3318P for you. The syncs were missing :)

zuckschwerdt, 2019-02-17:
The reception problems could be due to the "beacon" becoming too strong when your device is farther away/weaker. Try to tune down so the beacon is outside the band. E.g. try `-f 433.78M` (band will be 433.66–433.90, masking out the beacon at 433.98).

devZer0, 2019-02-17:
i set to -f 433.74M, with this the beacon is not visible in the graph anymore - but i still get intermittend 
detection jump between Kedsum and Conrad device

time      : 2019-02-17 15:30:32
model     : Kedsum-TH    ID        : 16
Channel   : 1            Battery   : LOW           Temperature: 69.40 F      Humidity  : 49 %          Integrity : CRC
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2019-02-17 15:30:35
model     : Kedsum-TH    ID        : 16
Channel   : 1            Battery   : LOW           Temperature: 69.40 F      Humidity  : 49 %          Integrity : CRC
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2019-02-17 15:30:37
model     : Conrad-S3318P House Code: 16
Channel   : 1            Battery   : OK            Button    : 1             Temperature: 69.40 F      Humidity  : 49 %          Integrity : CRC
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2019-02-17 15:30:39
model     : Kedsum-TH    ID        : 16
Channel   : 1            Battery   : LOW           Temperature: 69.40 F      Humidity  : 49 %          Integrity : CRC

devZer0, 2019-02-17:
does this look better for analysis ?

[g004_433.74M_250k.cu8.gz](https://github.com/merbanan/rtl_433/files/2873100/g004_433.74M_250k.cu8.gz)

zuckschwerdt, 2019-02-17:
Looks great! So it is actually identical to the Kedsum (15 syncs, 6 repeats) if received properly ;)
I'll add this sample to the repo, then merge #985 in a few days and with that support should be optimal.

devZer0, 2019-02-17:
i don't think so, because i am using your git version tagged zuckschwerdt:feat-kedsum for testing and i'm not receiving properly.  i still see it switching between Conrad-S3318P and Kedsum-TH and with Kedsum-TH it's telling "battery = low", which is untrue. and for both i'm getting CRC error

devZer0, 2019-02-17:
at different location i get stable Kedsum-TH output, but battery is still shown "low" (which isn't true) and showing "Integrity : CRC"  (did i understand correct that this means fail!? - being unsure, a more descriptive value "crc error/fail" should be better understandable)

zuckschwerdt, 2019-02-17:
`Integrity` will show the level of checks done. In this case a `CRC` **succeeded**.

The battery low code looks suspicious, I guess those bits mean something else ;)
Are you able to test a battery low situation? Do you have a known flat battery or better a variable voltage power supply?

devZer0, 2019-02-17:
i put in some old batteries and now get "Battery   : WEAK". but being curious why it shows "ok" with S3318P and "LOW" with ok batteries

time      : 2019-02-17 21:17:45
model     : Kedsum-TH    ID        : 157
Channel   : 1            Battery   : LOW           Temperature: 66.60 F      Humidity  : 41 %          Integrity : CRC
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2019-02-17 21:18:32
model     : Conrad-S3318P House Code: 157
Channel   : 1            Battery   : OK            Button    : 0             Temperature: 66.60 F      Humidity  : 41 %          Integrity : CRC

zuckschwerdt, 2019-02-18:
Well, it's just some bit in the data. Someday somebody assumed it might be the battery_low bit. I have never seen this case, where it's two bits: OK, WEAK, LOW. So that doesn't seem right.

I have now added flags debug to the branch `feat-kedsum` (you might need to reset the branch before pulling, or just clone the repo again). Test some scenarios and report what flags1 and flags2 could mean.

zuckschwerdt, 2019-02-28:
Any luck documenting the battery_low bit with the debug output currently in the `feat-kedsum` branch? Otherwise I'll clean that up and merge and we tackle that bit another day.

zuckschwerdt, 2019-03-10:
Code is merged now. Reopen if you discover how `flags` relates to the battery status.

closed 2019-03-10 by zuckschwerdt