merbanan/rtl_433 #989: Question: Disable Protocol

cbmboy, 2019-02-23:
I have read through the readme and many issues on here and Google and have not been able to find an answer.  Currently I am working to implement RTL_433 and it worked a little to well.

I appear to have a neighbor who has an Acurite 606tx sensor that is broadcasting 24/7 and bombarding my receiver.  I am sending this to an MQTT broker, and I wanted to disable this protocol so that it (hopefully) doesn't interfere with the devices I want to receive from.

Anyway, from what I understand I should be able to use -R -55 (negative 55) to disable the protocol.  But when I do this, 0 protocol's register.  It doesn't seem to be working.

So can someone tell me what I am doing wrong? Is this the correct (and simple) context:
RTL_433 -R -55

I have confirmed that using 55 instead of -55 does enable this one protocol, but using the negative (as the readme of RTL_433 outlines) does NOT appear to disable this one protocol.

Thanks in advanced!
DeadEnd

zuckschwerdt, 2019-02-23:
Are you using a recent version? The protocol selection only recently gained the negative options. It should look like this:
```
rtl_433 -R -55
rtl_433 version 18.12-113-g43bb177 branch master at 201902230920 inputs file rtl_tcp RTL-SDR SoapySDR
…
Registered 95 out of 120 device decoding protocols [ 1-4 8 11-12 15-17 19-21 23 25-26 29-36 38-54 56-60 62-64 67-71 73-100 102-103 108-116 119 ]
```

cbmboy, 2019-02-23:
When I start it I only get ## out of 90 devices, so I think not.

I am using a docker container. I believe I'll  need to modify it then to have it pull the latest instead of the version it was built with.

Learning a lot of new stuff!
Thanks for the help!

closed 2019-02-23 by cbmboy