merbanan/rtl_433 #991: Unitec W186-F works with AlectoV1 protocol

Heimetli, 2019-02-24:
A sensor called Unitec W186-F (bought from Migros) works fine using the AlectoV1 protocol.

However, I have to cleanup the output, because I sometimes get readings from another sensor with unbelievable values (e.g temperature -204.700).

This signal is rather strange because it always has a sensor id of 154, which is not mine. It is possible that there is another sensor near my location, but a temperature of -200 is unbelievable. The temperature is not constant, I have also seen values of 102.55 and 0.100

In short: it works for me, but users should be careful when analyzing the output.

zuckschwerdt, 2019-02-24:
Can you grab a sample (`rtl_433 -S all`) of the alien signal? Use http://triq.net/iqs to visualy inspect the samples. We can then improve false positive detection.

Heimetli, 2019-02-24:
> Can you grab a sample (`rtl_433 -S all`) of the alien signal?
It is probably a collision of two sensors, the Prologue and the AlectoV1.

When all protocols are enabled I also get wrong readings for "Springfield Temperature & Moisture" and  "TFA Twin Plus 30.3049".

> Use http://triq.net/iqs to visualy inspect the samples.
I did, but I am missing a zoom function...

Attached is the wrong reading for the AlectoV1.The temperature at the time was about 8.4 degrees celsius.

Do you want the other traces too? (Springfield and TFA)
[trace.zip](https://github.com/merbanan/rtl_433/files/2898202/trace.zip)

zuckschwerdt, 2019-02-24:
I'm always happy to look through samples that are no supported by our protocols but produce collisions.
Just look at the spectrogram (http://triq.net/iqs) to make sure it's a proper sample, no need to analyze it ;)

zuckschwerdt, 2019-02-24:
Looks like that signal is a Prologue-type sensor which by accident matches the AlectoV1 checksum. I'll have a closer look soon.
```
time      : @0.124240s
model     : Prologue sensor                        id        : 5
rid       : 147          Channel   : 1
Battery   : OK            Button    : 0
Temperature: 7.60 C       Humidity  : 20 %
```

Heimetli, 2019-02-24:
> Looks like that signal is a Prologue-type sensor which by accident matches the AlectoV1 checksum.
The symptoms would support this:
* it does not occur that often
* it can happen a few times in a row

Files with other errors are attached. Do you want more of them?

[tfa.zip](https://github.com/merbanan/rtl_433/files/2898355/tfa.zip)
[springfield.zip](https://github.com/merbanan/rtl_433/files/2898356/springfield.zip)

zuckschwerdt, 2019-04-13:
There is nothing to distinguish the "51.30 C" signal (g200_433.92M_250k.cu8) from other AlectoV1 transmissions. It fits the format an decodes cleanly. Either the checksum is too weak or that other sensor uses the same protocol with different data.

zuckschwerdt, 2019-08-28:
Reopen if anyone has a working idea how to distinguish those sensor types.

closed 2019-08-28 by zuckschwerdt