merbanan/rtl_433 #1000: transmit to ambient weather reciever

dwc54, 2019-02-28:
i know this is program is not made for transmitting i am trying to find how to encode the data 

i want to transmit to a ambient weather display its sensors use protocol 20 Ambient Weather Temperature 
i have a 433mhz transmitter how can i find how to encode the data and the timing for this protocol?

zuckschwerdt, 2019-02-28:
If that is one of those OOK transmitters: try `-w OOK:FILENAME.txt` or `-w OOK:-` for output to the terminal. Works live or when reading sample data.

Word of warning. Wire the transmitter wrong, or toggle the pin wrong and you will have a jammer for the whole neighbourhood. Always check operation with a second recevier.

merbanan, 2019-02-28:
Find or record a signal for this sensor. Then use sigrok-open.sh to look how the bits are sent over the air. Then look at the code to match the bits to the payload. When you know that try to clone a working signal. When you got that right you can start playing with sending arbitrary data.

closed 2019-02-28 by merbanan

dwc54, 2019-03-01:
when i run -w OOK:- it outputs two column of numbers is that the on time and off time of the pulses?

zuckschwerdt, 2019-03-01:
Yes, that should be the timing for any OOK transmitter.

dwc54, 2019-03-01:
how is that converted into binary  data for decoding?

dwc54, 2019-03-01:
when i use -vvvv it says 
`pulse_demod_manchester_zerobit(): Ambient Weather Temperature Sensor
bitbuffer:: Number of rows: 1
[00] {219} 7f ff ff c0 0a 2f 69 21 78 65 58 7f f5 17 b4 90 bc 32 ac 3f fa 8b da 48 5e 19 56 00
`

so i want to transmit 
7f ff ff c0 0a 2f 69 21 78 65 58 7f f5 17 b4 90 bc 32 ac 3f fa 8b da 48 5e 19 56 00
using "pulse_demod_manchester_zerobit() " how does "pulse_demod_manchester_zerobit()" work?

merbanan, 2019-03-01:
@dwc54 you need to read up on how manchester coding works. 219 bits will require some work to get correct.