merbanan/rtl_433 #997: Add Model and ID to MQTT topic

cbmboy, 2019-02-28:
Hello whoever is reading this.
I am very new to github and just started toying the RTL_433.
This may be the incorrect way to do this (I am thinking it might be a pull request) - but I know you prefer to keep this as simple as possible, so this might not be something you want to put into the repository.

BUT - I was very excited to figure this out (since I am so inexperienced).
I have multiple identical devices that I want to use RTL_433 to send MQTT messages for.  The problem is I want to be able to separate them with topics, but RTL_433 by default does not do this.

SO - my google-fu led me to a working solution.
Here are the simple steps I used to do this:

1) Add **_jq_** to the apt install list
2) Add the following modification to rtl2mqtt.sh

After the first echo $line add:

> DEVICE="$(echo $line | jq --raw-output '.model' | tr ' ' '_')"
> ID="$(echo $line | jq --raw-output '.id')"

Change the mosquitto_pub topic to:
>"RTL_433/Raw/$DEVICE/$ID"

Well, that's it... 
This might be common knowledge for those here who are experienced, but I was happy to solve my problem, and wanted to share :-)

Cheers!
DeadEnd

zuckschwerdt, 2019-02-28:
That's a good post-processing hint. We currently rely on either that shell script to bridge to MQTT or something like https://github.com/zuckschwerdt/rtl_433_relay/blob/master/rtl_433_mqtt_relay.py – which is easier to customize and has better performance.

We also have working code for MQTT in rtl_433 to support that directly. It's just not clear how topics and payloads should be formed.
E.g. identify the device by `model/id` or `model/id/channel` or `model/(channel otherwise id)`?
Put JSON at that topic or split some/all values to sub-topic, e.g. `model/channel/temperature` seems nice. And what about units? Some one has to investigate all this, and make a proposal

mikejac, 2019-03-04:
Maybe use the Homie convention: [An MQTT Convention for IoT/M2M](https://homieiot.github.io) ??

/michael

closed 2019-03-06 by DeadEnded

bademux, 2019-03-12:
Sorry for resurecting closed topic.
@zuckschwerdt [OpenHab supports](https://www.openhab.org/blog/2018-12-16-mqtt-arrives-in-the-modern-openhab-2-x-architecture.html#mqtt-thing-auto-discovery)  [Homie 3.x specification](https://homieiot.github.io/) and  [HomeAssistant MQTT Components specification](https://www.home-assistant.io/docs/mqtt/discovery/) so maybe one of those?

zuckschwerdt, 2019-03-13:
Thanks for the suggestions and documentation links. Both formats seem very verbose if we need to support our very generic sensor types. But I guess we can add both as options. (e.g. something like `-F mqtt:192.168.1.2:1883,homie=1,hamcs=1`)
For a first test and to develop the necessary translation we should use https://github.com/zuckschwerdt/rtl_433_relay/blob/master/rtl_433_mqtt_relay.py

bademux, 2019-03-13:
Sounds great! 
Just small suggestion: 
- I'm not shure what *hamcs* means, but if it stays for HomeAssistans (guys use [*hass* as shortage](http://hass.io)), than it is better to use hassmcs (yeah too *ssss*) or just hass.
- Do we need ability to send 2 types of messages to the same topic at the same time? How about type={homie,hass}
- *mosquitto_pub* supports *-L* (*--url*) option that allows to provide connection in standard way as URI mqtt(s)://[username[:password]@]host[:port]/topic.
I looks like it will fit into rt_433 console arg design with no pain.
In that way we can add *homie=1,hamcs=1* can be implemented as root path: mqtt(s)://[username[:password]@]host[:port]/{homie,hass}/topic

Anyway thanks for the project!