merbanan/rtl_433 #998: More unified looks

gvanem, 2019-02-28:
Currently different sensors could display the `Temperature: xx.x C` at different positions
on the screen. Example screen-output:
[image]

I do not understand how it could be different.  
But could the data in the `output_fields[]` be unified somehow?

PS. I used no config-file (`-c NUL` ) above. So nothing in `rtl_433.conf` could be the guilty part.

zuckschwerdt, 2019-02-28:
There is no sorting, it's just fields as the decoder adds them. We should perhaps rewrite decoders to have some fixed order of well known keys:
- model (ok)
- id (ok, not sure where that rid fits)
- channel (ok, but might be missing)
- temperature (move this up)
- humidity (move this up)
- button, battery, … (move down)

But there is rather much diversity, there are probes with two temperature values, the temperature might not be the important part (TPMS), … Hard to say if there could be some consensus.

gvanem, 2019-02-28:
> * temperature (move this up)

Maybe if the `.name` contains `Temperature`, that should determine what that **main-value** is?

zuckschwerdt, 2019-09-05:
Todo: Key order (for Prologue here) should generally be
(type), model, (subtype), id, channel, battery_ok, temperature_x, humidity, …, and mic last. At least that seems to be the most used order in decoders.

zuckschwerdt, 2019-09-06:
Most of this is done with 5e4c7dc now.

closed 2019-09-06 by zuckschwerdt