merbanan/rtl_433 #1002: Norgo NGE101 device support

martinkaltoft, 2019-03-08:
It would be nice if someone would look into adding device support for the 'Norgo NGE101 Energy Meter'

Test files:
https://github.com/merbanan/rtl_433_tests/tree/master/tests/NGE101

The hard work of analyzing and decoding the protocol have been done here:
http://blog.bitheap.net/2011/01/nge101-norgo-wireless-energy-meter-part_25.html

I simply lack the ability to make this info into a working decoder.

jamaron, 2019-03-24:
Thanks to protocol description by Jesper Hansen, I was able to implement NGE101 decoder for rtl_433. I have been using it for some time now and it is working very well. Unfortunately I do not have time to finalize it for releasing right now, but hope I can find some time to do that in near future.

martinkaltoft, 2019-03-30:
That sounds great jamaron, I hope you find the time to make it public.

zuckschwerdt, 2019-04-06:
Just dump the code here and I'll clean that up and integrate it.
Otherwise if somebody wants to go through those blog posts and do a short writeup I might write a decoder.

jamaron, 2019-04-08:
[norgo.zip](https://github.com/merbanan/rtl_433/files/3055511/norgo.zip)

Please find attached norgo.c and a photo of the device. Pulse width parameters may still need some fine tuning.

zuckschwerdt, 2019-04-08:
The code already pretty much looked good, and works. PR in #1042, needs a bit more clean up. Closing this in favor of the PR.

closed 2019-04-08 by zuckschwerdt