merbanan/rtl_433 #1004: Problem with 3-in-1 Temperature decoding

RDK-45, 2019-03-10:
I'm using the RTL_433 functions on a Raspberry Pi running Debian/Stretch.  The RTL_433 system was install from the GIT site in November 2018.  It is monitoring several Acurite sensors (towers, outdoor air&liquid temperature sensor, 5-in-1 and 3-in-1).  Except as indicated below it is working fine.

My 3-in-1 has been giving bad temperature data off and on.  One evening I was able to capture some raw data records (rtl_433 -R 40 -F json -D &> debug030419.txt) when it was reporting -140F temperature values.  The next morning it was working correctly so I captured more records (debug030519.txt) when it was reporting correct temperatures (mid 50’s F).  In both cases the Acurite Dashboard was reporting correct temperatures during those periods, so I do not suspect that the sensor is failing.
[debug030519.txt](https://github.com/merbanan/rtl_433/files/2949225/debug030519.txt)
[debug030419.txt](https://github.com/merbanan/rtl_433/files/2949226/debug030419.txt)

Note the attached files also have outdoor data for a 5-in-1 unit for temperature comparisons.  The units are about 50 m apart and depending on time of day (morning or evening) one will be in the sunlight but not the other.  The tower unit also being reported in not outside.

Perhaps the attached files will help identify the issue??

Thanks...Rob

zuckschwerdt, 2019-03-10:
The Acurite code sadly obfuscates the data layout somewhat. Take a look at [acurite.c](https://github.com/merbanan/rtl_433/blob/master/src/devices/acurite.c) and construct the meaning of the bits in e.g. [this BitBench](http://triq.net/bitbench?c=%5B%20Acurite%203n1%20sensor%2C%20ID%203650%2C%20Ch%20A%2C%20Seq%200%2C%20batt%20OK%2C%20type%2032%20%5D&c=CE42609C903006D2%20%5B%206%20MPH%2C%20-143.2%20F%2C%2028%20%25%20%5D&c=CE42609C903006D2%20%5B%206%20MPH%2C%20-143.2%20F%2C%2028%20%25%20%5D&c=CE4260210FEB018C%20%5B%201%20MPH%2C%20%20%2054.7%20F%2C%2033%20%25%20%5D&c=CE4260210F6A020C%20%5B%202%20MPH%2C%20%20%2054.6%20F%2C%2033%20%25%20%5D&f=8h%20). The you can see if maybe the top temperature bit is perhaps something else.

zuckschwerdt, 2019-03-10:
The format looks like perhaps (the `1x` are parity bits)
```
C:2d R:2d ID:12d 1x BATT:1b TYPE:6h 1x ?2b W:5b 1x 3b T:4b 1x 7b S: 1x 7d
```
And I guess the temperature is just missing the top bit by the looks of it.

zuckschwerdt, 2019-03-10:
Should be fixed (with 34287e1) now.

RDK-45, 2019-03-10:
Christian…..I have compared the copy of acurite.c you referenced and that which I installed in November.  The copy I’m using uses a function “acurite_getTemp” to calculate the raw_temp for both the 5-in-1 and the 3-in-1, but with different intercepts.  The copy you referenced has removed that function and the temperatures are calculated separately in each 5-in-1 or 3-in-1 section and the algorithms are not the same, with that for the 3-in-1 having a different mask for the high bit (0x1F vs 0x0f).  

 

I can’t do it now, but perhaps Tuesday I will reinstall the RTL_433 package with all new code downloaded from the GIT site.

 

I’ll get back to you when I have done this….RDK

Rob Koller Sr

zuckschwerdt, 2019-03-10:
Yes, I figured out the Acurite format in the meantime and fixed the temperature value. The "0x1F vs 0x0f" change is that fix.

In the process I moved all the helper function code, like acurite_getTemp, to the lines where it is actually used. Makes it much easier to glance where all the bits go.

Should be good to use. Install current master, test for a while (temperatures over 56.7 F show that it works), then close this issue.

RDK-45, 2019-03-10:
Christian…I did not see this before I send my last note with comments about the differences in the acurite.c modules. 

 

Will your modifications be in the latest clone from the GIT site?...RDK

 

Rob Koller Sr

zuckschwerdt, 2019-03-10:
Yes, the current Git "master" branch has this fix included.

RDK-45, 2019-03-11:
Ok, I have updated that Pi and it is not throwing that error right now.  I'll watch it for a few days before closing the issue.

By the way, is there a build number or something I can display for this install that would differentiate it from a previous install? 

Many thanks for your efforts and this solution...RDK

zuckschwerdt, 2019-03-11:
The version is the first line of output and should currently be:
```
rtl_433 version 18.12-133-g5a960e9 branch master at 201903101441
```

RDK-45, 2019-03-12:
I'm trying to collect some more data as now I'm seeing another error not related to the one in this issue, but the previous command `rtl_433 -R 40 -F json -D &> debug031219.txt` no longer gives me the HEX raw data record for the decoded data values.  Below is an example from my previous use of this command and shows the raw hex data:
`acurite_txr Parity: 10000000
Acurite 5n1 raw msg: CE 42 60 9C 90 30 03 CF
{"time" : "2019-03-04 18:37:38", "model" : "Acurite 3n1 sensor", "sensor_id" : 3650, "channel" : "A", "sequence_num" : 0, "battery" : "OK", "message_type" : 32, "wind_speed_mph" : 3.000, "temperature_F" : -143.200, "humidity" : 28}
`
The output says the -D is depreciated and to try -v.  I have tried all of the -v options (ie -v, -vv, -vvv and -vvvv) but none show the hex data record.  Is there another option or command I should be using? Thanks...RDK

zuckschwerdt, 2019-03-12:
Verbosity level 2 (`-vv`) should work for the Acurite sensors.

RDK-45, 2019-03-12:
Christian…As I said, I tried all 4 levels of verbosity without success….RDK

 

Rob Koller Sr

France 33-6-74909740

 <http://www.rdkscorner.com/rdk/> WWW.RDKsCorner.Com

zuckschwerdt, 2019-03-12:
Can you confirm which "model" key exactly you are seeing?

RDK-45, 2019-03-12:
Christian…Here are the first few records for this command: 

rtl_433 -R 40 -F json -vv &> /mnt/usbdrive/debug031219vv.txt

 

rtl_433 version 18.12-133-g5a960e9 branch master at 201903101441 inputs file rtl_tcp RTL-SDR

Trying conf file at "rtl_433.conf"...

Trying conf file at "/home/rob/.config/rtl_433/rtl_433.conf"...

Trying conf file at "/usr/local/etc/rtl_433/rtl_433.conf"...

Trying conf file at "/etc/rtl_433/rtl_433.conf"...

 

     Consider using "-M newmodel" to transition to new model keys. This will become the default someday.

     A table of changes and discussion is at https://github.com/merbanan/rtl_433/pull/986.

 

Registered 1 out of 121 device decoding protocols

Found 1 device(s)

 

trying device  0:  Realtek, RTL2838UHIDIR, SN: 00000001

Detached kernel driver

Found Rafael Micro R820T tuner

Using device 0: Generic RTL2832U OEM

Exact sample rate is: 250000.000414 Hz

[R82XX] PLL not locked!

Sample rate set to 250000 S/s.

Bit detection level set to 0 (Auto).

Tuner gain set to Auto.

Reading samples in async mode...

Tuned to 433.920MHz.

{"time" : "2019-03-12 14:27:29", "model" : "Acurite tower sensor", "id" : 11824, "sensor_id" : 11824, "channel" : "A", "temperature_C" : 20.200, "humidity" : 50, "battery_low" : 0}

{"time" : "2019-03-12 14:27:29", "model" : "Acurite tower sensor", "id" : 11824, "sensor_id" : 11824, "channel" : "A", "temperature_C" : 20.200, "humidity" : 50, "battery_low" : 0}

 

…RDK

Rob Koller Sr

France 33-6-74909740

 <http://www.rdkscorner.com/rdk/> WWW.RDKsCorner.Com

zuckschwerdt, 2019-03-12:
Oh I see! The options are positional, which is quite a pitfall. You actually need
```
rtl_433 -vv -R 40 -F json
```
so the verbosity is set when the protocol gets initialized, sorry for the confusion. I hope to address some of those option troubles soon.

RDK-45, 2019-03-13:
Christian.....Hmmm, I tried the change `rtl_433 -vv -R 40 -F json &> /mnt/usbdrive/debug031319vv.txt` but I still do not see the raw hex data.  Here is the first few lines from the output:

rtl_433 version 18.12-133-g5a960e9 branch master at 201903101441 inputs file rtl_tcp RTL-SDR
Trying conf file at "rtl_433.conf"...
...
trying device  0:  Realtek, RTL2838UHIDIR, SN: 00000001
Detached kernel driver
Found Rafael Micro R820T tuner
Using device 0: Generic RTL2832U OEM
Exact sample rate is: 250000.000414 Hz
[R82XX] PLL not locked!
Sample rate set to 250000 S/s.
Bit detection level set to 0 (Auto).
Tuner gain set to Auto.
Reading samples in async mode...
Tuned to 433.920MHz.
acurite_txr Parity: 0100001
{"time" : "2019-03-13 06:05:13", "model" : "Acurite tower sensor", "id" : 15602, "sensor_id" : 15602, "channel" : "A", "temperature_C" : 8.300, "humidity" : 52, "battery_low" : 0}
acurite_txr Parity: 0100001
{"time" : "2019-03-13 06:05:13", "model" : "Acurite tower sensor", "id" : 15602, "sensor_id" : 15602, "channel" : "A", "temperature_C" : 8.300, "humidity" : 52, "battery_low" : 0}
acurite_txr Parity: 0100001
{"time" : "2019-03-13 06:05:13", "model" : "Acurite tower sensor", "id" : 15602, "sensor_id" : 15602, "channel" : "A", "temperature_C" : 8.300, "humidity" : 52, "battery_low" : 0}
pulse_demod_pwm(): Acurite 592TXR Temp/Humidity, 5n1 Weather Station, 6045 Lightning 
bitbuffer:: Number of rows: 3 
[00] {57} fc f2 44 b4 88 bb 29 00 
[01] {57} fc f2 44 b4 88 bb 29 00 
[02] {57} fc f2 44 b4 88 bb 29 00 

Thanks....RDK

zuckschwerdt, 2019-03-13:
Isn't the last three lines what you are looking for? Otherwise you can try `-vvv` but that's a lot of output.

RDK-45, 2019-03-13:
Christian…..I’m not sure since further down there are fewer of those lines relative to the number of translated lines.  I’ll attach the latest output file for your command

 

What I think I’m looking for is what I used to get:

acurite_txr Parity: 01000000

Acurite 5n1 raw msg: C5 51 78 81 87 AF 21 66

{"time" : "2019-03-05 07:12:12", "model" : "Acurite 5n1 sensor", "sensor_id" : 1361, "channel" : "A", "sequence_num" : 0, "battery" : "OK", "message_type" : 56, "wind_speed_mph" : 4.736, "temperature_F" : 54.300, "humidity" : 33}

 

Where the raw hex is right up front with each translated record….RDK

 

Rob Koller Sr

 

Registering protocol [40] "Acurite 592TXR Temp/Humidity, 5n1 Weather Station, 6045 Lightning"

acurite_txr Parity: 01000000
Acurite 5n1 raw msg: { 8} c5 : 11000101 
{"time" : "2019-03-13 06:05:19", "model" : "Acurite 5n1 sensor", "sensor_id" : 1361, "channel" : "A", "sequence_num" : 0, "battery" : "OK", "message_type" : 49, "wind_speed_kph" : 0.000, "wind_dir_deg" : 180.000, "rain_inch" : 40.740}
acurite_txr Parity: 11000000
Acurite 5n1 raw msg: { 8} d5 : 11010101 
{"time" : "2019-03-13 06:05:19", "model" : "Acurite 5n1 sensor", "sensor_id" : 1361, "channel" : "A", "sequence_num" : 1, "battery" : "OK", "message_type" : 49, "wind_speed_kph" : 0.000, "wind_dir_deg" : 180.000, "rain_inch" : 40.740}
acurite_txr Parity: 11000001
Acurite 5n1 raw msg: { 8} e5 : 11100101 
{"time" : "2019-03-13 06:05:19", "model" : "Acurite 5n1 sensor", "sensor_id" : 1361, "channel" : "A", "sequence_num" : 2, "battery" : "OK", "message_type" : 49, "wind_speed_kph" : 0.000, "wind_dir_deg" : 180.000, "rain_inch" : 40.740}
acurite_txr Parity: 0000001
{"time" : "2019-03-13 06:05:25", "model" : "Acurite tower sensor", "id" : 2606, "sensor_id" : 2606, "channel" : "C", "temperature_C" : 4.700, "humidity" : 50, "battery_low" : 0}
acurite_txr Parity: 0000001
{"time" : "2019-03-13 06:05:25", "model" : "Acurite tower sensor", "id" : 2606, "sensor_id" : 2606, "channel" : "C", "temperature_C" : 4.700, "humidity" : 50, "battery_low" : 0}
acurite_txr Parity: 0000001
{"time" : "2019-03-13 06:05:25", "model" : "Acurite tower sensor", "id" : 2606, "sensor_id" : 2606, "channel" : "C", "temperature_C" : 4.700, "humidity" : 50, "battery_low" : 0}
[00] {57} 0a 2e 44 b2 88 17 cd 00 
[01] {57} 0a 2e 44 b2 88 17 cd 00 
[02] {57} 0a 2e 44 b2 88 17 cd 00 
Acurite 5n1 raw msg: { 8} ce : 11001110 
{"time" : "2019-03-13 06:05:26", "model" : "Acurite 3n1 sensor", "sensor_id" : 3650, "channel" : "A", "sequence_num" : 0, "battery" : "OK", "message_type" : 32, "wind_speed_mph" : 0.000, "temperature_F" : 40.500, "humidity" : 80}
Acurite 5n1 raw msg: { 8} ce : 11001110 
{"time" : "2019-03-13 06:05:26", "model" : "Acurite 3n1 sensor", "sensor_id" : 3650, "channel" : "A", "sequence_num" : 0, "battery" : "OK", "message_type" : 32, "wind_speed_mph" : 0.000, "temperature_F" : 40.500, "humidity" : 80}
acurite_txr Parity: 0000000
{"time" : "2019-03-13 06:05:27", "model" : "Acurite tower sensor", "id" : 11824, "sensor_id" : 11824, "channel" : "A", "temperature_C" : 19.600, "humidity" : 50, "battery_low" : 0}
acurite_txr Parity: 0000000
{"time" : "2019-03-13 06:05:27", "model" : "Acurite tower sensor", "id" : 11824, "sensor_id" : 11824, "channel" : "A", "temperature_C" : 19.600, "humidity" : 50, "battery_low" : 0}
acurite_txr Parity: 0000000
{"time" : "2019-03-13 06:05:27", "model" : "Acurite tower sensor", "id" : 11824, "sensor_id" : 11824, "channel" : "A", "temperature_C" : 19.600, "humidity" : 50, "battery_low" : 0}
[00] {57} ee 30 44 b2 09 ac c9 00 
[01] {57} ee 30 44 b2 09 ac c9 00 
[02] {57} ee 30 44 b2 09 ac c9 00 
{"time" : "2019-03-13 06:05:30", "model" : "Acurite tower sensor", "id" : 15602, "sensor_id" : 15602, "channel" : "A", "temperature_C" : 8.300, "humidity" : 52, "battery_low" : 0}
{"time" : "2019-03-13 06:05:30", "model" : "Acurite tower sensor", "id" : 15602, "sensor_id" : 15602, "channel" : "A", "temperature_C" : 8.300, "humidity" : 52, "battery_low" : 0}
{"time" : "2019-03-13 06:05:30", "model" : "Acurite tower sensor", "id" : 15602, "sensor_id" : 15602, "channel" : "A", "temperature_C" : 8.300, "humidity" : 52, "battery_low" : 0}
acurite_txr Parity: 01000001
Acurite 5n1 raw msg: { 8} c5 : 11000101 
{"time" : "2019-03-13 06:05:38", "model" : "Acurite 5n1 sensor", "sensor_id" : 1361, "channel" : "A", "sequence_num" : 0, "battery" : "OK", "message_type" : 56, "wind_speed_kph" : 0.000, "temperature_F" : 39.900, "humidity" : 81}
acurite_txr Parity: 11000000
Acurite 5n1 raw msg: { 8} d5 : 11010101 
{"time" : "2019-03-13 06:05:38", "model" : "Acurite 5n1 sensor", "sensor_id" : 1361, "channel" : "A", "sequence_num" : 1, "battery" : "OK", "message_type" : 56, "wind_speed_kph" : 0.000, "temperature_F" : 39.900, "humidity" : 81}
acurite_txr Parity: 11000000
Acurite 5n1 raw msg: { 8} e5 : 11100101 
{"time" : "2019-03-13 06:05:38", "model" : "Acurite 5n1 sensor", "sensor_id" : 1361, "channel" : "A", "sequence_num" : 2, "battery" : "OK", "message_type" : 56, "wind_speed_kph" : 0.000, "temperature_F" : 39.900, "humidity" : 81}
acurite_txr Parity: 0000001
{"time" : "2019-03-13 06:05:41", "model" : "Acurite tower sensor", "id" : 2606, "sensor_id" : 2606, "channel" : "C", "temperature_C" : 4.700, "humidity" : 50, "battery_low" : 0}
acurite_txr Parity: 0000001
[... 33 lines cut ...]
acurite_txr Parity: 0000001
[... cut ...]
{"time" : "2019-03-13 06:06:03", "model" : "Acurite tower sensor", "id" : 15602, "sensor_id" : 15602, "channel" : "A", "temperature_C" : 8.300, "humidity" : 52, "battery_low" : 0}
{"time" : "2019-03-13 06:06:03", "model" : "Acurite tower sensor", "id" : 15602, "sensor_id" : 15602, "channel" : "A", "temperature_C" : 8.300, "humidity" : 52, "battery_low" : 0}
{"time" : "2019-03-13 06:06:03", "model" : "Acurite tower sensor", "id" : 15602, "sensor_id" : 15602, "channel" : "A", "temperature_C" : 8.300, "humidity" : 52, "battery_low" : 0}
Signal caught, exiting!
Reattached kernel driver

zuckschwerdt, 2019-03-13:
Those lines are only there for the 5N1, the TXR doesn't have that. But you can copy
https://github.com/merbanan/rtl_433/blob/master/src/devices/acurite.c#L519
to https://github.com/merbanan/rtl_433/blob/master/src/devices/acurite.c#L485 to get that.

RDK-45, 2019-03-13:
Christian...OK, I'm beginning to understand that for the "Tower" sensors the "RAW Hex" data line was not displayed.  I missed that as before I was focused  on the 3-in-1 packets.  
I am now interested in seeing that info for both the "Tower" and "5-in-1/3-in-1" sensor types.  What I'm observing now is that the previous version of rtl_433 which I was using gave me this information line for the "5-in-1/3-in-1" sensors: 
`Acurite 5n1 raw msg: CE 42 60 9C 90 30 03 CF`
where now I get
`Acurite 5n1 raw msg: { 8} c5 : 11000101 `

OK, from your previous response (and I do apologize for hijacking issue 1004 thread for this unrelated issue) I assume I need to modify my version of acurite.c. with something like this for both the tower and 5-in-1 sections to get the complete raw hex data?
`        if (debug_output) {
            fprintf(stderr, "Acurite 5n1 raw msg: %02X %02X %02X %02X %02X %02X %02X %02X\n",
                bb[0], bb[1], bb[2], bb[3], bb[4], bb[5], bb[6], bb[7]);
        }`
Thanks again for your time and patience....RDK

zuckschwerdt, 2019-03-13:
The line above is already the full `bb` content. The interesting bits will be in other bitbuffer rows.
For a single row use: `bitrow_printf(row, len, "Message: ");`
And for the whole buffer: `bitrow_printf(bitbuffer, "Message: ");`
In your case `bitbuffer_printf(bitbuf, "Acurite TXR raw msg: ");`.

RDK-45, 2019-03-13:
I'll get back to this in a couple days since I think I need to reflect on this a bit and do some experiments.  In the mean time, since I have not observed any 3-in-1 temperature errors, I will assume you have solved the initial issue for this thread and thus will close it.  Thanks for your efforts....RDK

closed 2019-03-13 by RDK-45