merbanan/rtl_433 #1005: Problems decoding Acurite 606TX

doughecka, 2019-03-11:
I have 3, 606TX sensors that are not able to be reliably picked up. Every once in awhile I'll see one get picked up, but never with regularity. I am picking up my neighbors Acurite 5n1 sensor very well. 
When I run with -vv and force one of the sensors to transmit, I see this in the console:
AlectoV1 Checksum/Parity error
I can verify that message pops up each time one of the 606TXs transmit. If I try excluding the AlectoV1 (-R -16) I still see that message. If I enable ONLY the 606TX (-R 55) I see nothing in the console. 
I've attached the capture of the signal. 
I'm running the latest code from Master as of a few hours ago. 

[g010_433.92M_250k.zip](https://github.com/merbanan/rtl_433/files/2950253/g010_433.92M_250k.zip)

zuckschwerdt, 2019-03-11:
You seem to have a newer version that doesn't send sync pulses. I'll have a fix shortly. Your sample reads:
```
model     : Acurite 606TX Sensor                   id        : 198
Battery   : OK           Temperature: 23.4 C
```

zuckschwerdt, 2019-03-11:
Fixed (with 64139f3) in current master.

closed 2019-03-11 by zuckschwerdt