merbanan/rtl_433 #1006: Automated testing based on rtl_433_tests

ciaranmooney, 2019-03-11:
This is a proposal that automatic tests should be integrated (or easily run) using the very large set of tests available in the rtl_433_test repository.

I suggest that rather than just submitting recordings of new devices, that each recording should be accompanied by a file describing the expected output.

Hopefully, this will catch changes in the code base that that change the device authors expected output and prevent changes breaking peoples projects.

I intend to start writing such a test for the device I created (wg-pb12v2). if there is little or no appetite for this functionality then I won't worry about spending a lot of effort on it.

If implemented that will be a large back catalogue of recordings that will be missing output descriptions but this is a dull but easy thing to resolve.

zuckschwerdt, 2019-03-11:
That's what we currently do. Look at the json files and the python runner.

zuckschwerdt, 2019-03-12:
Things could be documented better though. If someone with a fresh view wants to write wiki texts, please do! I'll help out if questions arise.

closed 2019-03-12 by zuckschwerdt