merbanan/rtl_433 #1007: Chinese TPMS with external sensors

SensorsIot, 2019-03-12:
I like your software and want to thank you for your effort. Recently I got these TPMS sensors: http://bit.ly/2VVxIrP and would like to build a beer pressure sensor using these sensors. I did some investigations using GQRX and URH and think I know the waveforms, the signals, the position of the information as well as the formula. I put everything together in a file. 
https://www.dropbox.com/s/cb5l64d241p63uz/Docu%20TPMS.docx?dl=0

I also can provide you with the URH files if you want. 
It would be cool if these sensors would be detected by rtl_433. Since I got a few more sensors I would also be interested in a tutorial on how to build custom sensors (Also with the -M qualifier)

zuckschwerdt, 2019-03-12:
Hi Andreas, well that is a worthy application of TPMS! The signal you documented (50ms half-bit width, NRZ, Manchester) is common for TPMS. Shouldn't be any trouble to implement that in the next days. It would be great if you can supply a sample (preferably recorded with `rtl_433 -S all`) -- just attach it here.
In the meantime I'll take a closer look at your documented findings.

zuckschwerdt, 2019-03-12:
Here is a [BitBench](http://triq.net/bitbench?c=0aa6b79cc0011c79fffffeab5bce60008e3c&c=0aa6b79cc3291c71fffffeab5bce61948e38&c=0947fcad81251d81fffffe53fe56c0928ec0&c=0947fcad818d26c5fffffe53fe56c0c69362&c=0947fcad822d29c1fffffe53fe56c11694e0&c=0947fcad83052c3dfffffe53fe56c182961e&c=0947fcad80cd1d79fffffe53fe56c0668ebc&c=08a6342a416cf1f9fffffe2b1a1520b678fc&c=08a6342a41a91ce5fffffe2b1a1520d48e72&f=ID%3A36h%20%3F28h%20%3F24h%20%3F28h%20PRESSURE%3A12d%20TEMP%3A12d%20CHK%3F4h) to play with the decoding.

zuckschwerdt, 2019-03-12:
Actually if I look at the NRZ data you provide
```
54aaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaacb32d5554b334d2aab4b4cad4cd52d555555555559665aaaa96669a555696995a99aa58
```
I see the same string repeated twice, stuff one bit in the `555555` part and you get
```
9665aaaa96669a555696995a99aa5a5
9665aaaa96669a555696995a99aa58
```
which Manchester decoded gives `94ff95b019a3af` (might need inversion or 1-3 bit shifts to look sensible).

Can you provide a couple of raw NRZ lines? Should work with
```
rtl_433 -R 0 -X 'n=TPMS,m=FSK_PCM,s=50,l=50,r=1000'
```
Also is there an ID marked on the sensors (usually 7 hex digits)?

SensorsIot, 2019-03-12:
Hi Christian,
The URH offers 2 Manchester codes ( I and II). I found that Manchester II is also called Biphase L code. I have no knowledge of codes.

Here is the file from your first command. I added log because Github only accepted it like that.

[Chinese.log](https://github.com/merbanan/rtl_433/files/2958801/Chinese.log)

The second command resulted in an error message:
sdr@ThinkPad ~ $ rtl_433 -R 0 -X 'n=TPMS,m=FSK_PCM,s=50,l=50,r=1000'
Disabling all device decoders.
Bad flex spec, missing modulation!
Use -X <spec> to add a general purpose decoder. For usage use -X help

Concerning codes: These sensors are so small they have nothing printed on them. But I had 4 of them and found that the information I marked was sensor specific. I do not care if it is right or rong because I will use what my receiver shows ;-)

zuckschwerdt, 2019-03-13:
Biphase Level is the classical Manchester (G.E. Thomas) a very common choice.

Oh, sorry, you seem to have an older version installed (rtl_433 is moving pretty fast). The old syntax was:
```
rtl_433 -R 0 -X 'TPMS:FSK_PCM:50:50:1000'
```

The captured file is just silence (drop it onto http://triq.net/iqs to see the spectrogram).
Grab a few samples and use the spectrogram to look for a pattern (FSK should be two red "lines").
The URH sample files will also work, but the format is important:

- `.complex` is known as `.cf32` in rtl_433
- `.complex16u` is known as `.cu8` in rtl_433
- `.complex16s` is known as `.cs8` in rtl_433

If we take a bit string (or a Manchester decoded one) we need to decide if the bits are flipped (if it's actually G.E.Thomas-Manchester or rather IEEE-Manchester).
And we need to align the hex representation (shift by 0 to 3 bits). That's easier if we can spot the ID code ;) But it's not really needed.

The format could be `28 bit ID, 12 bit Pressure, 12 bit Temperature, 8 bit Checksum`.
(28 bit for the ID field is the common case for TPMS, 8 bit Checksum is seen on all other devices so far. 12 bit temperature is a higher precision that normal.)
Here is a [BitBench](http://triq.net/bitbench?c=ab5bce60008e3c%20%20%5BFL%20%20%200%20kPa%2021%20C%5D&c=ab5bce61948e38%20%20%5BFL%20340%20kPa%2021%20C%5D&c=53fe56c0928ec0%20%20%5BFR%20120%20kPa%2022%20C%5D&c=53fe56c0c69362%20%20%5BFR%20160%20kPa%2023%20C%5D&c=53fe56c11694e0%20%20%5BFR%20230%20kPa%2024%20C%5D&c=53fe56c182961e%20%20%5BFR%20320%20kPa%2025%20C%5D&c=53fe56c0668ebc%20%20%5BFR%20080%20kPa%2021%20C%5D&c=2b1a1520b678fc%20%20%5BRL%20150%20kPa%2010%20C%5D&c=2b1a1520d48e72%20%20%5BRL%20170%20kPa%2021%20C%5D&f=ID%3A28h%20P%3A12d%20T(HALF%20DEG%20C%20-FIFTY)%3A8d%20CHK%3A8h) with the decoding so far.

The temperature looks like it's Celsius scaled by 2 offset 50, i.e. raw_value/2-50 °C (for a total range of -50 °C to 77.5 °C).
The pressure looks like it could be PSI scaled by 8, or kPa scaled by 1/0.85.

The last byte is usually a checksum (CRC, add, or xor). To make a good guess we'd need code lines that are very close together (e.g. temperature or pressure changed by a very small amount).
The lines you decoded so far work well, the part after `ffffe` is already nicely aligned (as you already found out). If the command works, can you provide those code lines?

SensorsIot, 2019-03-13:
I worked now the whole day on my infrastructure to discover and test codes. I use URH to register data from the sensors ( and a bicycle wheel to create variable pressure and fluid air to cool the sensor down). Then I decode it with URH and create files from the packets (.complex). As the last step, I use GRC to send the packets and check on the display what it displays. I use these values to name the packets (RL- rear left sensor, first number pressure in bar, second number temperature in centigrade).

https://www.dropbox.com/s/da6o4bht9yfdkp0/TPMS.zip?dl=0

Here you find a zip file with plenty of examples. Unfortunately, I was not successful with your commands (in the meantime I installed the newest version of RTL_433). The "rtl_433 -S all" only created empty files and the rtl_433 -R 0 -X 'TPMS:FSK_PCM:50:50:1000' a lot of what I think is garbage.
You should find packets with temperature as well as pressure zero in the zip file. Also packets with only one degree difference.
I hope this helps you. I still do not get how the bitbench works :-(, even with your examples)

zuckschwerdt, 2019-03-14:
Thanks, the sample files are of good quality and usable. I take it you recorded with a HackRF/Pluto/SDRplay/LimeSDR at 8 MSps, not an RTL-SDR (at 250k or 1024k)? The timings seem to suggest that.

The FSK signal from the sensor is (very nicely) bandwidth limited (GFSK or MFSK) and the current detector in rtl_433 can't handle that very well -- that's why the bit output isn't usable.
I'll try to work out a solution in the next days.

No worries about the BitBench, it's a single purpose tool to inspect and showcase bit strings. Something like a hex editor, somewhat comparable to [Kaitai](http://kaitai.io/) (s.a. the discussion about alternatives in [BitBench's Hackaday article](https://hackaday.com/2018/11/06/bitbench-helps-parse-binary-data/)) but very limited to simple rows of data and simple formats to extract values. Basically enter some data rows, and write a "format" how to display the rows -- it's good for data that isn't aligned to nibbles and for data fields that are an odd width.

SensorsIot, 2019-03-14:
I used an SDR dongle on 2M because my URH on Linux does not offer HackRF. The playback was done with a HackRF.
I have to look a little more at BitBench as I got a few other TPMS sensors (without display). They all have different formats. One is somehow recognized by RTL_433 but as an Interlogics contact.
But first I concentrate on the external ones and I am glad you are willing to investigate into it. One other question: Do you know of a simple receiver which would decode such signals for an Arduino?

zuckschwerdt, 2019-03-14:
The samples are recorded at 2M? Then the timing isn't 50µs for a half-bit but 200µs. So far I've only seen 50µs and once 100µs with TPMS sensors.

I've you have the time to write a configuration for a dedicated recevier then I'd try a CC1101, it supports (G)FSK with synchronous frames and also asynchronous bit banging. Specifically I have the RF1100SE which has some library support but also isn't too hard to setup by hand.

SensorsIot, 2019-03-14:
I checked it again. The preamble 0 or one lasts 100 uS and a 01 sequence takes around 200 uS (measured with URH). 
Thank you for your suggestion concerning the CC1101. I also looked at it and was not sure. So I will try it.  Do you have any hints on how to get it in the "bit banging" mode and which library to use?

zuckschwerdt, 2019-03-14:
With a slight modification to the FSK demodulator (needs more testing before merge) I can reliably get the data from your samples.
This is where [BitBench shines](http://triq.net/bitbench?c=%5BFL%2008%2022%5D55ade732344864&c=%5BFL%2010%2022%5D55ade7304048f4&c=%5BFL%2010%2022%5D55ade7304048f4&c=%5BFL%2010%2022%5D55ade7304048f4&c=%5BFL%2010%2022%5D55ade7304048f4&c=%5BFL%2014%2022%5D55ade732544825&c=%5BFL%2014%2022%5D55ade732544825&c=%5BFL%2014%2022%5D55ade732544825&c=%5BFL%2014%2022%5D55ade732544825&c=%5BFL%2014%2022%5D55ade732544825&c=%5BFL%2014%2022%5D55ade732544825&c=%5BFL%2014%2022%5D55ade732544825&c=%5BFL%2014%2022%5D55ade732544825&c=%5BFL%2021%2023%5D55ade730814950&c=%5BFL%2021%2023%5D55ade730814950&c=%5BFL%2021%2023%5D55ade730814950&c=%5BFL%2021%2023%5D55ade730814950&c=%5BFL%2021%2023%5D55ade730814950&c=%5BFL%2021%2023%5D55ade730814950&c=%5BFL%2021%2023%5D55ade730814950&c=%5BFL%2021%2023%5D55ade730814950&c=%5BFL%2025%2023%5D55ade7329549f1&c=%5BFL%2025%2023%5D55ade7329549f1&c=%5BFL%2025%2023%5D55ade7329549f1&c=%5BFL%2025%2023%5D55ade7329549f1&c=%5BFL%2025%2023%5D55ade7329549f1&c=%5BFL%2025%2023%5D55ade7329549f1&c=%5BFL%2025%2023%5D55ade7329549f1&c=%5BFL%2025%2023%5D55ade7329549f1&c=%5BFL%2028%2023%5D55ade732a84992&c=%5BFL%2028%2023%5D55ade732a84992&c=%5BFL%2028%2023%5D55ade732a84992&c=%5BFL%2028%2023%5D55ade732a84992&c=%5BFL%2028%2023%5D55ade732a84992&c=%5BFL%2028%2023%5D55ade732a84992&c=%5BFL%2028%2023%5D55ade732a84992&c=%5BFL%2028%2023%5D55ade732a84992&c=%5BFL%2029%2023%5D55ade730af496b&c=%5BFL%2029%2023%5D55ade730af496b&c=%5BFL%2029%2023%5D55ade730af496b&c=%5BFL%2029%2023%5D55ade730af496b&c=%5BFL%2029%2023%5D55ade730af496b&c=%5BFL%2029%2023%5D55ade730af496b&c=%5BFL%2029%2023%5D55ade730af496b&c=%5BFL%2029%2023%5D55ade730af496b&c=%5BFR%2029%2022%5D29ff2b60ae4812&c=%5BFR%2029%2022%5D29ff2b60ae4812&c=%5BFR%2029%2022%5D29ff2b60ae4812&c=%5BFR%2029%2022%5D29ff2b60ae4812&c=%5BFR%2029%2023%5D29ff2b62ae494e&c=%5BFR%2029%2023%5D29ff2b62ae494e&c=%5BFR%2029%2023%5D29ff2b62ae494e&c=%5BFR%2029%2023%5D29ff2b62ae494e&c=%5BFR%2029%2023%5D29ff2b62ae494e&c=%5BFR%2029%2023%5D29ff2b62ae494e&c=%5BFR%2029%2023%5D29ff2b62ae494e&c=%5BFR%2029%2023%5D29ff2b62ae494e&c=%5BFR%2030%2022%5D29ff2b62b448b4&c=%5BFR%2030%2022%5D29ff2b62b448b4&c=%5BFR%2030%2022%5D29ff2b62b448b4&c=%5BFR%2030%2022%5D29ff2b62b448b4&c=%5BFR%2030%2022%5D29ff2b62b448b4&c=%5BFR%2030%2022%5D29ff2b62b448b4&c=%5BFR%2030%2022%5D29ff2b62b448b4&c=%5BFR%2030%2022%5D29ff2b62b448b4&c=%5BFR%2033%2023%5D29ff2b62c84937&c=%5BFR%2033%2023%5D29ff2b62c84937&c=%5BFR%2033%2023%5D29ff2b62c84937&c=%5BFR%2033%2023%5D29ff2b62c84937&c=%5BFR%2033%2023%5D29ff2b62c84937&c=%5BFR%2033%2023%5D29ff2b62c84937&c=%5BFR%2033%2023%5D29ff2b62c84937&c=%5BFR%2033%2023%5D29ff2b62c84937&c=%5BFR%2036%2023%5D29ff2b62d749d4&c=%5BFR%2036%2023%5D29ff2b62d749d4&c=%5BFR%2036%2023%5D29ff2b62d749d4&c=%5BFR%2036%2023%5D29ff2b62d749d4&c=%5BFR%2036%2023%5D29ff2b62d749d4&c=%5BFR%2036%2023%5D29ff2b62d749d4&c=%5BFR%2036%2023%5D29ff2b62d749d4&c=%5BFR%2036%2023%5D29ff2b62d749d4&c=%5BFR%2040%2023%5D29ff2b62ee4927&c=%5BFR%2040%2023%5D29ff2b62ee4927&c=%5BFR%2040%2023%5D29ff2b62ee4927&c=%5BFR%2040%2023%5D29ff2b62ee4927&c=%5BFR%2040%2023%5D29ff2b62ee4927&c=%5BFR%2040%2023%5D29ff2b62ee4927&c=%5BFR%2042%2023%5D29ff2b60f949ca&c=%5BFR%2042%2023%5D29ff2b60f949ca&c=%5BFR%2042%2023%5D29ff2b60f949ca&c=%5BFR%2042%2023%5D29ff2b60f949ca&c=%5BRL%2000%2020%5D158d0a940046aa&c=%5BRL%2000%2020%5D158d0a940046aa&c=%5BRL%2000%2020%5D158d0a940046aa&c=%5BRL%2000%2020%5D158d0a940046aa&c=%5BRL%2000%2020%5D158d0a940046aa&c=%5BRL%2011%2021%5D158d0a96454773&c=%5BRL%2011%2021%5D158d0a96454773&c=%5BRL%2011%2021%5D158d0a96454773&c=%5BRL%2011%2021%5D158d0a96454773&c=%5BRL%2012%2000%5D158d0a924b3231&c=%5BRL%2012%2000%5D158d0a924b3231&c=%5BRL%2012%2000%5D158d0a924b3231&c=%5BRL%2012%2000%5D158d0a924b3231&c=%5BRL%2012%2000%5D158d0a924b3231&c=%5BRL%2012%2000%5D158d0a924b3231&c=%5BRL%2012%2000%5D158d0a924b3231&c=%5BRL%2012%2000%5D158d0a924b3231&c=%5BRL%2013%2000%5D158d0a925132fe&c=%5BRL%2013%2000%5D158d0a925132fe&c=%5BRL%2013%2000%5D158d0a925132fe&c=%5BRL%2013%2000%5D158d0a925132fe&c=%5BRL%2013%2000%5D158d0a925132fe&c=%5BRL%2013%2000%5D158d0a925132fe&c=%5BRL%2013%2000%5D158d0a925132fe&c=%5BRL%2013%2000%5D158d0a925132fe&c=%5BRL%2014%2021%5D158d0a9458474d&c=%5BRL%2014%2021%5D158d0a9458474d&c=%5BRL%2014%2021%5D158d0a9458474d&c=%5BRL%2014%2021%5D158d0a9458474d&c=%5BRL%2014%2021%5D158d0a9458474d&c=%5BRL%2014%2021%5D158d0a9458474d&c=%5BRL%2014%2021%5D158d0a9458474d&c=%5BRL%2014%2021%5D158d0a9458474d&c=%5BRL%2015%2005%5D158d0a905d3739&c=%5BRL%2015%2005%5D158d0a905d3739&c=%5BRL%2015%2005%5D158d0a905d3739&c=%5BRL%2015%2005%5D158d0a905d3739&c=%5BRL%2015%2005%5D158d0a905d3739&c=%5BRL%2015%2005%5D158d0a905d3739&c=%5BRL%2015%2005%5D158d0a905d3739&c=%5BRL%2015%2005%5D158d0a905d3739&c=%5BRL%2015%2011%5D158d0a905d3dfb&c=%5BRL%2015%2011%5D158d0a905d3dfb&c=%5BRL%2015%2011%5D158d0a905d3dfb&c=%5BRL%2015%2011%5D158d0a905d3dfb&c=%5BRL%2015%2011%5D158d0a905d3dfb&c=%5BRL%2015%2011%5D158d0a905d3dfb&c=%5BRL%2015%2011%5D158d0a905d3dfb&c=%5BRL%2015%2011%5D158d0a905d3dfb&c=%5BRL%2016%2015%5D158d0a925f4136&c=%5BRL%2016%2015%5D158d0a925f4136&c=%5BRL%2016%2015%5D158d0a925f4136&c=%5BRL%2016%2015%5D158d0a925f4136&c=%5BRL%2016%2015%5D158d0a925f4136&c=%5BRL%2016%2015%5D158d0a925f4136&c=%5BRL%2016%2015%5D158d0a925f4136&c=%5BRL%2016%2015%5D158d0a925f4136&c=%5BRL%2017%2021%5D158d0a966647d7&c=%5BRL%2017%2021%5D158d0a966647d7&c=%5BRL%2017%2021%5D158d0a966647d7&c=%5BRL%2017%2021%5D158d0a9 [... cut ...] he decoding. (Press the "C" icon next to "Pad Left and Pad Right below." to see the annotations.
The Format is: `ID:28h S:4b P:8d T:8d CHK:8h`
- 28 bit ID
- 4 bit Status (deflation alarm, battery low etc)
- 8 bit Pressure (best guess quarter PSI, i.e. ~0.58 kPa)
- 8 bit Temperature (deg. C offset by 70)
- 8 bit Checksum
The preamble is `0xaa..aa9` (or `0x55..556` depending on the choosen polarity).

I'll test the changes to the demodulator more in depth, then add support for this sensor shortly.

zuckschwerdt, 2019-03-14:
The branch [feat-jansitetpms](https://github.com/merbanan/rtl_433/tree/feat-jansitetpms) should be able to decode the sensor at 2M sample rate. I still need to recover the checksum, could be an LFSR digest, not obviously a CRC though.
The hack to the FSK demodulator will take some time to get right, you could try a 1024k sample rate, but I'd say 250k will not work currently.

enavarro222, 2019-03-14:
This sensors looks similar to the one discus here https://github.com/merbanan/rtl_433/issues/856
So that's really great ! :)

I have to test your branch with my sensors ! I'll try to do this next week.

zuckschwerdt, 2019-03-14:
@enavarro222 your Zeepin C240 is Gaussian OOK if I remember correctly. The Jansite is Gaussian FSK -- very likely a different sensor. The FSK is be easier to fix for Gaussian filtering (softer edges -> narrow bandwidth). But we are still working on improvements to the demodulators (detectors really). And I'll see if I can apply the FSK ideas from here to the OOK part somehow.

SensorsIot, 2019-03-15:
Thank you for the branch. I have to finish my video and will continue with this project over the WE.

zuckschwerdt, 2019-03-15:
If the decoder works well, it'd be great if you can collect a bigger list of distinct codes. With about 1000 it should be possible to deduce the checksum and make the decoding more reliable. Even if it is a LFSR digest the key derivation can be worked out given enough codes. Basically work out the key for each of the 48 bits and then search for a generator key.

SensorsIot, 2019-03-16:
I created now around 500 different sets: https://www.dropbox.com/s/i87vd0szlbyhbl5/Checksums.xlsm?dl=0. The decoder worked good, I only had to filter a few completely wrong records.
Can you check if it fits your needs?

zuckschwerdt, 2019-03-16:
Thanks, about 300 looks plausible (sometimes the last bits seeme chopped off, i.e = 0).
Should be enough to make some first speculations ;)

zuckschwerdt, 2019-03-16:
Ok, it's not a simple sum (addition or xor aka LRC / BCC) of any width, and a CRC-8 doesn't fit. But also it doesn't seem a true secure digest (checksum difference for single bit change are only partially depended on the prefix, the lower nibble seems to be computed prefix-free). I've seen this before [#831](https://github.com/merbanan/rtl_433/pull/831#issuecomment-437492120) and still can figure out if some key is used on the upper nibble.

SensorsIot, 2019-03-19:
Do you need more samples? I should be able to create more of them.
Do you know this project? http://reveng.sourceforge.net/ ?
I tinkered now for one day (including a large part of the night) with the CC1101 but was not able to get a decent demodulated signal :-(  I used a library and started to change the initialization registers according to the recommendations of SmartRFstudio. I was not sure about the needed bandwidth as well as the frequency shift (is 433.92 the middle, the high, or the low frequency of the chip). Too many "moving parts" (registers to chose)...

zuckschwerdt, 2019-03-19:
If you can capture codes at a reasonable rate I can try to figure out the checksum still. But a few 10.000 samples would be needed. Just the codes in a plain file, either with duplicates or with a counter -- so I can get a confidence for the codes, as there is  a good chance that some may be wrong.
Basically I would look at all same single bit changes in a wide range of different codes.

The samples you posted show a center frequency of 433.927M with a deviation of +-10kHz, a spacing of 20kHz, i.e. lower and upper FSK are at 433.917M and 433.937M. I would pick a RX filter bandwidth of maybe 50kHz for that.
It's also important that the modulation is GFSK not just FSK. I need to see if I can replay your signals and test some settings on the CC1101 / RF1100se.

SensorsIot, 2019-03-19:
You find now quite a few new values in the same table. In "Tabelle2" you find all combinations without duplicates. In "Tablelle1" you find the new samples with numbers, I also colored the different sensors. One thing is important to know: I had all four sensors together in a pressure tank. So sometimes they sent at the same time and disturbed each-other.
Thank you for your information about the modulation. I see I can learn a lot from you. Where did you get this information from? On the spectrum analyzer as well as on the waterfall it was not so clear to me. 
Another question: Would it be possible to skype with you? My skype name is spiessa.

zuckschwerdt, 2019-03-19:
I "zoomed the frequency in" by reducing the bandwidth with a decimation of 8 using
```
sox -t raw -e floating-point -b 32 -L -c 2 -r 2000000 FL-08-22_433.92M_2000k.cf32 -t raw -e unsigned-integer -b 8 -L -c 2 -r 250000 FL-08-22_433.92M_250k.cu8
```
Then used crosshairs on the spectrogram. (`2000000` might actually be `2048000` if you used an RTL-SDR, also this assumes the center frequency was indeed 433.920M).

I'll ping you tomorrow about a chat.

zuckschwerdt, 2019-03-23:
The checksum is though. It's CRC-like but weaker. We have two sets of codes with the same prefix:

- `158d0a90623f 76` `158d0a906248 39` `158d0a906249 2c`
- `158d0a90703d b8` `158d0a90703e e7` `158d0a907049 a8`

The question now is how can we remove the last data byte and then get matching checkums? I.e.

- `op(0x3f, 0x76) == op(0x48, 0x39) == op(0x49, 0x2c)`
- `op(0x3d, 0xb8) == op(0x3e, 0xe7) == op(0x49, 0xa8)`
- What is `op()` ?

merbanan, 2019-03-23:
I looked at the source and found payloads like this:
158d0a90623f 60
158d0a90623f 76

This all looks bogus to me. If it is an integrity check 2 payloads can't result in different values. The only option left is that there is a counter that is not part of the payload. And then we have the payloads with lots of 0000. All this combined makes me not to trust the conclusion that the last byte is a crc.

zuckschwerdt, 2019-03-23:
Yes, some bitstring end early. I disregarded all codes with zeros at the end. Also in for the example above I only picked data with at least 100 repetitions from the linked execl sheet.

zuckschwerdt, 2019-03-23:
I have looked at  codes with only the last data bit changing and the xor-distance in the checksum looks very suspicous. But I can tell where the state comes from, could be some multiplication or folding operation, perhaps.
```
158d0a905840 de to 158d0a905841 cb xor is 15
158d0a906248 39 to 158d0a906249 2c xor is 15
158d0a907f3c 56 to 158d0a907f3d 43 xor is 15
2a0215c08d38 d8 to 2a0215c08d39 cd xor is 15
158d0a903f42 e5 to 158d0a903f43 d0 xor is 35
158d0a907a3c da to 158d0a907a3d ef xor is 35
29ff2b604d40 5b to 29ff2b604d41 6e xor is 35
2a0215c03c42 f2 to 2a0215c03c43 c7 xor is 35
55ade7302544 c5 to 55ade7302545 f0 xor is 35
2a0215c0753c 20 to 2a0215c0753d 55 xor is 75
55ade7300546 c3 to 55ade7300547 b6 xor is 75
29ff2b603542 0f to 29ff2b603543 fa xor is f5
```

merbanan, 2019-03-23:
It's not a linear CRC :/

zuckschwerdt, 2019-03-25:
It could be like a Fletcher-8 checksum. I.e. a sum above the nibbles with two different (4 bit wide) sums in the checksum byte. But the lower nibble would be an XOR and the upper nibble also doesn't look like a counter -- the bits seem to "fill up" rather than count up.
I'm putting this aside for the moment, it could be worth another try if we could observe every single of the last 8 data bits changing -- that could tell wether the "sum" is nibble or byte wide and if it's reflected.

SensorsIot, 2019-03-25:
There are two different information in the packets. One is transmitted once and the other    5 times or so. What is the difference between the two? Why o they transmit one only once and the other a few times?  AFAIK you decoded the first one. Am I right?

gcormier, 2019-03-25:
Silly question, but if you replay the packet you are receiving, and put in a random value for the last byte in question, does the receiver still accept it?

zuckschwerdt, 2019-03-26:
@gcormier it would be a way to map out all bits, yes. But for one, constructing a viable signal isn't that easy (I have [an example how to create a TPMS signal from just the payload data](https://github.com/zuckschwerdt/tx_tools/blob/master/examples/pmv107j.txt) -- but I didn't have time to test it in any way.)
And you would have to stare at the receiver while sending all those possible codes and decide which one made it. Happy to help if anyone wants to try such an adeventure :)

zuckschwerdt, 2019-03-26:
@SensorsIot To the best I could see a "transmission" is just two "packets" with some padding between. Both the first and second part looked identical. The decoder will pick up both packets in the [bitbuffer_search() loop](https://github.com/merbanan/rtl_433/blob/master/src/devices/tpms_jansite.c#L86).
Using `rtl_433 -R 123` you should get about 16 payloads in 8 "groups" -- 2 per group with the same timestamp.

RandomSporadicProjects, 2019-04-08:
Hi
I think this filter may be being falsely triggered by other TPMS signals
I get repeats like this

{"time":"2019-04-08 14:33:33","model":"Ford","type":"TPMS","id":"463382d9","code":"7ed846","mic":"CHECKSUM"}
{"time":"2019-04-08 14:33:33","model":"Jansite","type":"TPMS","id":"463382d","flags":13,"pressure_kPa":214.2,"temperature_C":166,"code":"463382d97ed846"}

{"time":"2019-04-08 14:11:11","model":"Citroen","type":"TPMS","state":"13","id":"984b64ab","flags":0,"repeat":2,"pressure_kPa":249.612,"temperature_C":18,"maybe_battery":127,"mic":"CHECKSUM"}
{"time":"2019-04-08 14:11:11","model":"Jansite","type":"TPMS","id":"13984b6","flags":6,"pressure_kPa":290.7,"temperature_C":-48,"code":"13984b64ab02b7"}

{"time":"2019-04-08 14:57:31","model":"Renault","type":"TPMS","id":"0526b2","flags":"36","pressure_kPa":189,"temperature_C":15,"mic":"CRC"}
{"time":"2019-04-08 14:11:11","model":"Jansite","type":"TPMS","id":"13984b6","flags":6,"pressure_kPa":290.7,"temperature_C":-48,"code":"13984b64ab02b7"}

There looks to be some overlap of the data in each repeat - my untrained eye cannot see that in the Renault data though
.
This picked up with an SDR then converted to JSON

Not a biggie for me, just letting you know (the only contribution I can make to your very useful project).

zuckschwerdt, 2019-04-08:
This decoder is default disabled. You shouldn't see this unless you request it. Perhaps you used `-G`? Don't do that, there is a reason we disable decoders ;)

The reason here is that the checksum isn't known. It's a custom algorithm and not obvious how it's computed.

RandomSporadicProjects, 2019-04-12:
That's fine - thanks.  Actually it's because I'm nosey and generally set my SDR to log all signals to a db.  I've been collecting TPMS data (and lots of other sensor data for home automation) for approx a year, so when I saw a new TPMS decoder I enabled it.

Thanks for all your work.

zuckschwerdt, 2019-06-23:
Closing as this is supported and "just" the checksum is missing. If somebody has insight on that checksum please reopen!

closed 2019-06-23 by zuckschwerdt

merbanan, 2020-01-29:
I had a hypothesis that the high nibble only affected the high nibble in the checksum but that doesn't hold.
```
 158d0a907949 e7
 158d0a906949 77

 29ff2b605c48 2f
 29ff2b603c48 8e

 29ff2b604348 80
 29ff2b608348 00
 29ff2b609348 e9

 29ff2b604a48 c7
 29ff2b608a48 1a

 29ff2b606f48 cb
 29ff2b609f48 86

 29ff2b608a49 2f
 29ff2b609a49 3b

 29ff2b609149 60
 29ff2b60b149 c4
```

merbanan, 2020-01-30:
Swapping 2 bytes give different checksum.
```
29ff2b604142 5e
29ff2b604241 05
```
So the checksum is position dependent.

mr-pl3r, 2020-07-20:
Hi guys @SensorsIot  @zuckschwerdt ,

I am new in all this radio technology, that's why I decided to follow some of the tutorials shown by Andreas on how to hack 433 Mhz devices and the usage of the rtl433, even reading everything related to TPMS. 
So far to start working on the TPMS, I decided to buy the exact same model Jansite TPMS  to make sure everything will match the video, then installed the rtl433 and pretty much everything is running as expected, but not with the TPMS sensors - for some reason they seem to be sending different information (data) when compared to yours. I have checked the 315Mhz band (considering the I am in the US), but still no success...I still can't find what I'm missing, when I run URH I cannot see the same graph shown in you video -related to this issue, but when I run the  command "rtl_433 -R 0 -X 'TPMS:FSK_PCM:50:50:1000' " I am able to see packages coming thru. Any suggestions or how to start troubleshooting will be appreciated.  Thank you in advance.

zuckschwerdt, 2020-07-21:
Grab a sample to verfify it's really 50µs timing: `rtl_433 -S all` then run `rtl_433 -w YOURFILE.ook YOURFILE.cu8` and drop on https://triq.org/pdv/ to view, maybe also post the ook here.
Then collect the codes and see if you can correlate them with the data you want.

mr-pl3r, 2020-07-24:
@zuckschwerdt 
I have completed a bunch of different readings, but I can't seem to match this format `ID:28h S:4b P:8d T(-FIFTY):8d CHK:8h`  or see the logic behind what I am reading on the bitbench.  As you can see attached to this msg is the `rtl_433 -A` file which shows multiple modulations. The one that I kept trying is  `rtl_433` -R 0 -X 'n=name,m=FSK_PWM,s=52,l=104,r=112'` based on the -A file, in addition I have included the ook files, which after plugging them in the "pulseview"
[rttl_433.zip](https://github.com/merbanan/rtl_433/files/4969680/rttl_433.zip)
 the only one file to produce a slicer was the file g007_433.92M_250k, the rest don't show any other information. 
BTW - The sensors are not the same model as the ones used by @SensorsIot, mine are a different model TY-568-US.

zuckschwerdt, 2020-07-24:
Note that there are gaps and pulses of 52 µs and 104 µs. That's not PWM, roughly PWN = fixed gaps (or fixed "pulse+gap") and data in pulse length. And conversly PPM = fixed pulses and data in gap length.
Here we have data in pulses and gaps: PCM, roughly one time unit gap = 0, one time unit pulse = 1.
And on top the data is very regular with only "base width" x1 and x2. That's Manchester encoded. So:
```
rtl_433 -R 0 -X 'n=TPMS,m=FSK_MC_ZEROBIT,s=50,l=50,r=1000'
```

mr-pl3r, 2020-07-26:
Based on your previous comment, I was able to pull the information from the sensors, based on this other comment you made, which uses the same type of modulation [https://github.com/merbanan/rtl_433/issues/856#issuecomment-450005471](url).

I ended up using the next command: `rtl_433 -R 0 -X 'n=TPMS,m=FSK_MC_ZEROBIT,s=50,l=50,r=1000,preamble=ffee'`, with that said I was able to pull now the ID, Pressure & Temp in the next bitbench [https://triq.net/bitbench?c=d01a70d0259ab8&c=d01a70d0259ab8&c=d01a7168840964&c=d01a7168840964&c=d01a7168840964&c=d01a7168840964&c=d01a7168840964&c=d01a7168840964&c=d01a7168840964&c=d01a7168840964&c=d01a7168840960&c=d01a7168840964&c=d01a7168840964&c=d01a7168840964&c=d01a7180867964&c=d01a7180867964&c=d01a7180867964&c=d01a7180867964&c=d01a71b086d960&c=d01a71b086d964&c=d01a71b086d964&c=d01a71b086d964&c=d01a725882a948&c=d01a725882a948&c=d01a725882a948&c=d01a725882a948&c=d01a7250869954&c=d01a7250869950&c=d01a7250869954&c=d01a7250869954&c=d01a7290841954&c=d01a7290841954&c=d01a7290841954&c=d01a7290841954&c=d01a72d885a950&c=d01a72d885a954&c=d01a72d885a954&c=d01a72d885a954&c=d01a7318852948&c=d01a7318852948&c=d01a7318852948&c=d01a7318852948&c=d01a7398822954&c=d01a7398822950&c=d01a7398822954&c=d01a7398822954&c=d01a7390861948&c=d01a7390861948&c=d01a7390861948&c=d01a7390861948&c=d01a7388864948&c=d01a7388864948&c=d01a7388864948&c=d01a7388864948&c=d01a738886494c&c=d01a7388864948&c=d01a738886494c&c=d01a738886494c&c=d01a7380827954&c=d01a7380827954&c=d01a7380827954&c=d01a7380827954&f=%20ID%3Ahhhh%20b%2C%20CH3d%2C%20PRESSURE%3A12d%2C%20TEMP%3Ad%2C%20CRC%3A8h%20%7C%208h%2016h%2016h%20](bitbench - Going Up).

And then I tried it again- but this time going down and with the sensor inside the fridge -this to capture the temperature info. [https://triq.net/bitbench?c=d01a12e8450c54&c=d01a12e8450c54&c=d01a12e8450c54&c=d01a12e8450c54&c=d01a1280407c48&c=d01a1280407c48&c=d01a1280407c48&c=d01a1280407c4c&c=d01a123047dc54&c=d01a123047dc54&c=d01a123047dc54&c=d01a123047dc50&c=d01a11e8430c7c&c=d01a11e8430c78&c=d01a11e8430c78&c=d01a11e8430c7c&c=d01a11a042bc78&c=d01a11a042bc78&c=d01a11a042bc78&c=d01a11a042bc78&c=d01a113845ec64&c=d01a113845ec64&c=d01a113845ec60&c=d01a113845ec60&c=d01a1100417c7c&c=d01a1100417c78&c=d01a1100417c78&c=d01a1100417c78&c=d01a10d841ac64&c=d01a10d841ac64&c=d01a10d841ac64&c=d01a10d841ac64&c=d01a10b044dc78&c=d01a10b044dc78&c=d01a10b044dc78&c=d01a10b044dc7c&c=d01a1088404c64&c=d01a1088404c64&c=d01a1088404c64&c=d01a1088404c64&c=d01a1050429c64&c=d01a1050429c64&c=d01a1050429c64&c=d01a1050429c64&c=d01a103847ec78&c=d01a103847ec78&c=d01a103847ec78&c=d01a103847ec78&c=d01a102047bc78&c=d01a102047bc7c&c=d01a102047bc78&c=d01a102047bc7c&c=d01a1010471c78&c=d01a1010471c78&c=d01a1010471c78&c=d01a1010471c78&c=d01a1008474c78&c=d01a1008474c78&c=d01a1008474c78&c=d01a1008474c78&c=d01a1000437c64&c=d01a1000437c60&c=d01a1000437c64&c=d01a1000437c64&f=%20ID%3Ahhhh%20b%2C%20CH3d%2C%20PRESSURE%3A12d%2C%20TEMP%3Ad%2C%20CRC%3A8h%20%7C%208h%2016h%2016h%20](bitbench - Going Down)

Based on the fact that I was able to read the data on the little screen that comes with the sensors, I was able to match the data with the following formula:
Pressure [PSI] = (bitbench pressure # *2/10) / 6.89
Temperature [Celsius] = (bitbench temp #) / 5.74

I will keep performing tests during the week to make sure this formula is stable in different conditions.

Thank you for your advice.

zuckschwerdt, 2020-07-27:
It is 1 psi = 6.89 kPa. So pressure is in kPa. Either in 0.2 (1/5) increments or in tenths (0.1) and you need to add a final bit.

Observed temperature " / 5.74" could indicate that you are missing the last bit: one bit more would give " / 10" which is very common.

The field marked "CRC" does not look like a good CRC or checksum, it is not random enough.

fusionstream, 2021-05-29:
Just thought I'd comment that neither 123 nor 180 work with a Hippcron Internal Sensor. The device looks identical to the Jansite so I'm not sure if it's just the same shell with a chip change or the internal one has a different protocol. Been trying in vain to decode the signals but with fairly congested roads it's difficult to pinpoint which is the exact signal. The most common decoded signals I get are an Abarth Spider as well as Hyundai VDO. I did also get an Interlogix signal a couple of times like Andreas.

fusionstream, 2021-05-31:
An update. This is the solar device I use (internal sensor option) https://www.aliexpress.com/item/4001119160861.html

Recording at 1024k sps and with option -R 123 shows the addresses of the sensors in the 'codes' albeit prefixed with 'a' but with 0kPa and weird temperature values (multiple). This suggests to me that they payload containing data does not really use the same protocol or something is off (since I have to specifically record at 1024k). Side note: 2M and 3Msps produces no usable data. Further compounding the matter is 2 of the 4 sensors are not correctly received by my receiver anymore (which is why I expedited this process)

zuckschwerdt, 2021-05-31:
When using `-A`, do you get a "pdv" link for viewing the raw data?

Otherwise get the raw data with `-X "n=mysolar,m=FSK_PCM,s=52,l=52,r=150"`

fusionstream, 2021-05-31:
I do have links. I also ran it with '-S all' during the capture and have 880 files so... haha. I'll try to muck around and see what I can find. This is nowhere near as easy as decoding my fan controls. :(

fusionstream, 2021-06-09:
Seemingly no usable data was captured beyond the addresses. The vendor should be sending me 2 replacements (for the 2 bad sensors) whereupon I will test it again. Additionally, I also will test it before installing it in the wheels which should make this infinitely easier.