merbanan/rtl_433 #1008: Ventus W132

df7nf, 2019-03-12:
Hello,

i use your Programm to record the Ventus W132 Speedsensor Data.
It works but can i change the time between recieving data. Actually the period is round about 30Seconds. But i will use it on an raspberry on my sailing boot, so i try to get faster informations about windspeed and direction - less then 30Seconds. 

How can i refresh data faster?

Thank you for your help

merbanan, 2019-03-12:
Not possible I would guess. There are limits on how much air time you can use on the ISM bands. Anyway this does not matter for rtl_433, it will decode as often as there is a signal transmitted. Ask the manufacturer of the device or seller.

closed 2019-03-12 by merbanan

df7nf, 2019-03-12:
okay thanks