merbanan/rtl_433 #1012: Honeywell 5890PI motion sensor

fanlessfan, 2019-03-19:
Hello,

I am wondering the Honeywell 5890PI motion sensor is supported or not. I can receive/decode Honeywell window/door sensor 5816. Here is the link to 5890. 

http://library.ademconet.com/MWT/fs2/5890PI/5890PI-Data-Sheet.PDF

Thx

merbanan, 2019-03-19:
Most likely, one of the protocol decoders for Honeywell mentions PIR sensors.

https://github.com/merbanan/rtl_433/blob/master/src/devices/honeywell_wdb.c

But as always no guarantees. Try and report back if it works or not.

closed 2019-03-19 by merbanan

fanlessfan, 2019-03-19:
Hi @merbanan ,

Thank you for your fantastic work. I will compile and try the latest.

pbendersky, 2019-04-07:
@fanlessfan are you able to decode those sensors? Do you know what frequency should I use?

zuckschwerdt, 2019-04-07:
see #1013