merbanan/rtl_433 #1014: Incorrect decoding of Nexa events?

lasseo, 2019-03-20:
I am a first time user of RTL-433. I wanted to sniff existing communication between Nexa equipments to be able to add this to Hass.io. However, unsure if I interpret data wrong or if "RTL-433" is giving me wrong info.

I have a Nexa dimmer unit, and Nexa wallbuttons to controll the dimmer (self-learning)
Sniffing traffic when turning light off with the Nexa buttons gave with RTL-433 gave the following output.

time      : 2019-03-19 22:19:16
model     : Proove       House Code: 67084961
Channel   : 3            State     : OFF           Unit      : 2             Group     : 1
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2019-03-19 22:19:16
model     : Nexa         House Code: 67084961
Group     : 1            Channel   : 3             State     : OFF           Unit      : 2
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _

However, this does not match with information received with TelldusCenter ( sniffing )
Remote Code / House code: 23902  Unit code: 2

Even TelldusCenter sniffing got some random values first, but then it stabilised on correct and working values.  

I feel there is something wrong in RTL-433 parsing, but it could equally be me not understanding the application good enough. Please reject issue if incorrect. I am happy to provide more "data" if needed.

zuckschwerdt, 2019-03-20:
`67084961` is `23902` inverted ("bit flipped"). Can't say which one is "right". Can you reference the code to something "official"?

lasseo, 2019-03-20:
What do you mean with official?

- Products are the following:
  - Nexa Transmitter,  http://www.nexa.se/vara-produkter/system-nexa/sandare/wt
  - Nexa Receiver, something like http://www.nexa.se/vara-produkter/system-nexa/mottagare-dimmers/cmr-101

  - TelldusCenter used, https://telldus.com/se/resurser/    ( Mac version )
     ( choosing Self-learning and then scan to get codes )

I have currently no official documentation on the "format" sent out from Nexa units. So currently trusting TelldusCenter for the scanning feature.  Will try to see if I can find any other scanning application, to see what that reports.

Using devices in Hassio with Tellstick, house code of 23902 works while 67084961 is not. Hence assuming that 23902 is correct value, if not Tellstick have an odd way of configuring devices with inverted house codes.

merbanan, 2019-03-20:
I think that a house code of 5 digits are more reasonable compared to 8. I am for to invert that value.

zuckschwerdt, 2019-03-20:
It would be strange if only some bits need inversion. Perhaps the Manchester decode is off. I'll take a closer look at that.

Mindavi, 2019-03-20:
My neighbors have these sensors too, I think. Does it help if I record some samples? I don't know exactly what's happening there but I see these events fired quickly after each other when I see them.

Mindavi, 2019-03-30:
I've got over 1400 samples now, but none of them is of the sensor I actually wanted to record :( gotta set up the system on another place with better reception

Mindavi, 2019-04-01:
It's not much, but I've got one sample file where the data is contained in. Somehow it's hard to get these samples, as they're only ever caught when another message is played right behind them (I saw some come by, but they never generated a file when I used `-S known`). There might be a bug in the code for those sensors, which maybe also explains the behavior seen here.

@zuckschwerdt do you have any idea how to capture this data? does it help to use the `-S all` flag? I didn't use it because I thought it was 'known' and `-S all` generates a lot of files, but if needed I can use it. However, if the decoder fails live I suspect it will also fail when run on the files, meaning I won't know which file to use (mm, maybe keeping the logs might help finding them).

[The sample](https://github.com/merbanan/rtl_433/files/3029314/g115_433.92M_250k.cu8.zip)

rtl_433 output for this sample is (as of version `18.12-175-g897f9db branch master at 201903290959`):

```
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _

time      : @0.320056s
model     : Proove
House Code: 50538725
Channel   : 3
State     : OFF
Unit      : 1
Group     : 1
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _

time      : @0.320056s
model     : Nexa
House Code: 50538725
Group     : 1
Channel   : 3
State     : OFF
Unit      : 1
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _

time      : @0.400244s
model     : Proove
House Code: 50538725
Channel   : 3
State     : OFF
Unit      : 1
Group     : 1
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _

time      : @0.400244s
model     : Nexa
House Code: 50538725
Group     : 1
Channel   : 3
State     : OFF
Unit      : 1
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _

time      : @0.480516s
model     : Proove
House Code: 50538725
Channel   : 3
State     : OFF
Unit      : 1
Group     : 1
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _

time      : @0.480516s
model     : Nexa
House Code: 50538725
Group     : 1
Channel   : 3
State     : OFF
Unit      : 1
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _

time      : @0.560708s
model     : Proove
House Code: 50538725
Channel   : 3
State     : OFF
Unit      : 1
Group     : 1
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _

time      : @0.560708s
model     : Nexa
House Code: 50538725
Group     : 1
Channel   : 3
State     : OFF
Unit      : 1
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _

time      : @0.640896s
Model     : Oregon Scientific SL109H
Id        : 0
Channel   : 3
Celsius   : 0.00 C
Humidity  : 0 %
Status    : 0
Integrity : CHECKSUM
```

zuckschwerdt, 2019-04-03:
Thanks for the PR. Does it work now, can this be closed?

lasseo, 2019-04-03:
Tested latest code, and I still see the same issue:

RTL_433 gives my the following:

> time      : 2019-04-03 23:49:49
> model     : Proove       House Code: 67088581
> Channel   : 3            State     : OFF           Unit      : 2             Group     : 1
> 20282
> 
> time      : 2019-04-03 23:49:49
> model     : Nexa         House Code: 67088581
> Channel   : 3            State     : OFF           Unit      : 2             Group     : 1

While TelldusCenter still gives me:  20282 as house/remote code.
20282 is the value that I e.g. uses in hass.io in tellstick configuration to get it to work.

So for me, the issue is still open.

merbanan, 2019-04-04:
@zuckschwerdt the inverted house code has been requested before. Maybe we should output both in this case. Or add a config option to select one or the other?

zuckschwerdt, 2019-04-04:
Sorry, I just wasn't following too closely. Inverting the id to me is no problem, but I keep wondering if perhaps everything is somehow inverted -- would be too strange otherwise.

I'll have another look today and PR the id change so everybody gets a fair warning :)

Mindavi, 2019-04-04:
> 
> 
> Thanks for the PR. Does it work now, can this be closed?

Nope, it was an unrelated issue with the decoder where it didn't report that it had found a signal. This meant that I couldn't record the signals and maybe help here.

Anyway, as the signals are from a system at the neighbors, I don't know which id is correct and I don't think my neighbors are going to tell me. I can only provide signals to look at, if needed.

Mindavi, 2019-04-04:
@kajksa and @LinuxChristian have authored the proove and nexa decoder, where kajksa made the first contribution and (I guess) Christian built upon that. It seems like kajksa isn't active anymore, but Christian might be able to shed some light on this discussion.

@LinuxChristian , have you ever had another source that helped you decoding the signal?

This might be an impactful change if people use the id to refer to the sensor, but as merbanan said, it might be reasonable to think the shorter house code is correct.

@lasseo , did you try to contact either Telldus or Nexa? They might not want to tell you anything, but I guess it's worth trying.

I've found some articles describing the protocol, but they're all unofficial. Some use the tellstick as a leading source.

- https://www.elektronikforumet.com/wiki/index.php?title=RF_Protokoll_-_JULA-Anslut
- https://gist.github.com/Mumakil/3b378060a0daab5b4c972faf4d47c818
- https://forum.jeelabs.net/node/87.html

Telldus has made a description of the protocol themselves, it seems. See:

- http://svn.telldus.com/svn/rf_ctrl/nexa_2_6_driver/trunk/NexaProtocol.txt

zuckschwerdt, 2019-04-04:
The "Telldus" protocol linked looks like EV1527 (12 bits ternary + stop bit). Since There is no convention which ternary states to use there is confusion. This now makes sense.

fyi: the original protocol calls
- short-short: `0` (pull to GND)
- short-long: `F` (floating)
- long-long: `1` (pull to Vcc)

But since OC outputs are common, most actually use `F` (open) as `0`, and `0` (pull-down) as `1` :)

zuckschwerdt, 2019-04-04:
There seems to be no way to distinguish Nexa, Proove, and KlikAanKlikUit. Packet length might be 64 or 72 bits (with last bits a dimmer value). But that's not per remote but 64 for off and 64 or 72 for on.

I guess I will merge all three into one of those and disable two (pending a removal).

merbanan, 2019-04-04:
The Nexus/Rubicson/Solight protocols could be merged also.

zuckschwerdt, 2019-04-04:
I have clean up Kaku, Nexa, and Proove with bbca524.
The change for the ID (and group flag) is in #1036

Mindavi, 2019-04-12:
@lasseo does this solve your problem?

lasseo, 2019-04-16:
Hi,
Tested latest source today, and works as I wanted. 
Thanks and hopefully we all agree that this is the right value to present. 

I can now use the RTL_433 application to listen devices and use these values directly in e.g. hass.io and tellstick configuration.

Cheers

Mindavi, 2019-06-03:
I guess this issue can be closed.

merbanan, 2019-06-03:
Lets assume so.

closed 2019-06-03 by merbanan