merbanan/rtl_433 #1021: time options, time_t and timeval

RussellSenior, 2019-03-25:
For processing convenience, it would be nice to have options for time_t (seconds since unix epoch) and timeval (the "hires" version of time_t). If I'm piping the data into something else, I don't want to have to parse complex time strings.

zuckschwerdt, 2019-03-25:
How to you parse the output, something like a bash script? Would `timestamp: 1553530376.123456` do?
Also the time value is a fixed character format of YYYY-MM-DD hh:mm:ss" (it should be ISO-8601, i.e. a "T" instead of the space really).

RussellSenior, 2019-03-25:
Personally, I just want time_t (the integer version). I'm parsing with awk and other typical unix tools from the -F csv output. But the analog of hires might be useful to others.

zuckschwerdt, 2019-03-26:
But awk can use the floating point variant and just use the part before the dot, right? I want to avoid adding two timestamp options if it can be avoided.

RussellSenior, 2019-03-26:
How about having an option to provide a format string to strftime(), which could default to "%Y-%m-%d %H:%M:%S" for current behavior, but which I could override with "%s" to give me what I want.  Adding the "hires" option would tack on the fractional seconds, without further modification. See GNU coreutils "date" utility for an example of such customization: $ date --help.

fanlessfan, 2019-04-06:
@zuckschwerdt It will be great, if we can do ISO-8601  YYYY-MM-DD**T**hh:mm:ss ("T" instead of the space really). When I integrated it to homeassisstant, ios app only recognize the "T" format of time. Currently, I have to convert it to "T" format manually.

thx

zuckschwerdt, 2019-04-06:
So, we want:
- none
- default `YYYY-MM-DD hh:mm:ss`
- default hires  `YYYY-MM-DD hh:mm:ss.uuuuuu`
- iso `YYYY-MM-DDThh:mm:ss`
- iso hires  `YYYY-MM-DDThh:mm:ss.uuuuuu`
- timestamp `s`
- timestamp hires `s.uuuuuu`
- reltime  `@s`

Additionally "utc".

Maybe a spec like this?
- `-M time:0`
- `-M time[:1][:hires][:utc]` (the `1` being default, i.e. `time:hires` will work)
- `-M time:iso[:hires][:utc]`
- `-M time:unix[:hires][:utc]`
- `-M time:rel`

zuckschwerdt, 2019-04-18:
Resolved with #1046

closed 2019-04-18 by zuckschwerdt