merbanan/rtl_433 #1022: Help with Flex decoder

gcormier, 2019-03-25:
Trying to hop on another TPMS brand. I let some air out to trigger some messages so I finally know I got something.

[image]

```
Detected FSK package	@0.342148s
Analyzing pulses...
Total count:   46,  width: 13.82 ms		( 3455 S)
Pulse width distribution:
 [ 0] count:    1,  width:    0 us [0;0]	(   0 S)
 [ 1] count:    2,  width:  492 us [492;492]	( 123 S)
 [ 2] count:   34,  width:  120 us [120;124]	(  30 S)
 [ 3] count:    8,  width:  240 us [240;244]	(  60 S)
 [ 4] count:    1,  width:   56 us [56;56]	(  14 S)
Gap width distribution:
 [ 0] count:   35,  width:  120 us [120;140]	(  30 S)
 [ 1] count:   10,  width:  236 us [196;248]	(  59 S)
Pulse period distribution:
 [ 0] count:    1,  width:  140 us [140;140]	(  35 S)
 [ 1] count:    2,  width:  612 us [612;612]	( 153 S)
 [ 2] count:   27,  width:  244 us [240;248]	(  61 S)
 [ 3] count:   12,  width:  360 us [320;368]	(  90 S)
 [ 4] count:    3,  width:  488 us [488;488]	( 122 S)
Level estimates [high, low]:  15871,     12
RSSI: -0.1 dB SNR: 30.9 dB Noise: -31.0 dB
Frequency offsets [F1, F2]:     360, -21528	(+1.4 kHz, -82.1 kHz)
Guessing modulation: No clue...
```

I've been trying this

` rtl_433 -r g010_315M_250k.cu8 -X "name=test,modulation=FSK_PCM,s=120,l=120,s=1000,r=10000"

The 120uS is close enough to the 100uS which I got from the datasheet. I'm 

Reset is some sort of time after which we reset the decoder I assume? 

I'm not really sure what SYNC represents nor what to put in that value.

gcormier, 2019-03-25:
I did get it working, but I still don't understand what SYNC represents.

`rtl_433 -X "name=test,modulation=FSK_PCM,s=120,l=120,s=120,r=9000" -r g167_315M_250k.cu8`

zuckschwerdt, 2019-03-26:
`sync` is a third optional width to detect sync-pulses. The full range would be:
- short width
- long width
- sync width
I guess those two 492 µs pulses could be sync. You need to confirm that in Pulseview.

`reset` is the maxiumum gap after which decoding is aborted. I should be slightly longer that the longest expected gap. 300 µs in this example.

gcormier, 2019-03-26:
Those 2 pulses are indeed 492.
[image]

However, it gives me no results
`-X "name=test,modulation=FSK_PCM,s=120,l=120,s=192,r=600"`

Omitting it all together gives me nice results though, comparable to before minus the random 0 at the end

`rtl_433 -r g025_315M_250k.cu8 -X "name=test,modulation=FSK_PCM,s=120,l=120,r=300"`

```
time      : @0.341740s
model     : test         count     : 1             num_rows  : 1             rows      : 
len       : 112          data      : 7aaaaaaaf4d4b34ad2b2d4ad54ac
codes     : {112}7aaaaaaaf4d4b34ad2b2d4ad54ac
```

zuckschwerdt, 2019-03-27:
Strange. PCM currently is missing the implementation for sync thus both should work the same.
I'd say adding a preamble match to clean the output up is all that's needed. `,preamble=aaaf` or similar depending on the bit alignment needed. (aaaf = aaab = 5557 = 5552, see [this BitBench](http://triq.net/bitbench?c=aaaf&f=8b%0A8h) and try the shift button `>>`

gcormier, 2019-04-04:
Thanks again for the help!

closed 2019-04-04 by gcormier