merbanan/rtl_433 #1024: New Device - Nissan Leaf TPMS

gcormier, 2019-03-26:
Test data
https://github.com/merbanan/rtl_433_tests/pull/269

Figured I would bring the discussion here.

Ignore the PR notes, as I realize now I just spend the evening decoding the data stream and not the actual decoded output :(

gcormier, 2019-03-27:
@zuckschwerdt Over my head a bit.

Is this output string the raw string, or the manchester decoded value?

```
rtl_433 -r g025_315M_250k.cu8 -X "name=test,modulation=FSK_PCM,s=120,l=120,r=300,preamble=aaaf"

time      : @0.341740s
model     : test         count     : 1             num_rows  : 1             rows      : 
len       : 76           data      : 4d4b34ad2b2d4ad54ac
codes     : {76}4d4b34ad2b2d4ad54ac
```

Before I go to town on BitBench, I want to know I'm using the right value. I'm hoping to check every other TPMS format to see if it matches any of those.

zuckschwerdt, 2019-03-28:
Sorry, no this isn't manchester decoded. You would need to apply that by hand. I'll add a few example scripts and maybe a button for BitBench soon.
There currently is no MC decode for FSK as you would need to align the data at a suitable preamble. The trick for subsequent manchester is to make sure all groups of two bits after the preamble are either `01` or `10`. Also the resulting hex code could be shifted 1-3 bits if the preamble to cut off was choosen to short or long.
Sorry again, I'll try to make MC utils available asap.

gcormier, 2019-03-28:
Sounds good, thanks!

I guess using the tx_tools generator with a suspected preamble would be a
nice way to graphically visualize the preamble to make sure it ends at the
right spot.

klohner, 2019-03-30:
I tried playing around with this data a bit.  After removing a sync/prefix and Differential Manchester decoding the data, I took a stab at trying to identify the fields.  I wasn't able to come up with much, but maybe this is progress? 

There seems to be a parity bit in there somewhere, and I'd guess it's the final bit.  There's a few other flags that may mean something, and there seems to be an 8-bit value that may be related to pressure, but maybe not?  Also, I'll assume these samples all come from the same device since bits that probably relate to a TPMS_ID don't change at all.

[BitBench analysis](http://triq.net/bitbench?c=%5B-X%20%22name%3Dtest%2Cmodulation%3DFSK_PCM%2Cs%3D120%2Cl%3D120%2Cs%3D120%2Cr%3D9000%22%5D&c=%5B%221111010101010101010101010101010111%22%20sync%20data%20prefix%20removed%2C%20Zero-Transition%20Differential%20Manchester%20Decoded%5D&c=&c=0y000000101110100101001101001011110000110%20%5B7aaaaaaaf554b34ad2b2d4b32ab2%5D%20%5BThis%20was%20captured%20days%20before%20without%20knowing%20I%20had%20it.%20Represents%20the%20highest%20pressure%20before%20any%20deflations%5D&c=&c=0y011000101110100101001101001011010110001%20%5B7aaaaaaaf354b34ad2b2d4b2d354%5D%20%5BDeflate%5D%20%5BTemp%20was%2025c%5D%20%20%20%20%20%20%20%20%20%20%20%20%20%20%20%20%20%20%20%20%20%20%20%20%20%20%20%20%20%20%20%20%20%20%20%20%20%20%20%20%20%20%20%20%20%20%20%20%20%20%20%20%20%20%20%20%20%20%20%20%20%20%20%20%20%20%20%20%20%20%20%20%20%20%20%20%20%20%20%20%20%20%20%20%20%20%20&c=0y011000101110100101001101001011010100110%20%5B7aaaaaaaf354b34ad2b2d4b2d2b2%5D%20%5BDeflate%5D&c=0y011000101110100101001101001011011111010%20%5B7aaaaaaaf354b34ad2b2d4b2ccd2%5D%20%5BDeflate%5D&c=0y011000101110100101001101001011001001000%20%5B7aaaaaaaf354b34ad2b2d4b2b52a%5D%20%5BDeflate%5D&c=0y011000101110100101001101001011001110100%20%5B7aaaaaaaf354b34ad2b2d4b2b34a%5D%20%5BDeflate%5D&c=0y011000101110100101001101001011001100101%20%5B7aaaaaaaf354b34ad2b2d4b2b2b4%5D%20%5BDeflate%5D&c=0y011000101110100101001101001011000111001%20%5B7aaaaaaaf354b34ad2b2d4b2acd4%5D%20%5BDeflate%5D%20%5BTemp%20was%2025c%5D&c=&c=0y001100101110100101001101001011000101101%20%5B7aaaaaaaf4d4b34ad2b2d4b2ad34%5D%20%5BReport%5D%20%5BTemp%20was%2025c%5D&c=0y001100101110100101001101001011000101101%20%5B7aaaaaaaf4d4b34ad2b2d4b2ad34%5D%20%5BReport%5D&c=0y001100101110100101001101001011000100010%20%5B7aaaaaaaf4d4b34ad2b2d4b2ad52%5D%20%5BReport%5D&c=0y001100101110100101001101001011000100010%20%5B7aaaaaaaf4d4b34ad2b2d4b2ad52%5D%20%5BReport%5D&c=0y001100101110100101001101001011000100010%20%5B7aaaaaaaf4d4b34ad2b2d4b2ad52%5D%20%5BReport%5D%20%5BTemp%20was%2025c%5D&c=&c=%5BRound%202%5D&c=0y011000101110100101001101001011000001100%20%5B7aaaaaaaf354b34ad2b2d4b2aaca%5D%20%5BDeflate%5D&c=&c=0y001100101110100101001101001001000011100%20%5B7aaaaaaaf4d4b34ad2b2d4ad54ca%5D%20%5Bg004%20-%20Just%20before%20cooling%5D&c=0y001100101110100101001101001001000011100%20%5B7aaaaaaaf4d4b34ad2b2d4ad54ca%5D%20%5Bg007%5D&c=0y001100101110100101001101001001000001101%20%5B7aaaaaaaf4d4b34ad2b2d4ad5534%5D%20%5Bg008%5D&c=0y001100101110100101001101001001000001101%20%5B7aaaaaaaf4d4b34ad2b2d4ad5534%5D%20%5Bg010%20coldest%3F%5D&c=&c=0y001100101110100101001101001001000001101%20%5B7aaaaaaaf4d4b34ad2b2d4ad5534%5D%20%5BFairly%20stable%5D&c=0y001100101110100101001101001001000010011%20%5B7aaaaaaaf4d4b34ad2b2d4ad54ac%5D%20%5Bg017%20-%20let%20a%20little%20air%20out%20just%20before%5D&c=0y001100101110100101001101001001000110001%20%5B7aaaaaaaf4d4b34ad2b2d4ad5354%5D%20%5Bg023%20let%20a%20bit%20more%20air%20just%20before%2C%2023c%3F%5D&c=&c=0y001100101110100101001101001001000010011%20%5B7aaaaaaaf4d4b34ad2b2d4ad54ac%5D%20%5Bg052%20-%3E%20Let%20out%20air%5D&c=0y001100101110100101001101001001000110001%20%5B7aaaaaaaf4d4b34ad2b2d4ad5354%5D%20%5Bg054%20%3D%20result.%5D&c=0y001100101110100101001101001001000110001%20%5B7aaaaaaaf4d4b34ad2b2d4ad5354%5D%20%5Bg067%20-%3E%20Let%20out%20TINY%20amount%20of%20air%5D&c=0y001100101110100101001101001001000111000%20%5B7aaaaaaaf4d4b34ad2b2d4ad532a%5D%20%5Bg069%20%3D%20result%5D&c=&c=0y001100101110100101001101001001000100110%20%5B7aaaaaaaf4d4b34ad2b2d4ad52b2%5D%20%5Bg083%20-%20Let%20a%20bunch%20of%20air%20out%20to%20reset%20the%20panic%20mode.%20Tire%20on%20balcony%20to%20cool.%5D&c=0y001100101110100101001101001001011000001%20%5B7aaaaaaaf4d4b34ad2b2d4ad3554%5D%20%5Bg085%5D&c=0y001100101110100101001101001001011000001%20%5B7aaaaaaaf4d4b34ad2b2d4ad3554%5D%20%5Bg086%20Pressure%20stabilized.%20Time%20for%20cooling.%5D&c=0y001100101110100101001101001001011001000%20%5B7aaaaaaaf4d4b34ad2b2d4ad352a%5D%20%5Bg093%5D&c=0y001100101110100101001101001001011011111%20%5B7aaaaaaaf4d4b34ad2b2d4ad34cc%5D%20%5Bg116%20around%209%20degrees%20outside%20rubber%5D&c=0y001100101110100101001101001001011010110%20%5B7aaaaaaaf4d4b34ad2b2d4ad34b2%5D%20%5Bg123%5D&c=0y001100101110100101001101001001011110100%20%5B7aaaaaaaf4d4b34ed2b2d4ad334a%5D%20%5Bg150%20-%20error%20decoding%2C%20should%20be%207aaaaaaaf4d4b34Ad2b2d4ad334a%5D&c=0y001100101110100101001101001001011111011%20%5B7aaaaaaaf4d4b34ad2b2d4ad332c%5D%20%5Bg167%5D&c=0y001100101110100101001101001001011101010%20%5B7aaaaaaaf4d4b34ad2b2d4ad32d2%5D%20%5Bg201%20Rubber%200%20degrees%5D&f=ZERO%3A1b%20DEFLATING%3A1b%20MODE%3A2b%20TPMS_ID%3F%3A24h%20FLAG%3A1b%20ONE%3A1b%20VALUE%3A8d%20PARITY%3A1b)

zuckschwerdt, 2019-03-30:
I would expect 3-4 bytes of ID, 1-2 bytes pressure, 1 byte temp, and 1 byte checksum. Haven't really seen shorter codes. I'll take a closer look this evening.

gcormier, 2019-03-30:
I did some quick googling to see if there was any Leaf tool that would spit out the TPMS ID's. There's an Android app (Leaf Spy), but even the Pro version doesn't report the ID's.

klohner, 2019-03-30:
@gcormier Would it be possible to get samples from other tires, and have a manual tire pressure reading associated with each sample?

zuckschwerdt, 2019-03-30:
@gcormier Just a crude first test, but BitBench now has Preamble/Sync options and Manchester decode also.

gcormier, 2019-03-30:
Yes, I will try to get readings for PSI as well as something from the other 3 wheels. They don't belong to me, so I'll see when I can get over to my friends to read them :)

zuckschwerdt, 2019-03-31:
This [BitBench has the data](http://triq.net/bitbench?c=%5B-X%20%22name%3Dtest%2Cmodulation%3DFSK_PCM%2Cs%3D120%2Cl%3D120%2Cs%3D120%2Cr%3D9000%22%5D&c=%5B%221111010101010101010101010101010111%22%20sync%20data%20prefix%20removed%2C%20Zero-Transition%20Differential%20Manchester%20Decoded%5D&c=&c=7aaaaaaaf554b34ad2b2d4b32ab2%20%5BThis%20was%20captured%20days%20before%20without%20knowing%20I%20had%20it.%20Represents%20the%20highest%20pressure%20before%20any%20deflations%5D&c=&c=7aaaaaaaf354b34ad2b2d4b2d354%20%5BDeflate%5D%20%5BTemp%20was%2025c%5D&c=7aaaaaaaf354b34ad2b2d4b2d2b2%20%5BDeflate%5D&c=7aaaaaaaf354b34ad2b2d4b2ccd2%20%5BDeflate%5D&c=7aaaaaaaf354b34ad2b2d4b2b52a%20%5BDeflate%5D&c=7aaaaaaaf354b34ad2b2d4b2b34a%20%5BDeflate%5D&c=7aaaaaaaf354b34ad2b2d4b2b2b4%20%5BDeflate%5D&c=7aaaaaaaf354b34ad2b2d4b2acd4%20%5BDeflate%5D%20%5BTemp%20was%2025c%5D&c=&c=7aaaaaaaf4d4b34ad2b2d4b2ad34%20%5BReport%5D%20%5BTemp%20was%2025c%5D&c=7aaaaaaaf4d4b34ad2b2d4b2ad34%20%5BReport%5D&c=7aaaaaaaf4d4b34ad2b2d4b2ad52%20%5BReport%5D&c=7aaaaaaaf4d4b34ad2b2d4b2ad52%20%5BReport%5D&c=7aaaaaaaf4d4b34ad2b2d4b2ad52%20%5BReport%5D%20%5BTemp%20was%2025c%5D&c=&c=%5BRound%202%5D&c=7aaaaaaaf354b34ad2b2d4b2aaca%20%5BDeflate%5D&c=&c=7aaaaaaaf4d4b34ad2b2d4ad54ca%20%5Bg004%20-%20Just%20before%20cooling%5D&c=7aaaaaaaf4d4b34ad2b2d4ad54ca%20%5Bg007%5D&c=7aaaaaaaf4d4b34ad2b2d4ad5534%20%5Bg008%5D&c=7aaaaaaaf4d4b34ad2b2d4ad5534%20%5Bg010%20coldest%3F%5D&c=&c=7aaaaaaaf4d4b34ad2b2d4ad5534%20%5BFairly%20stable%5D&c=7aaaaaaaf4d4b34ad2b2d4ad54ac%20%5Bg017%20-%20let%20a%20little%20air%20out%20just%20before%5D&c=7aaaaaaaf4d4b34ad2b2d4ad5354%20%5Bg023%20let%20a%20bit%20more%20air%20just%20before%2C%2023c%3F%5D&c=&c=7aaaaaaaf4d4b34ad2b2d4ad54ac%20%5Bg052%20-%3E%20Let%20out%20air%5D&c=7aaaaaaaf4d4b34ad2b2d4ad5354%20%5Bg054%20%3D%20result.%5D&c=7aaaaaaaf4d4b34ad2b2d4ad5354%20%5Bg067%20-%3E%20Let%20out%20TINY%20amount%20of%20air%5D&c=7aaaaaaaf4d4b34ad2b2d4ad532a%20%5Bg069%20%3D%20result%5D&c=&c=7aaaaaaaf4d4b34ad2b2d4ad52b2%20%5Bg083%20-%20Let%20a%20bunch%20of%20air%20out%20to%20reset%20the%20panic%20mode.%20Tire%20on%20balcony%20to%20cool.%5D&c=7aaaaaaaf4d4b34ad2b2d4ad3554%20%5Bg085%5D&c=7aaaaaaaf4d4b34ad2b2d4ad3554%20%5Bg086%20Pressure%20stabilized.%20Time%20for%20cooling.%5D&c=7aaaaaaaf4d4b34ad2b2d4ad352a%20%5Bg093%5D&c=7aaaaaaaf4d4b34ad2b2d4ad34cc%20%5Bg116%20around%209%20degrees%20outside%20rubber%5D&c=7aaaaaaaf4d4b34ad2b2d4ad34b2%20%5Bg123%5D&c=7aaaaaaaf4d4b34ed2b2d4ad334a%20%5Bg150%20-%20error%20decoding%2C%20should%20be%207aaaaaaaf4d4b34Ad2b2d4ad334a%5D&c=7aaaaaaaf4d4b34ad2b2d4ad332c%20%5Bg167%5D&c=7aaaaaaaf4d4b34ad2b2d4ad32d2%20%5Bg201%20Rubber%200%20degrees%5D&f=ZERO%3A1b%20DEFLATING%3A1b%20MODE%3A2b%20TPMS_ID%3F%3A24h%20FLAG%3A1b%20ONE%3A1b%20VALUE%3A8d%20PARITY%3A1b&a=Preamble&m=aaaf&d=DMC) with Preamble alignment and DMC coding. The didn't check/update the format  string thoug.

gcormier, 2019-03-31:
Some of my research notes (I think would support the MC theory)

Claims to be OEM : https://www.ebay.com/p/Nissan-407003AN1B-Genuine-OEM-TPMS-Sensor/1437046258?iid=132576949062

https://industrycanada.co/7812D-S180015B
https://fccid.io/KR5S180052015B

Remark: *7 WUP definition: - 
Modulation ASK for the WUP
 - 12 blocks in the WUP
 - 2 bytes à "00" Manchester, 3 bytes NRZ in each blocks 
WUP length = 10.0993536ms MAX 

Also, I have a note that there is some unfinished work done on a sensor by the same company, whether that helps or not : https://github.com/merbanan/rtl_433_tests/compare/master...FizzyWhiz:master for the KR5S180052056.

gcormier, 2019-03-31:
Interesting though - the picture with the FCC ID clearly has FSK written.

[image]

Perhaps I am misinterpreting that Remark *7. Maybe it is not specific to this device.

klohner, 2019-03-31:
Agreed that MC looks better than DMC for this.  I was able to parse out 8 bits that may relate to pressure, and using the same decoding for a PSI value that the Toyota TPMS sensor uses ((PSI+7)*4), I'm getting sane values.  

Here's another stab at decoding: [BitBench](http://triq.net/bitbench?c=%5B-X%20%22name%3Dtest%2Cmodulation%3DFSK_PCM%2Cs%3D120%2Cl%3D120%2Cs%3D120%2Cr%3D9000%22%5D&c=%5B0xf5555555e%20preamble%20removed%2C%20Manchester%20(G.E.%20Thomas)%20Decoded%2C%2037%20bits%20data%5D&c=&c=7aaaaaaaf554b34ad2b2d4b32ab2%20%5BThis%20was%20captured%20days%20before%20without%20knowing%20I%20had%20it.%20Represents%20the%20highest%20pressure%20before%20any%20deflations%5D&c=&c=7aaaaaaaf354b34ad2b2d4b2d354%20%5BDeflate%5D%20%5BTemp%20was%2025c%5D&c=7aaaaaaaf354b34ad2b2d4b2d2b2%20%5BDeflate%5D&c=7aaaaaaaf354b34ad2b2d4b2ccd2%20%5BDeflate%5D&c=7aaaaaaaf354b34ad2b2d4b2b52a%20%5BDeflate%5D&c=7aaaaaaaf354b34ad2b2d4b2b34a%20%5BDeflate%5D&c=7aaaaaaaf354b34ad2b2d4b2b2b4%20%5BDeflate%5D&c=7aaaaaaaf354b34ad2b2d4b2acd4%20%5BDeflate%5D%20%5BTemp%20was%2025c%5D&c=&c=7aaaaaaaf4d4b34ad2b2d4b2ad34%20%5BReport%5D%20%5BTemp%20was%2025c%5D&c=7aaaaaaaf4d4b34ad2b2d4b2ad34%20%5BReport%5D&c=7aaaaaaaf4d4b34ad2b2d4b2ad52%20%5BReport%5D&c=7aaaaaaaf4d4b34ad2b2d4b2ad52%20%5BReport%5D&c=7aaaaaaaf4d4b34ad2b2d4b2ad52%20%5BReport%5D%20%5BTemp%20was%2025c%5D&c=&c=%5BRound%202%5D&c=7aaaaaaaf354b34ad2b2d4b2aaca%20%5BDeflate%5D&c=&c=7aaaaaaaf4d4b34ad2b2d4ad54ca%20%5Bg004%20-%20Just%20before%20cooling%5D&c=7aaaaaaaf4d4b34ad2b2d4ad54ca%20%5Bg007%5D&c=7aaaaaaaf4d4b34ad2b2d4ad5534%20%5Bg008%5D&c=7aaaaaaaf4d4b34ad2b2d4ad5534%20%5Bg010%20coldest%3F%5D&c=&c=7aaaaaaaf4d4b34ad2b2d4ad5534%20%5BFairly%20stable%5D&c=7aaaaaaaf4d4b34ad2b2d4ad54ac%20%5Bg017%20-%20let%20a%20little%20air%20out%20just%20before%5D&c=7aaaaaaaf4d4b34ad2b2d4ad5354%20%5Bg023%20let%20a%20bit%20more%20air%20just%20before%2C%2023c%3F%5D&c=&c=7aaaaaaaf4d4b34ad2b2d4ad54ac%20%5Bg052%20-%3E%20Let%20out%20air%5D&c=7aaaaaaaf4d4b34ad2b2d4ad5354%20%5Bg054%20%3D%20result.%5D&c=7aaaaaaaf4d4b34ad2b2d4ad5354%20%5Bg067%20-%3E%20Let%20out%20TINY%20amount%20of%20air%5D&c=7aaaaaaaf4d4b34ad2b2d4ad532a%20%5Bg069%20%3D%20result%5D&c=&c=7aaaaaaaf4d4b34ad2b2d4ad52b2%20%5Bg083%20-%20Let%20a%20bunch%20of%20air%20out%20to%20reset%20the%20panic%20mode.%20Tire%20on%20balcony%20to%20cool.%5D&c=7aaaaaaaf4d4b34ad2b2d4ad3554%20%5Bg085%5D&c=7aaaaaaaf4d4b34ad2b2d4ad3554%20%5Bg086%20Pressure%20stabilized.%20Time%20for%20cooling.%5D&c=7aaaaaaaf4d4b34ad2b2d4ad352a%20%5Bg093%5D&c=7aaaaaaaf4d4b34ad2b2d4ad34cc%20%5Bg116%20around%209%20degrees%20outside%20rubber%5D&c=7aaaaaaaf4d4b34ad2b2d4ad34b2%20%5Bg123%5D&c=7aaaaaaaf4d4b34Ad2b2d4ad334a%20%5Bg150%20-%20manually%20fixed%20bit%20in%20decoding%5D&c=7aaaaaaaf4d4b34ad2b2d4ad332c%20%5Bg167%5D&c=7aaaaaaaf4d4b34ad2b2d4ad32d2%20%5Bg201%20Rubber%200%20degrees%5D&f=MODE%3A2d%20UNKNOWN%3A1b%20TPMS_ID%3A24h%20(PSI%2BSEVEN)*FOUR%3D8d%20UNKNOWN%3A2b&a=Preamble&m=f5555555e&d=MC)

gcormier, 2019-03-31:
Nice! I am heading over to my buddies tomorrow evening to get some captures from the other 3 wheels! I'll make a note of the PSI as well and ambient temp.

zuckschwerdt, 2019-04-01:
To get robust sync the preamble should be shorter  so some starting bits a re allowed to be missing, 16 bit seems enough `555e`.

gcormier, 2019-04-02:
Got the data. Might take a day or two to update the tests and PR, but here's one string from each tire. I have the PSI's all written down too, i just have to crossreference.

```
7aaaaaaaf4d4b34ad2b2d4ad54ca0
7aaaaaaaf4d52aab2d2b54ad4b34
7aaaaaaaf4cd4acccaab54ad4b4c
7aaaaaaaf4cd34ab2ccb54acd4d4
```

gcormier, 2019-04-02:
```

Tire 2 at 24psi
7aaaaaaaf4d52aab2d2b54ad34b4

Tire 3 at 22psi
7aaaaaaaf4cd4acccaab54ad2b520

Tire 4 at 20psi
7aaaaaaaf4cd34ab2ccb54acd4d4
```

klohner, 2019-04-02:
@gcormier Great, thanks!  I revised the BitBench a bit to include these new data points and reset the preamble value to what @zuckschwerdt suggested.  The TPMS_ID might be only 5 bytes long since the 6th byte doesn't seem to vary at all between all 4 tires.  Also, changing the formula for the pressure value to be interpreted as ((PSI+3)*4) seems to line up well with the manual readings you took.

[BitBench](http://triq.net/bitbench?c=%5B-X%20%22name%3Dtest%2Cmodulation%3DFSK_PCM%2Cs%3D120%2Cl%3D120%2Cs%3D120%2Cr%3D9000%22%5D&c=%5B0xf5555555e%20preamble%20removed%2C%20Manchester%20(G.E.%20Thomas)%20Decoded%2C%2037%20bits%20data%5D&c=&c=7aaaaaaaf554b34ad2b2d4b32ab2%20%5BThis%20was%20captured%20days%20before%20without%20knowing%20I%20had%20it.%20Represents%20the%20highest%20pressure%20before%20any%20deflations%5D&c=&c=7aaaaaaaf354b34ad2b2d4b2d354%20%5BDeflate%5D%20%5BTemp%20was%2025c%5D&c=7aaaaaaaf354b34ad2b2d4b2d2b2%20%5BDeflate%5D&c=7aaaaaaaf354b34ad2b2d4b2ccd2%20%5BDeflate%5D&c=7aaaaaaaf354b34ad2b2d4b2b52a%20%5BDeflate%5D&c=7aaaaaaaf354b34ad2b2d4b2b34a%20%5BDeflate%5D&c=7aaaaaaaf354b34ad2b2d4b2b2b4%20%5BDeflate%5D&c=7aaaaaaaf354b34ad2b2d4b2acd4%20%5BDeflate%5D%20%5BTemp%20was%2025c%5D&c=&c=7aaaaaaaf4d4b34ad2b2d4b2ad34%20%5BReport%5D%20%5BTemp%20was%2025c%5D&c=7aaaaaaaf4d4b34ad2b2d4b2ad34%20%5BReport%5D&c=7aaaaaaaf4d4b34ad2b2d4b2ad52%20%5BReport%5D&c=7aaaaaaaf4d4b34ad2b2d4b2ad52%20%5BReport%5D&c=7aaaaaaaf4d4b34ad2b2d4b2ad52%20%5BReport%5D%20%5BTemp%20was%2025c%5D&c=&c=%5BRound%202%5D&c=7aaaaaaaf354b34ad2b2d4b2aaca%20%5BDeflate%5D&c=&c=7aaaaaaaf4d4b34ad2b2d4ad54ca%20%5Bg004%20-%20Just%20before%20cooling%5D&c=7aaaaaaaf4d4b34ad2b2d4ad54ca%20%5Bg007%5D&c=7aaaaaaaf4d4b34ad2b2d4ad5534%20%5Bg008%5D&c=7aaaaaaaf4d4b34ad2b2d4ad5534%20%5Bg010%20coldest%3F%5D&c=&c=7aaaaaaaf4d4b34ad2b2d4ad5534%20%5BFairly%20stable%5D&c=7aaaaaaaf4d4b34ad2b2d4ad54ac%20%5Bg017%20-%20let%20a%20little%20air%20out%20just%20before%5D&c=7aaaaaaaf4d4b34ad2b2d4ad5354%20%5Bg023%20let%20a%20bit%20more%20air%20just%20before%2C%2023c%3F%5D&c=&c=7aaaaaaaf4d4b34ad2b2d4ad54ac%20%5Bg052%20-%3E%20Let%20out%20air%5D&c=7aaaaaaaf4d4b34ad2b2d4ad5354%20%5Bg054%20%3D%20result.%5D&c=7aaaaaaaf4d4b34ad2b2d4ad5354%20%5Bg067%20-%3E%20Let%20out%20TINY%20amount%20of%20air%5D&c=7aaaaaaaf4d4b34ad2b2d4ad532a%20%5Bg069%20%3D%20result%5D&c=&c=7aaaaaaaf4d4b34ad2b2d4ad52b2%20%5Bg083%20-%20Let%20a%20bunch%20of%20air%20out%20to%20reset%20the%20panic%20mode.%20Tire%20on%20balcony%20to%20cool.%5D&c=7aaaaaaaf4d4b34ad2b2d4ad3554%20%5Bg085%5D&c=7aaaaaaaf4d4b34ad2b2d4ad3554%20%5Bg086%20Pressure%20stabilized.%20Time%20for%20cooling.%5D&c=7aaaaaaaf4d4b34ad2b2d4ad352a%20%5Bg093%5D&c=7aaaaaaaf4d4b34ad2b2d4ad34cc%20%5Bg116%20around%209%20degrees%20outside%20rubber%5D&c=7aaaaaaaf4d4b34ad2b2d4ad34b2%20%5Bg123%5D&c=7aaaaaaaf4d4b34Ad2b2d4ad334a%20%5Bg150%20-%20manually%20fixed%20bit%20in%20decoding%5D&c=7aaaaaaaf4d4b34ad2b2d4ad332c%20%5Bg167%5D&c=7aaaaaaaf4d4b34ad2b2d4ad32d2%20%5Bg201%20Rubber%200%20degrees%5D&c=&c=%5BRound%203%5D&c=7aaaaaaaf4d4b34ad2b2d4ad54ca%20%5BTire%201%5D&c=7aaaaaaaf4d52aab2d2b54ad4b34%20%5BTire%202%5D&c=7aaaaaaaf4cd4acccaab54ad4b4c%20%5BTire%203%5D&c=7aaaaaaaf4d52aab2d2b54ad34b4%20%5BTire%202%20at%2024psi%5D&c=7aaaaaaaf4cd4acccaab54ad2b52%20%5BTire%203%20at%2022psi%5D&c=7aaaaaaaf4cd34ab2ccb54acd4d4%20%5BTire%204%20at%2020psi%5D&f=MODE%3A2d%20TPMS_ID%3A20h%20UNKNOWN%3A5b%20(PSI%2BTHREE)*FOUR%3D8d%20UNKNOWN%3A2b&a=Preamble&m=555e&d=MC)

gcormier, 2019-04-02:
Nice work!

It is odd, the picture shows more values for the sensor ID.. but I don't see where they would be unless they are all matching (what would be the odds?!) in the 4 tires I found.

klohner, 2019-04-02:
I'd guess each sensor is only advertising one ID.  Since the photo you provided is from a random ebay listing and not the exact sensor you're scanning, it's difficult to tell which ID printed on the sensor is being advertised, and if I aligned the bits correctly to pick it up that way.

Unless you can view the sensor you're scanning (I'm guessing that's unlikely), maybe you can use some kind of OBDII device to have the car tell you what it thinks the IDs of the sensors are?  I think that unless we can get some kind of confirmation of what the TPMS_ID of the device is supposed to be, we're just guessing at bits.

Also, I'm not seeing any indication that the sensor is advertising a temperature reading.

gcormier, 2019-04-02:
The Leaf app won't divulge the sensor ID unfortunately.

I concur on the temperature - I hit it with freezing spray and never noticed a difference.

One last idea is that perhaps there is another mode activated by regular driving that differs from the panic mode, and subsequent "frequent update" mode. Tail end of winter up here, but know that I know a bit of what we're looking for, I can go for a ride along when these wheels are back on the car to see what's getting spit out.

panarom, 2019-04-26:
@gcormier while you were working on this, did you happen to stumble across any information related to Honda TPMS sensors?

gcormier, 2019-04-26:
No I didnt.

alwilson, 2021-02-03:
I was able to ride around in a 2015 Nissan Leaf and record some data. I used what I found in this issue to write a decoder based off the Citroen decoder. After combing through the data looking for 4 distinct tires I think the data field may be 24 bits wide, b/c that gave me 4 distinct IDs and the time between each set of transmissions was ~65s.

I should put it under a feature branch, but right now I have changes for it sitting on master here:
https://github.com/alwilson/rtl_433/commits/master

Before: MODE:2d TPMS_ID:20h UNKNOWN:5b (PSI+THREE)*FOUR=8d UNKNOWN:2b
After: MODE:3d TPMS_ID:24h (PSI+THREE)*FOUR=8d UNKNOWN:2b

alwilson, 2021-02-03:
Timestamps would help, but here's [the data on BitBench](http://triq.net/bitbench?c=%5B-X%20%22name%3Dtest%2Cmodulation%3DFSK_PCM%2Cs%3D120%2Cl%3D120%2Cs%3D120%2Cr%3D9000%22%5D&c=%5B0xf5555555e%20preamble%20removed%2C%20Manchester%20(G.E.%20Thomas)%20Decoded%2C%2037%20bits%20data%5D&c=&c=%5B~10%20min%20drive%2C%20ANT500%20HackRF%20recording%20in%20passenger%20seat%20of%202015%20Nissan%20Leaf%5D&c=&c=f5555555e9aaa9aa599a66655998%20%5B%5D&c=f5555555e9aaa9aa599a66655998%20%5B%5D&c=f5555555e9aaa9aa599a66655998%20%5B%5D&c=f5555555e9a96666596559a566a4%20%5B%5D&c=f5555555e9a96666596559a566a4%20%5B%5D&c=f5555555eaa9666659655a6566a8%20%5B%5D&c=f5555555eaa9666659655a6566a8%20%5B%5D&c=f5555555eaa9666659655a6566a8%20%5B%5D&c=f5555555eaa9666659655a6566a8%20%5B%5D&c=f5555555eaaaa9aa599a66655a64%20%5B%5D&c=f5555555eaaaa9aa599a66655a64%20%5B%5D&c=f5555555eaaaa9aa599a66655a64%20%5B%5D&c=f5555555eaaaa9aa599a66655a64%20%5B%5D&c=f5555555eaa96666596559a56964%20%5B%5D&c=f5555555eaa96666596559a56964%20%5B%5D&c=f5555555eaaaa9aa599a66656568%20%5B%5D&c=f5555555eaaaa9aa599a66656568%20%5B%5D&c=f5555555e9a9666659655a6569a4%20%5B%5D&c=f5555555e9a9666659655a6569a4%20%5B%5D&c=f5555555eaa96666596559a56a54%20%5B%5D&c=f5555555eaa96666596559a56a54%20%5B%5D&c=f5555555eaa96666596559a56a54%20%5B%5D&c=f5555555eaa96666596559a56a54%20%5B%5D&c=f5555555eaaaa9aa599a666565a4%20%5B%5D&c=f5555555eaaaa9aa599a666565a4%20%5B%5D&c=f5555555eaaaa9aa599a666565a4%20%5B%5D&c=f5555555eaaaa9aa599a666565a4%20%5B%5D&c=f5555555eaa9666659655a656a68%20%5B%5D&c=f5555555eaa9666659655a656a68%20%5B%5D&c=f5555555eaa9666659655a656a68%20%5B%5D&c=f5555555eaa9666659655a656a68%20%5B%5D&c=f5555555eaaaa9aa599a66656658%20%5B%5D&c=f5555555eaaaa9aa599a66656658%20%5B%5D&c=f5555555eaaaa9aa599a66656658%20%5B%5D&c=f5555555e9a9666659655a659558%20%5B%5D&c=f5555555e9a9666659655a659558%20%5B%5D&c=f5555555e9a9666659655a659558%20%5B%5D&c=f5555555e9a9666659655a659558%20%5B%5D&c=f5555555eaa96666596559a56aa8%20%5B%5D&c=f5555555eaa96666596559a56aa8%20%5B%5D&c=f5555555eaa96666596559a56aa8%20%5B%5D&c=f5555555eaaaa9aa599a66656658%20%5B%5D&c=f5555555eaaaa9aa599a66656658%20%5B%5D&c=f5555555eaaaa9aa599a66656658%20%5B%5D&c=f5555555eaa9666659655a659568%20%5B%5D&c=f5555555eaa9666659655a659568%20%5B%5D&c=f5555555eaa9666659655a659568%20%5B%5D&c=f5555555eaa9666659655a659568%20%5B%5D&c=f5555555eaa9666659655aa56a64%20%5B%5D&c=f5555555eaa96666596559a59554%20%5B%5D&c=f5555555eaa96666596559a59554%20%5B%5D&c=f5555555eaa96666596559a59554%20%5B%5D&c=f5555555eaa96666596559a59554%20%5B%5D&c=f5555555eaaaa9aa599a66656964%20%5B%5D&c=f5555555eaaaa9aa599a66656964%20%5B%5D&c=f5555555eaaaa9aa599a66656964%20%5B%5D&c=f5555555eaaaa9aa599a66656964%20%5B%5D&c=f5555555eaa9666659655a659568%20%5B%5D&c=f5555555eaa9666659655a659568%20%5B%5D&c=f5555555eaa9666659655a659568%20%5B%5D&c=f5555555eaa9666659655a659568%20%5B%5D&c=f5555555eaa9666659655aa56a98%20%5B%5D&c=f5555555eaa9666659655aa56a98%20%5B%5D&c=f5555555eaa96666596559a59554%20%5B%5D&c=f5555555eaa96666596559a59554%20%5B%5D&c=f5555555eaa96666596559a59554%20%5B%5D&c=f5555555eaa96666596559a59554%20%5B%5D&c=f5555555eaaaa9aa599a66656964%20%5B%5D&c=f5555555eaaaa9aa599a66656964%20%5B%5D&c=f5555555eaaaa9aa599a66656964%20%5B%5D&c=f5555555eaaaa9aa599a66656964%20%5B%5D&c=f5555555eaa9666659655a6595a4%20%5B%5D&c=f5555555eaa9666659655a6595a4%20%5B%5D&c=f5555555eaa9666659655a6595a4%20%5B%5D&c=f5555555eaa9666659655a6595a4%20%5B%5D&c=f5555555eaa96666596559a595a8%20%5B%5D&c=f5555555eaa96666596559a595a8%20%5B%5D&c=f5555555eaaaa9aa599a66656998%20%5B%5D&c=f5555555eaaaa9aa599a66656998%20%5B%5D&c=f5555555eaaaa9aa599a66656998%20%5B%5D&c=f5555555eaa9666659655a659658%20%5B%5D&c=f5555555eaa9666659655a659658%20%5B%5D&c=f5555555eaa9666659655a659658%20%5B%5D&c=f5555555eaa9666659655aa59598%20%5B%5D&c=f5555555eaa96666596559a59664%20%5B%5D&c=f5555555eaa96666596559a59664%20%5B%5D&c=f5555555eaaaa9aa599a66656a54%20%5B%5D&c=f5555555eaaaa9aa599a66656a54%20%5B%5D&f=MODE%3A3d%20TPMS_ID%3A24h%20(PSI%2BTHREE)*FOUR%3D8d%20UNKNOWN%3A2b&a=Preamble&m=555e&d=MC&cw=4) with the 24-bit ID encoding.

Although, now that I look at how many times I saw each TPMS_ID I only saw one tire 4 times and only near the end of the drive.
```
fde56a [32]:
	@24.390045s     psi: 30.2 mode: 101 unknown: 01
	@24.600241s     psi: 30.2 mode: 101 unknown: 01
	@24.705465s     psi: 30.2 mode: 101 unknown: 01
	@86.369972s     psi: 30.5 mode: 111 unknown: 10
	@86.474937s     psi: 30.5 mode: 111 unknown: 10
	@86.580162s     psi: 30.5 mode: 111 unknown: 10
	@86.685417s     psi: 30.5 mode: 111 unknown: 10
	@148.351578s    psi: 31.0 mode: 111 unknown: 11
	@148.456558s    psi: 31.0 mode: 111 unknown: 11
	@210.336441s    psi: 31.2 mode: 111 unknown: 10
	@210.441452s    psi: 31.2 mode: 111 unknown: 10
	@210.546677s    psi: 31.2 mode: 111 unknown: 10
	@210.651947s    psi: 31.2 mode: 111 unknown: 10
	@272.373566s    psi: 31.5 mode: 111 unknown: 01
	@272.583801s    psi: 31.5 mode: 111 unknown: 01
	@272.689087s    psi: 31.5 mode: 111 unknown: 01
	@334.495361s    psi: 31.5 mode: 111 unknown: 01
	@334.600647s    psi: 31.5 mode: 111 unknown: 01
	@334.705933s    psi: 31.5 mode: 111 unknown: 01
	@396.437103s    psi: 32.0 mode: 111 unknown: 10
	@396.542114s    psi: 32.0 mode: 111 unknown: 10
	@396.647400s    psi: 32.0 mode: 111 unknown: 10
	@396.752716s    psi: 32.0 mode: 111 unknown: 10
	@458.468231s    psi: 32.0 mode: 111 unknown: 10
	@458.573242s    psi: 32.0 mode: 111 unknown: 10
	@458.678558s    psi: 32.0 mode: 111 unknown: 10
[... cut ...]
	@474.786804s    psi: 33.2 mode: 111 unknown: 10
	@474.894806s    psi: 33.2 mode: 111 unknown: 10
	@475.002838s    psi: 33.2 mode: 111 unknown: 10
	@539.238220s    psi: 33.5 mode: 111 unknown: 01
	@539.346191s    psi: 33.5 mode: 111 unknown: 01
	@539.454224s    psi: 33.5 mode: 111 unknown: 01
caa487 [4]:
	@351.041595s    psi: 32.5 mode: 111 unknown: 10
	@415.880249s    psi: 32.8 mode: 111 unknown: 01
	@416.209229s    psi: 32.8 mode: 111 unknown: 01
	@546.356567s    psi: 33.2 mode: 111 unknown: 01
```

MotorvateDIY, 2023-06-15:
> I was able to ride around in a 2015 Nissan Leaf and record some data. I used what I found in this issue to write a decoder based off the Citroen decoder. After combing through the data looking for 4 [...]

Background:
10 months ago, I started to figure out my 2011 G37 TPMS sensors. I read the TPMS IDs from the BCM and that make figuring out the layout fairly easy. PLUS "chzu" helped me to get started, and pointed out a few key parts.

It was just last week I came across your tpms_nissan.c update and it worked great!
The only change I made was to remove the "-3" psi correction, as the is elevation dependant and I want my TPMS display to match the RTL_433 values.

My Ask (Please!!)
I have a G37 focused Youtube channel (MotorvateDIY) and have just finished shooting an episode on how to change your own TPMS sensors and program them. I used your update to show the sensor IDs. (compiled locally)

Since most car guys may not know how to compile code, is there nay chance you can make a windows standalone exe?
With that, anyone can easily read the IDs and use them to program the BCM. (body control module)

I did try to create a windows standalone, and spent 2 days trying Visual Code and CMAKE, but it always gave me an error "this can't be executed" when on the other windows machine. (I work in the embedded field, so DLL are new to me)

Anyways, thanks for your work on this.

Here is my post from 10 months ago:
https://www.reddit.com/r/RTLSDR/comments/wnvyzv/nissaninfiniti_tpms_sensor_decode_question/

zuckschwerdt, 2023-06-15:
I'll try to review and, if possible, merge a decoder tomorrow. We can then create the "nightly" pre-release for Windows.

MotorvateDIY, 2023-06-16:
> I'll try to review and, if possible, merge a decoder tomorrow. We can then create the "nightly" pre-release for Windows.

Thank you, thank you!!

If possible, could you outline the steps you use to make a stand alone exe using CMAKE?
I am new to CMAKE and want to continue to learn more about it. (My first time using it, was last week)

Anyways, thank you again for all the great work you have done with the RTL_433 program!

zuckschwerdt, 2023-06-16:
> could you outline the steps you use to make a stand alone exe using CMAKE?

The steps are automated in https://github.com/merbanan/rtl_433/blob/master/.github/workflows/release.yml
and the build script is https://github.com/merbanan/rtl_433/blob/master/do_sysroot.sh
But really it should be as simple as installing the dependencies and then building like shown in https://github.com/merbanan/rtl_433/blob/master/docs/BUILDING.md#visual-studio-2017
I'm not sure if building librtlsdr is needed as Pothos should have those libs pre-built.

MotorvateDIY, 2023-06-16:
> But really it should be as simple as installing the dependencies and then building like shown in https://github.com/merbanan/rtl_433/blob/master/docs/BUILDING.md#visual-studio-2017

Well.... this is weird. I just followed those instructions again, moved the needed DLLs to a different windows 10 machine and it worked!!! I have no idea what I was doing wrong the last two days!!

I'll look at it again to make sure I understand all this crazy desktop dev stuff.

gdt, 2023-09-29:
Looks like support was merged in #2536. 
I'm calling this fixed; please comment and explain if I'm wrong.

alwilson, 2023-09-29:
@gdt #2536 is still open and unmerged though.
I haven't touched this in a while, but I can fix up the conflict and update the PR. It sounds like it got some testing, but I'd have to get access to a Nissan Leaf again to test.

gdt, 2023-09-29:
Sorry, misread.  Yes, if you can get this to mergeable state -- and feel free to open a PR yourself and we can close the one done on your behalf -- that would be great.  @zuckschwerdt this is about a PR you are a middleman for.

developerfromjokela, 2023-10-26:
@alwilson  I have access to leaf, how to test?

zuckschwerdt, 2023-10-26:
Updated and manually merged with d34e48d
Please try from current master with `-R 248` (there is no MIC so it's default disabled).

developerfromjokela, 2023-10-26:
> Updated and manually merged with [d34e48d](https://github.com/merbanan/rtl_433/commit/d34e48dedc1857ce3e5a1176cd72bc6789ec05c7) Please try from current master with `-R 248` (there is no MIC so it's default disabled).

Alright, I've compiled and running it with ./rtl_433 -R 248.

djadala, 2023-12-23:
checksum is here:
https://github.com/merbanan/rtl_433/blob/a7c5ef124f69a7101a50944b499321f963e4b51f/src/devices/schraeder.c#L237

gdt, 2024-06-05:
Where are we on this?  It seems there is some support in master.  Anything else to do?    Starting 30-day feedback timer to close, unless there's a reason to keep it open.

djadala, 2025-01-15:
> Where are we on this?

link points to nissan  TPMS  checksum,
but here are implementation in GO:
```go
// 00 = OK
func sum2N(message [5]byte) byte {

        sum := message[0] >> 7
        sum += message[0] >> 5
        sum += message[0] >> 3
        sum += message[0] >> 1
        sum += message[0] << 1

        sum += message[1] >> 7
        sum += message[1] >> 5
        sum += message[1] >> 3
        sum += message[1] >> 1
        sum += message[1] << 1

        sum += message[2] >> 7
        sum += message[2] >> 5
        sum += message[2] >> 3
        sum += message[2] >> 1
        sum += message[2] << 1

        sum += message[3] >> 7
        sum += message[3] >> 5
        sum += message[3] >> 3
        sum += message[3] >> 1
        sum += message[3] << 1

        sum += message[4] >> 7
        sum += message[4] >> 5
        sum += message[4] >> 3

        return (^sum) & 0x03
}

```

gdt, 2025-07-01:
@djadala I cannot parse your reply.

Please explain specifically what needs to be done to the rtl_433 codebase.   If the checksum code is wrong, or missing, and there is correct code someplace else, please be clear on that.  Also if the code is in another project, please point to it so we can understand licensing.  If you are the author and are offering it under rtl_433's license, say that.

djadala, 2025-07-01:
@gdt Nissan Leaf TPMS
> Please explain specifically what needs to be done to the rtl_433 codebase.

as i remember, rtl_433 does not have checksum in nissan tpms driver, code is  missing.
i have implemented checksum in go, but i still don't put my code in public repository.
my code is based on :
[rtl_433/src/devices/schraeder.c](https://github.com/merbanan/rtl_433/blob/a7c5ef124f69a7101a50944b499321f963e4b51f/src/devices/schraeder.c#L237)
lines 237 .. 250, this code is in rtl_433
i am the author of published here in previous comment go code, and i offer it under rtl_433's license.

djadala, 2025-07-01:
@gdt may be it is not clear, by 'go', i mean go programming language 
https://go.dev/