merbanan/rtl_433 #1026: Oregon Scientific V3 - WGR800X

ukshady, 2019-03-30:
I have been experimenting with a new Oregon_scientific sensor.  The existing the oregon_scientific decoder does not recognise the WGR800X device.  I cannot determine what needs to change to address this..

I have uploaded the samples to rtl_tests and have been using bitbench to test, but I have been unable to split the repeat messages into separate messge blocks.  I can see that this contains 3 repeats of the same message.

rtl_433 -X "n=WGR800X,m=OOK_MC_ZEROBIT,s=440,l=2400,r=2400"

time      : 2019-03-27 20:00:16
model     : WGR800X      count     : 1             num_rows  : 1             rows      :
len       : 335          data      : 000000b122400e0e06000000f9e8000000b122400e0e06000000f9e8000000b122400e0e06000000f9e8
codes     : {335}000000b122400e0e06000000f9e8000000b122400e0e06000000f9e8000000b122400e0e06000000f9e8
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2019-03-27 20:00:30
model     : WGR800X      count     : 1             num_rows  : 1             rows      :
len       : 335          data      : 000000b122400e0e06000000f9e8000000b122400e0e06000000f9e8000000b122400e0e06000000f9e8
codes     : {335}000000b122400e0e06000000f9e8000000b122400e0e06000000f9e8000000b122400e0e06000000f9e8
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2019-03-27 20:00:44
model     : WGR800X      count     : 1             num_rows  : 1             rows      :
len       : 335          data      : 000000b122400e0e06000000f9e8000000b122400e0e06000000f9e8000000b122400e0e06000000f9e8
codes     : {335}000000b122400e0e06000000f9e8000000b122400e0e06000000f9e8000000b122400e0e06000000f9e8
[http://triq.net/bitbench?c=000000b122400e0006000000197c&c=000000b122400e0006000000197c%20%5BNorth%5D&c=000000b122400e0406000000589e%20%5BEast%5D&c=000000b122400e0206000000384c%20%5BSouth%5D&c=000000b122400e060600000079ae%20%5BWest%5D&c=000000b122400e16060000017922%20%5BWest%5D&c=000000b122400e16060000017922%20%5BWest%5D&c=&c=000000b122400e100600000119f0%20%5BNE%5D&c=000000b122400e100600000119f0%20%5BNorth%20East%5D&c=000000b122400e0c06000000d8d8%20%5BSouth%20East%5D&c=000000b122400e0a06000000b80a%20%5BSW%5D&c=000000b122400e0a06000000b80a%20%5BSW%5D&c=000000b122400e14060000015812%20%5BEES%5D&c=000000b122400e1c06000001d854%20%5BESS%5D&c=000000b122400e0e06000000f9e8%20%5BNW%5D&c=000000b122400e1e0600080105cc%20%5BWind%5D&c=000000b122400e00061008019878%20%5Bnorth%20wind%5D&c=000000b122400e1e060004018530%20%5Bnw%5D&c=000000b122400e1606001000f872%20%5Bwest%5D&c=000000b122400e0e07c01000c574%20%5Bnw%5D&f=PRE%3A24h%20SYNC%3Ah%20ID%3A16h%20CHAN%3Ah%20BAT%3Ah%20ROLL%3Ah%20DIR%3A7b%20DIR%3Ad%20GUST%3Add%20AVG%3Add%20CRC%3A24h](url)

ukshady, 2019-04-01:
I don't understand the v3 protocol code well enough in /src/devices/oregon_scientific.c.  I think its a matter of adding a new pattern match for this particular sync nibble.  I got this erro by running the following command  rtl_433 -f 433920000 -p 33 -M Oregon-WGR800 -vv
Possible Oregon Scientific v3 message, but sync nibble wasn't found
Raw Data: {335} 00 00 00 b1 22 40 0e 10 06 00 00 01 19 f0 00 00 00 b1 22 40 0e 10 06 00 00 01 19 f0 00 00 00 b1 22 40 0e 10 06 00 00 01 19 f0

zuckschwerdt, 2019-04-01:
Yes, that OS code looks really unfriendly. I hope to refactor it some day.

zuckschwerdt, 2019-04-12:
There is some debug code in #1045 now. It will just split the message an print some debug values. Please test, refine the fields/values and collect a number of lines (say 50) to get the checksum.

ukshady, 2019-04-18:
Using rtl_433 -f 433920000 -p 33 -M Oregon-WGR800 -vv

Tuned to 433.920MHz.
Badly formatted OS v2.1 message: {335} 00 00 00 b1 22 40 0e 00 06 00 00 00 19 7c 00 00 00 b1 22 40 0e 00 06 00 00 00 19 7c 00 00 00 b1 22 40 0e 00 06 00 00 00 19 7c
Possible Oregon Scientific v3 message, but sync nibble wasn't found
Raw Data: {335} 00 00 00 b1 22 40 0e 00 06 00 00 00 19 7c 00 00 00 b1 22 40 0e 00 06 00 00 00 19 7c 00 00 00 b1 22 40 0e 00 06 00 00 00 19 7c

Badly formatted OS v2.1 message: {335} 00 00 00 b1 22 40 0e 18 06 00 00 01 99 b6 00 00 00 b1 22 40 0e 18 06 00 00 01 99 b6 00 00 00 b1 22 40 0e 18 06 00 00 01 99 b6
Possible Oregon Scientific v3 message, but sync nibble wasn't found
Raw Data: {335} 00 00 00 b1 22 40 0e 18 06 00 00 01 99 b6 00 00 00 b1 22 40 0e 18 06 00 00 01 99 b6 00 00 00 b1 22 40 0e 18 06 00 00 01 99 b6
preamble 1  547 5471

Badly formatted OS v2.1 message: {326} 00 00 00 b1 22 40 0e 18 06 00 00 01 99 b6 00 00 62 44 80 1c 30 0c 00 00 03 33 6c 00 00 01 62 44 80 1c 30 0c 00 00 03 33 6c
Possible Oregon Scientific v3 message, but sync nibble wasn't found
Raw Data: {326} 00 00 00 b1 22 40 0e 18 06 00 00 01 99 b6 00 00 62 44 80 1c 30 0c 00 00 03 33 6c 00 00 01 62 44 80 1c 30 0c 00 00 03 33 6c
wt450_callback: wrong size of bit per row 3
preamble 1  567 5671
preamble 1  595 5951

I will capture some more data tomorrow.

dgrostoto, 2019-07-14:
i added support for 2 other oregon sensors (THGR328N and RTGR328N) in the same file.
see https://github.com/merbanan/rtl_433/issues/1107
This may lead to some merge work in the oregon_scientific.c file

ukshady, 2019-08-26:
Things look promising using branch feat-wgr800x 
Here are some results.  Let me know if you want more results, as today is rather wind free :)

Thanks
Adrian

pi@greenpi:~/rtl_433 $ ../wgr800x/rtl_433/build/src/rtl_433 -d 1
rtl_433 version 18.12-196-g800ae08 branch feat-wgr800x at 201904161332 inputs file rtl_tcp RTL-SDR
Trying conf file at "rtl_433.conf"...
Trying conf file at "/home/pi/.config/rtl_433/rtl_433.conf"...
Trying conf file at "/usr/local/etc/rtl_433/rtl_433.conf"...
Trying conf file at "/etc/rtl_433/rtl_433.conf"...

        Consider using "-M newmodel" to transition to new model keys. This will become the default someday.
        A table of changes and discussion is at https://github.com/merbanan/rtl_433/pull/986.

Registered 98 out of 125 device decoding protocols [ 1-4 8 11-12 15-17 19-21 23 25-26 29-36 38-60 62-63 67-71 73-100 102-103 108-116 119 121 124-125 ]
Detached kernel driver
Found Rafael Micro R820T tuner
Exact sample rate is: 250000.000414 Hz
[R82XX] PLL not locked!
Sample rate set to 250000 S/s.
Tuner gain set to Auto.
Tuned to 433.920MHz.
WGR800X: {160} 0b 12 24 00 e0 e0 60 00 00 0f 9e 80 00 00 00 00 00 00 00 00
WGR800X: SYNC:0 ID: e0e CHAN:0 BAT:6 ROLL:0 DIR:0 GUST:15 AVG:40576 CRC: 0
WGR800X: {160} 0f 9e 80 00 00 0b 12 24 00 e0 e0 60 00 00 00 00 00 00 00 00
WGR800X: SYNC:0 ID:   0 CHAN:b BAT:1 ROLL:2 DIR:36 GUST:224 AVG:57440 CRC: 0
WGR800X: {160} 0b 12 24 00 e0 e0 60 00 00 0f 9e 80 00 00 00 00 00 00 00 00
WGR800X: SYNC:0 ID: e0e CHAN:0 BAT:6 ROLL:0 DIR:0 GUST:15 AVG:40576 CRC: 0
WGR800X: {131} 0f 9e 80 00 00 0b 12 24 00 e0 e0 60 00 00 00 00 00
WGR800X: SYNC:0 ID:   0 CHAN:b BAT:1 ROLL:2 DIR:36 GUST:224 AVG:57440 CRC: 0
WGR800X: {91} 0b 12 24 00 e0 e0 60 00 00 0f 9e 80
WGR800X: SYNC:0 ID: e0e CHAN:0 BAT:6 ROLL:0 DIR:0 GUST:15 AVG:40576 CRC: 0
WGR800X: {19} 0f 9e 80 : 00001111 10011110 100
WGR800X: SYNC:0 ID:   0 CHAN:0 BAT:0 ROLL:0 DIR:0 GUST:0 AVG:0 CRC: 0
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2019-08-26 16:37:20
model     : Nexus Temperature/Humidity             House Code: 31
Channel   : 2            Battery   : LOW           Temperature: 31.60 C      Humidity  : 49 %
WGR800X: {160} 0b 12 24 00 e0 60 60 00 00 07 9a e0 00 00 00 00 00 00 00 00
WGR800X: SYNC:0 ID: e06 CHAN:0 BAT:6 ROLL:0 DIR:0 GUST:7 AVG:39648 CRC: 0
WGR800X: {160} 0f 35 c0 00 00 16 24 48 01 c0 c0 c0 00 00 00 00 00 00 00 00
WGR800X: SYNC:0 ID:   1 CHAN:6 BAT:2 ROLL:4 DIR:72 GUST:448 AVG:49344 CRC: 0
WGR800X: {160} 0b 12 24 00 e0 60 60 00 00 07 9a e0 00 00 00 00 00 00 00 00
WGR800X: SYNC:0 ID: e06 CHAN:0 BAT:6 ROLL:0 DIR:0 GUST:7 AVG:39648 CRC: 0
WGR800X: {130} 0f 35 c0 00 00 16 24 48 01 c0 c0 c0 00 00 00 00 00
WGR800X: SYNC:0 ID:   1 CHAN:6 BAT:2 ROLL:4 DIR:72 GUST:448 AVG:49344 CRC: 0
WGR800X: {91} 0b 12 24 00 e0 60 60 00 00 07 9a e0
WGR800X: SYNC:0 ID: e06 CHAN:0 BAT:6 ROLL:0 DIR:0 GUST:7 AVG:39648 CRC: 0
WGR800X: {18} 0f 35 c0 : 00001111 00110101 11
WGR800X: SYNC:0 ID:   0 CHAN:0 BAT:0 ROLL:0 DIR:0 GUST:0 AVG:0 CRC: 0
WGR800X: {160} 0b 12 24 00 e0 60 60 00 00 07 9a e0 00 00 00 00 00 00 00 00
WGR800X: SYNC:0 ID: e06 CHAN:0 BAT:6 ROLL:0 DIR:0 GUST:7 AVG:39648 CRC: 0
WGR800X: {160} 0f 35 c0 00 00 16 24 48 01 c0 c0 c0 00 00 00 00 00 00 00 00
WGR800X: SYNC:0 ID:   1 CHAN:6 BAT:2 ROLL:4 DIR:72 GUST:448 AVG:49344 CRC: 0
WGR800X: {160} 0b 12 24 00 e0 60 60 00 00 07 9a e0 00 00 00 00 00 00 00 00
WGR800X: SYNC:0 ID: e06 CHAN:0 BAT:6 ROLL:0 DIR:0 GUST:7 AVG:39648 CRC: 0
WGR800X: {130} 0f 35 c0 00 00 16 24 48 01 c0 c0 c0 00 00 00 00 00
WGR800X: SYNC:0 ID:   1 CHAN:6 BAT:2 ROLL:4 DIR:72 GUST:448 AVG:49344 CRC: 0
WGR800X: {91} 0b 12 24 00 e0 60 60 00 00 07 9a e0
WGR800X: SYNC:0 ID: e06 CHAN:0 BAT:6 ROLL:0 DIR:0 GUST:7 AVG:39648 CRC: 0
WGR800X: {18} 0f 35 c0 : 00001111 00110101 11
WGR800X: SYNC:0 ID:   0 CHAN:0 BAT:0 ROLL:0 DIR:0 GUST:0 AVG:0 CRC: 0
WGR800X: {160} 0b 12 24 00 e1 80 60 00 00 19 9b 60 00 00 00 00 00 00 00 00
WGR800X: SYNC:0 ID: e18 CHAN:0 BAT:6 ROLL:0 DIR:0 GUST:25 AVG:39776 CRC: 0
WGR800X: {160} 0c cd b0 00 00 05 89 12 00 70 c0 30 00 00 00 00 00 00 00 00
WGR800X: SYNC:0 ID:   0 CHAN:5 BAT:8 ROLL:9 DIR:18 GUST:112 AVG:49200 CRC: 0
WGR800X: {160} 0b 12 24 00 e1 80 60 00 00 19 9b 60 00 00 00 00 00 00 00 00
WGR800X: SYNC:0 ID: e18 CHAN:0 BAT:6 ROLL:0 DIR:0 GUST:25 AVG:39776 CRC: 0
[... 57 lines cut ...]
WGR800X: SYNC:0 ID:   0 CHAN:0 BAT:0 ROLL:0 DIR:0 GUST:0 AVG:0 CRC: 0
WGR800X: {160} 0b 12 24 00 e0 00 60 00 00 01 97 c0 00 00 00 00 00 00 00 00
WGR800X: SYNC:0 ID: e00 CHAN:0 BAT:6 ROLL:0 DIR:0 GUST:1 AVG:38848 CRC: 0
WGR800X: {160} 0c be 00 00 00 58 91 20 07 00 03 00 00 00 00 00 00 00 00 00
WGR800X: SYNC:0 ID:   5 CHAN:8 BAT:9 ROLL:1 DIR:32 GUST:1792 AVG:768 CRC: 0
WGR800X: {160} 0b 12 24 00 e0 00 60 00 00 01 97 c0 00 00 00 00 00 00 00 00
WGR800X: SYNC:0 ID: e00 CHAN:0 BAT:6 ROLL:0 DIR:0 GUST:1 AVG:38848 CRC: 0
WGR800X: {128} 0c be 00 00 00 58 91 20 07 00 03 00 00 00 00 00
WGR800X: SYNC:0 ID:   5 CHAN:8 BAT:9 ROLL:1 DIR:32 GUST:1792 AVG:768 CRC: 0
WGR800X: {91} 0b 12 24 00 e0 00 60 00 00 01 97 c0
WGR800X: SYNC:0 ID: e00 CHAN:0 BAT:6 ROLL:0 DIR:0 GUST:1 AVG:38848 CRC: 0
WGR800X: {16} 0c be : 00001100 10111110
WGR800X: SYNC:0 ID:   0 CHAN:0 BAT:0 ROLL:0 DIR:0 GUST:0 AVG:0 CRC: 0
WGR800X: {160} 0b 12 24 00 e1 00 60 00 00 11 9f 00 00 00 00 00 00 00 00 00
WGR800X: SYNC:0 ID: e10 CHAN:0 BAT:6 ROLL:0 DIR:0 GUST:17 AVG:40704 CRC: 0
[... cut ...]
WGR800X: {160} 0b 12 24 00 e1 00 60 00 00 11 9f 00 00 00 00 00 00 00 00 00
WGR800X: SYNC:0 ID: e10 CHAN:0 BAT:6 ROLL:0 DIR:0 GUST:17 AVG:40704 CRC: 0
WGR800X: {132} 08 cf 80 00 00 05 89 12 00 70 80 30 00 00 00 00 00
WGR800X: SYNC:0 ID:   0 CHAN:5 BAT:8 ROLL:9 DIR:18 GUST:112 AVG:32816 CRC: 0
WGR800X: {91} 0b 12 24 00 e1 00 60 00 00 11 9f 00
WGR800X: SYNC:0 ID: e10 CHAN:0 BAT:6 ROLL:0 DIR:0 GUST:17 AVG:40704 CRC: 0
WGR800X: {20} 08 cf 80 : 00001000 11001111 1000
WGR800X: SYNC:0 ID:   0 CHAN:0 BAT:0 ROLL:0 DIR:0 GUST:0 AVG:0 CRC: 0

zuckschwerdt, 2019-08-26:
Thanks, I'll look through the codes if something stands out, still unclear where the preamble/sync-word is and what checksum there is. Should be easier to figure out the fields once we have that.

zuckschwerdt, 2019-08-26:
If we take the 11 bytes that actually have bits we get [this BitBench](http://triq.net/bitbench?c=b122400e0e06000000f9e8&c=f9e8000000b122400e0e06&c=b122400e060600000079ae&c=f35c0000016244801c0c0c&c=b122400e180600000199b6&c=ccdb000000589120070c03&c=b122400e0206000000384c&c=e130000002c48900380818&c=b122400e0006000000197c&c=cbe0000005891200700030&c=b122400e100600000119f0&c=8cf8000000589120070803&f=v) (duplicates removed, all codes had duplicates which indicates good data). There is some rhythm, looks like two alternating different message types.

zuckschwerdt, 2019-08-26:
Oh, wrong. The second lines are just shifted badly and actually contain the same `b122400e` pattern. It really boils down [to these codes](http://triq.net/bitbench?c=b122400e0e06000000f9e8&c=b122400e060600000079ae&c=b122400e180600000199b6&c=b122400e0206000000384c&c=b122400e0006000000197c&c=b122400e100600000119f0&f=8h%20).
```
b122400e0e06000000f9e8
b122400e060600000079ae
b122400e180600000199b6
b122400e0206000000384c
b122400e0006000000197c
b122400e100600000119f0
```

zuckschwerdt, 2019-08-26:
Looks like there is a simple byte-wide sum, but it's strangely not byte aligned (off by a nibble).
```
b1 + 22 + 40 + 0e + 00 + 06 + 10 + 00 - 011 = 126
b1 + 22 + 40 + 0e + 00 + 06 + 00 + 08 - 009 = 126
b1 + 22 + 40 + 0e + 00 + 06 + 40 + 08 - 00d = 126
b1 + 22 + 40 + 0e + 0e + 06 + 00 + 00 - 00f = 126
b1 + 22 + 40 + 0e + 06 + 06 + 00 + 00 - 007 = 126
b1 + 22 + 40 + 0e + 18 + 06 + 00 + 00 - 019 = 126
b1 + 22 + 40 + 0e + 02 + 06 + 00 + 00 - 003 = 126
b1 + 22 + 40 + 0e + 00 + 06 + 00 + 00 - 001 = 126
b1 + 22 + 40 + 0e + 10 + 06 + 00 + 00 - 011 = 126
```

zuckschwerdt, 2019-08-26:
Can you list some of the `0b 12 24 00 …` codes with the values from the head unit? We can start to pinpoint some data values then.

ukshady, 2019-08-29:
At the moment I only have the Wind sensor and no head unit.  My past bench tests gave me these results. [bitbench](http://triq.net/bitbench?c=000000b122400e0006000000197c&c=000000b122400e0006000000197c%20%5BNorth%5D&c=000000b122400e0406000000589e%20%5BEast%5D&c=000000b122400e0206000000384c%20%5BSouth%5D&c=000000b122400e060600000079ae%20%5BWest%5D&c=000000b122400e16060000017922%20%5BWest%5D&c=000000b122400e16060000017922%20%5BWest%5D&c=&c=000000b122400e100600000119f0%20%5BNE%5D&c=000000b122400e100600000119f0%20%5BNorth%20East%5D&c=000000b122400e0c06000000d8d8%20%5BSouth%20East%5D&c=000000b122400e0a06000000b80a%20%5BSW%5D&c=000000b122400e0a06000000b80a%20%5BSW%5D&c=000000b122400e14060000015812%20%5BEES%5D&c=000000b122400e1c06000001d854%20%5BESS%5D&c=000000b122400e0e06000000f9e8%20%5BNW%5D&c=000000b122400e1e0600080105cc%20%5BWind%5D&c=000000b122400e00061008019878%20%5Bnorth%20wind%5D&c=000000b122400e1e060004018530%20%5Bnw%5D&c=000000b122400e1606001000f872%20%5Bwest%5D&c=000000b122400e0e07c01000c574%20%5Bnw%5D&f=PRE%3A24h%20SYNC%3Ah%20ID%3A16h%20CHAN%3Ah%20BAT%3Ah%20ROLL%3Ah%20DIR%3A7b%20DIR%3Ad%20GUST%3Add%20AVG%3Add%20CRC%3A24h) results.

zuckschwerdt, 2019-08-30:
Do you know the channel and battery flag positions for sure? Did you test them?

The wind direction values seem to correlate but are a bit wild, can you manually fix a direction and take good readings for the 16 directions?

| value | dir |
| --- | --- |
| 00 (0000000) | 0 deg |
| 01 (0000001) | 180 deg |
| 02 (0000010) | 90 deg |
| 03 (0000011) | 270 deg |
| 04 (0000100) | (should be 45 deg)? |
| 05 (0000101) | 225 deg |
| 06 (0000110) | 135 deg |
| 07 (0000111) | 315 deg |
| 08 (0001000) | 45 deg (should be 67)? |
| 09 (0001001) | (should be 247)? |
| 10 (0001010) | 112 deg |
| 11 (0001011) | 270 deg (should be 292)? |
| 12 (0001100) | (should be 337)? |
| 13 (0001101) | (should be 157)? |
| 14 (0001110) | 157 deg (should be 67)? |
| 15 (0001111) | 315 deg (should be 247)? |

ukshady, 2019-09-03:
I did some tests this evening and have determined the wind direction encoding, see table below.  

Binary | Dec |   | Actual Cardinal
-- | -- | -- | --
0 | 0 |   | North
1 | 1 |   | South
10 | 2 |   | East
11 | 3 |   | West
100 | 4 |   | NE
101 | 5 |   | SW
110 | 6 |   | SE
111 | 7 |   | NW
1000 | 8 |   | NNE
1001 | 9 |   | SSW
1010 | 10 |   | ESE
1011 | 11 |   | WNW
1100 | 12 |   | ENE
1101 | 13 |   | WSW
1110 | 14 |   | SSE
1111 | 15 |   | NNW

The actual direction encoding is **not** as indicated in this [protocol document](http://www.osengr.org/WxShield/Downloads/Weather-Sensor-RF-Protocols.pdf) see page 27.

The bitbench results are  [here](http://triq.net/bitbench?c=000000b122400e0006000000197c&c=000000b122400e0006000000197c%20%5BNorth%5D&c=000000b122400e0406000000589e%20%5BEast%5D&c=000000b122400e0206000000384c%20%5BSouth%5D&c=000000b122400e060600000079ae%20%5BWest%5D&c=000000b122400e16060000017922%20%5BWest%5D&c=000000b122400e16060000017922%20%5BWest%5D&c=&c=000000b122400e100600000119f0%20%5BNE%5D&c=000000b122400e100600000119f0%20%5BNorth%20East%5D&c=000000b122400e0c06000000d8d8%20%5BSouth%20East%5D&c=000000b122400e0a06000000b80a%20%5BSW%5D&c=000000b122400e0a06000000b80a%20%5BSW%5D&c=000000b122400e14060000015812%20%5BEES%5D&c=000000b122400e1c06000001d854%20%5BESS%5D&c=000000b122400e0e06000000f9e8%20%5BNW%5D&c=000000b122400e1e0600080105cc%20%5BWind%5D&c=000000b122400e00061008019878%20%5Bnorth%20wind%5D&c=000000b122400e1e060004018530%20%5Bnw%5D&c=000000b122400e1606001000f872%20%5Bwest%5D&c=000000b122400e0e07c01000c574%20%5Bnw%5D&f=PRE%3A24h%20SYNC%3Ah%20ID%3A16h%20CHAN%3Ah%20BAT%3Ah%20ROLL%3Ah%20DIR%3A7b%20DIR%3Ad%20GUST%3Add%20AVG%3Add%20CRC%3A24h)

See page 32 [protocol document](http://www.osengr.org/WxShield/Downloads/Weather-Sensor-RF-Protocols.pdf).  Current debug output is here. data payload is in **bold** typeface, I believe the SYNC

WGR800X: {160} **0b 12 24 14 63 e0 60 00 00 0a 40 c0** 00 00 00 00 00 00 00 00
WGR800X: SYNC:1 ID:463e CHAN:0 BAT:6 ROLL:0 DIR:0 GUST:10 AVG:16576 CRC: 0
WGR800X: {131} 0a 40 c0 00 00 0b 12 24 14 63 e0 60 00 00 00 00 00
WGR800X: SYNC:0 ID:   0 CHAN:b BAT:1 ROLL:2 DIR:36 GUST:5219 AVG:57440 CRC: 0
WGR800X: {91} **0b 12 24 14 63 e0 60 00 00 0a 40 c0**
WGR800X: SYNC:1 ID:463e CHAN:0 BAT:6 ROLL:0 DIR:0 GUST:10 AVG:16576 CRC: 0
WGR800X: {19} 0a 40 c0 : 00001010 01000000 110
WGR800X: SYNC:0 ID:   0 CHAN:0 BAT:0 ROLL:0 DIR:0 GUST:0 AVG:0 CRC: 0

Earlier experiments using
rtl_433 -X "n=WGR800X,m=OOK_MC_ZEROBIT,s=440,l=2400,r=2400"
had the raw data as

Raw Data: {335} 
00 00 00 b1 22 40 0e 00 06 00 00 00 19 7c 
00 00 00 b1 22 40 0e 00 06 00 00 00 19 7c 
00 00 00 b1 22 40 0e 00 06 00 00 00 19 7c

So this is why I have my bitbech results as 
PRE:24h SYNC:h ID:16h CHAN:h BAT:h ROLL:h DIR:7b DIR:d GUST:dd AVG:dd CRC:24h

Thanks

Adrian

zuckschwerdt, 2019-09-04:
Great work on the wind direction! Do you know what the `ROLL:h`,  `DIR:7b` (not `DIR:d` that's as above) mean? And did you check if `GUST:dd` and `AVG:dd` are sensible values?

zuckschwerdt, 2019-09-04:
Oh well, actually if you reverse the 4 bits the mapping is straight forward! Maybe other bits are also reversed?

Binary | Dec | Angle | Reversed | Dir
-- | -- | -- | -- | --
0000 |  0 |     0 | 0 *22.5 | North
1000 |  8 |  22.5 | 1 *22.5 | NNE
0100 |  4 |    45 | 2 *22.5 | NE
1100 | 12 |  67.5 | 3 *22.5 | ENE
0010 |  2 |    90 | 4 *22.5 | East
1010 | 10 | 112.5 | 5 *22.5 | ESE
0110 |  6 |   135 | 6 *22.5 | SE
1110 | 14 | 157.5 | 7 *22.5 | SSE
0001 |  1 |   180 | 8 *22.5 | South
1001 |  9 | 202.5 | 9 *22.5 | SSW
0101 |  5 |   225 | 10 *22.5 | SW
1101 | 13 | 247.5 | 11 *22.5 | WSW
0011 |  3 |   270 | 12 *22.5 | West
1011 | 11 | 292.5 | 13 *22.5 | WNW
0111 |  7 |   315 | 14 *22.5 | NW
1111 | 15 | 337.5 | 15 *22.5 | NNW

zuckschwerdt, 2019-09-04:
The existing OS v3 decoder can already decode the WGR800X with the WGR800 code. The preamble is just a few bits too short for the current sync algorithm. Doh.
I'll be reworking the sync and then we are done.

zuckschwerdt, 2019-09-04:
Can you check #1045 again and verify all data fields? It should work fine now.

ukshady, 2019-09-04:
Top work zuckshwedt 🥇 

pi@greenpi:~/wgr800x/rtl_433/build/src $ ./rtl_433
rtl_433 version 19.08-4-g017af9e branch feat-wgr800x at 201909041405 inputs file rtl_tcp RTL-SDR
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "rtl_433.conf"...
Trying conf file at "/home/pi/.config/rtl_433/rtl_433.conf"...
Trying conf file at "/usr/local/etc/rtl_433/rtl_433.conf"...
Trying conf file at "/etc/rtl_433/rtl_433.conf"...

        Consider using "-M newmodel" to transition to new model keys. This will become the default someday.
        A table of changes and discussion is at https://github.com/merbanan/rtl_433/pull/986.

Registered 106 out of 136 device decoding protocols [ 1-4 8 11-12 15-17 19-21 23 25-26 29-36 38-60 63 67-71 73-100 102-103 108-116 119 121 12                                                                  4-128 131-136 ]
usb_claim_interface error -6
Detached kernel driver
Found Rafael Micro R820T tuner
Exact sample rate is: 250000.000414 Hz
[R82XX] PLL not locked!
Sample rate set to 250000 S/s.
Tuner gain set to Auto.
Tuned to 433.920MHz.
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2019-09-04 17:37:09                    brand     : OS
model     : WGR800       House Code: 197
Channel   : 0            Battery   : OK            Gust      : 0.0 m/s       Average   : 0.0 m/s       Direction : 0.0 degrees
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2019-09-04 17:37:23                    brand     : OS
model     : WGR800       House Code: 197
Channel   : 0            Battery   : OK            Gust      : 0.0 m/s       Average   : 0.0 m/s       Direction : 180.0 degrees
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2019-09-04 17:37:29
model     : Nexus Temperature/Humidity             House Code: 31
Channel   : 2            Battery   : LOW           Temperature: 15.40 C      Humidity  : 62 %
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2019-09-04 17:37:37                    brand     : OS
model     : WGR800       House Code: 197
Channel   : 0            Battery   : OK            Gust      : 0.0 m/s       Average   : 0.0 m/s       Direction : 270.0 degrees
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2019-09-04 17:37:51                    brand     : OS
model     : WGR800       House Code: 197
Channel   : 0            Battery   : OK            Gust      : 0.0 m/s       Average   : 0.0 m/s       Direction : 90.0 degrees
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2019-09-04 17:38:05                    brand     : OS
model     : WGR800       House Code: 197
Channel   : 0            Battery   : OK            Gust      : 0.0 m/s       Average   : 0.0 m/s       Direction : 90.0 degrees
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2019-09-04 17:38:19                    brand     : OS
model     : WGR800       House Code: 197
Channel   : 0            Battery   : OK            Gust      : 0.4 m/s       Average   : 0.0 m/s       Direction : 337.5 degrees
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2019-09-04 17:38:33                    brand     : OS
model     : WGR800       House Code: 197
Channel   : 0            Battery   : OK            Gust      : 2.5 m/s       Average   : 0.2 m/s       Direction : 0.0 degrees
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2019-09-04 17:38:36
model     : Nexus Temperature/Humidity             House Code: 31
Channel   : 2            Battery   : LOW           Temperature: 15.30 C      Humidity  : 63 %
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2019-09-04 17:38:47                    brand     : OS
model     : WGR800       House Code: 197
Channel   : 0            Battery   : OK            Gust      : 4.7 m/s       Average   : 0.9 m/s       Direction : 0.0 degrees
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2019-09-04 17:39:01                    brand     : OS
model     : WGR800       House Code: 197
Channel   : 0            Battery   : OK            Gust      : 3.6 m/s       Average   : 1.8 m/s       Direction : 0.0 degrees

zuckschwerdt, 2019-09-05:
Support in #1045 merged.

closed 2019-09-05 by zuckschwerdt