merbanan/rtl_433 #1029: Lacrosse WS7000/2500 Sensors

ReMiOS, 2019-04-02:
Is it possible to add the Lacrosse Sensors ?
        WS2500-19 brightness sensor
        WS7000-20 meteo sensor (temperature/humidity/pressure)
        WS7000-16 Rain Sensor
        WS7000-15 wind sensor
       
it appears there is some code in elv.c that could be used 
rtl_433 detects a "ELV-WS2000" 

I've made an protocol description and attached it to this issue
based on http://www.f6fbb.org/domo/sensors/index.php
         and http://plevenon-meteo.info/technique/station/WS2500_sensors_infos.html 

also i've attached cu8 files made with the command "rtl_433 -a -S unknown"

Lastly i've attached CSV output from my Rigol DS1104z scope of the frame sent.
( .CSV format and .SR Pulseview format https://sigrok.org/wiki/Downloads )

Attachments: 
[Lacrosse WS2500&WS7000.docx](https://github.com/merbanan/rtl_433/files/3033500/Lacrosse.WS2500.WS7000.docx)
[RTL_433_tests.zip](https://github.com/merbanan/rtl_433/files/3033502/RTL_433_tests.zip)
[Metingen.zip](https://github.com/merbanan/rtl_433/files/3033506/Metingen.zip)

merbanan, 2019-04-02:
rtl_433 -A -r WS2500-19_g001_433.92M_250k.cu8 -> [00] {66} 00 35 e8 d3 54 ce 63 ee 40
rtl_433 -A -r WS7000-15_g001_433.92M_250k.cu8 -> [00] {61} 00 39 ed 53 10 84 6b 98
rtl_433 -A -r WS7000-16_g001_433.92M_250k.cu8 -> [00] {46} 00 29 eb d3 b4 ec
rtl_433 -A -r WS7000-20_g001_433.92M_250k.cu8 -> [00] {81} 00 25 f9 6a 99 f4 b5 9f 53 b9 80

merbanan, 2019-04-02:
elv.c has some code but that code needs a cleanup. It is probably best to add a new decoder for these sensors. The documentation is complete and bit recovery works fine. So adding these should be fairly easy.

ReMiOS, 2019-04-02:
Thanks for your fast response !

For the 4 sensor cu8 files i've sent the values as described in the docx

Lacrosse WS2500-19
  Frame: 5=0101 7=0111 8=1000 2=0010 5=0101 2=0010 9=1001 9=1001 8=1000 7=0111 3=0011
  Light = 52800 lux;  
  Exposure = 899 min ;

Lacrosse WS7000-15
  Frame: 3=0011 7=0111 5=0101 2=0010 1=0001 0=0000 0=0000 8=1000 A=1010 9=1001
  Windspeed = 12.5 km/h;    
  Direction = 200 degrees +/- 0.1 degree;

Lacrosse WS7000-16
  Frame: 2=0010 7=0111 E=1110 2=0010 B=1011 2=0010 B=1011
  Counts  = 2862;  

Lacrosse WS7000-20
  Frame: 4=0100 F=1111 4=0100 5=0101 2=0010 9=1001 7=0111 4=0100 5=0101 9=1001 7=0111 2=0010 B=1011 9=1001
  Temperature = -25.4 C;    
  Humidity = 47.9 %;    
  Pressure = 995 hPa;

merbanan, 2019-04-02:
Yes, I just checked if rtl_433 could recover the bits properly. It matched the documentation so the only thing left is to code the support.

zuckschwerdt, 2019-04-02:
I don't think the ELV code is related. It's better look at something like [`ambientweather_tx8300.c`](https://github.com/merbanan/rtl_433/blob/master/src/devices/ambientweather_tx8300.c) which is reasonably clean and friendly code.

ReMiOS, 2019-04-02:
This looks promising, although i'm not an C++ wizzard ..

It seems this code has an variable bit length, the Lacrosse sensors 
use an 1200us length for both "1" and "0" bits.
- Bit 1 : 1970us pulse with 3888 us gap
- Bit 0 : 1970us pulse with 1936 us gap

I've also added the 2 remaining cu8 files of the sensors 
Lacrosse WS7000-22
Frame: 1=0001 F=1111 4=0100 5=0101 2=10 9=1001 7=0111 4=0100 7=0111 B=1011
Temperature = -25.4 C;    
Humidity = 47.9 %;    

Lacrosse WS7000-27
Frame:  0=0000 F=1111 3=0011 5=0101 2=0010 B=1011 9=1001
Temperature = -25.3;   

[Lacrosse2.zip](https://github.com/merbanan/rtl_433/files/3034015/Lacrosse2.zip)

zuckschwerdt, 2019-04-02:
We wouldn't want C++ here anyway ;) I might just implement this tonight, the documentation looks solid.

zuckschwerdt, 2019-04-02:
btw. the new align option in BitBench allows to [quickly inspect this data](http://triq.net/bitbench?c=%7B66%7D%2000%2035%20e8%20d3%2054%20ce%2063%20ee%2040&c=%7B69%7D%2000%2006%20bd%201a%206a%2099%20cc%207d%20c8&c=%7B60%7D%2000%2073%20da%20a6%2021%2008%20d7%2030&c=%7B63%7D%2000%200e%207b%2054%20c4%2021%201a%20e6&c=%7B45%7D%2000%2053%20d7%20a7%2069%20d8&c=%7B80%7D%2000%204b%20f2%20d5%2033%20e9%206b%203e%20a7%2073&c=%7B80%7D%2000%204b%20f2%20d5%2033%20e9%206b%203e%20a7%2073&c=%7B60%7D%2000%2063%20f2%20d5%2033%20e9%207b%20b0&c=%7B62%7D%2000%2018%20fc%20b5%204c%20fa%205e%20ec&c=%7B45%7D%2000%2043%20fc%20d5%203b%2098&c=%7B45%7D%2000%2043%20fc%20d5%203b%2098&f=%5E4b%201h%20%0A....%5E4d%201x%20&a=Preamble&m=01).

zuckschwerdt, 2019-04-02:
Truly excellent documentation. See #1030

closed 2019-04-03 by zuckschwerdt