merbanan/rtl_433 #1033: Links to projects working with rtl_433

zuckschwerdt, 2019-04-03:
Please post interesting projects related to rtl_433 here. I will collect them for a wiki page or Readme section.
Name, Link, short description prefered.

kpreid, 2019-04-05:
ShinySDR. https://shinysdr.switchb.org/ Web remote-controllable SDR receiver application supporting multiple simultaneous hardware devices and demodulators, including rtl_433 and other decoding tools.

[It doesn't do anything clever with the collected data other than listing the most recent message from each observed device ID. Yet. Patches welcome.]

fanlessfan, 2019-04-06:
this project convert rtl433 output to mqtt.
https://github.com/james-fry/hassio-addons/blob/master/rtl4332mqtt/rtl2mqtt.sh

winterrace, 2019-04-07:
Still work in progress, but feel free to link to it:

rtl_fl2k_433: an RX/TX prototyping tool. Aims to be a comfortable, GUI-based bridge between >= 10$ RTL-SDR dongles on RX side and >= 5$ FL2K dongles on TX side. Currently, the GUI is available for Win64 only, but this might change in future...
https://github.com/winterrace2/rtl_fl2k_433

Under the hood, it uses a custom lib-style version of rtl_433 located at https://github.com/winterrace2/librtl_433. Beneath being usable as a lib, librtl_433 allows for redirection of stdout/stderr to a callback function of the hosting application and allows the application to get the pulse data information passed for each detected message. I provided it with an equivalent export scheme like librtlsdr or some other osmo libs have, so it should be usable on multiple platforms (a rudimentary linux build script is also contained and working).

A command-line frontend that uses this librtl_433 can be found at https://github.com/winterrace2/rtl_433, which also compiles under Linux (rudimentary build script contained) and MS visual studio. This is mainly for demonstration purposes about how to use the lib because, from the functional view, it has quite identical functionality than the original rtl_433,

loengrin2000, 2019-04-23:
rtl_433 with Snap7 running on Raspberry Pi to inject weather data to industrial control system (PLC - Siemens S7-300 or compatible VIPA) coming from Weather station WH1080 on 868.3MHz.
[RTL433_Snap7_integration_description.zip](https://github.com/merbanan/rtl_433/files/3107420/RTL433_Snap7_integration_description.zip)

dgrostoto, 2019-07-14:
I mention domoticz :
rtl_433 is usable from domoticz with a quite good integration: Domoticz launch rtl_433 with no data detection (relaunch rtl_433 if so) and process csv output format. All command line arguments are usable.

matthewwall, 2019-08-23:
the weewx-sdr driver gets data from rtl_433 and feeds it into weewx.  from there the data can be combined with data from other sources, displayed using any of the many weewx skins, and/or uploaded to many different web services.  the first weewx-sdr release was in 2016.

http://weewx.com/

https://github.com/matthewwall/weewx-sdr

https://github.com/weewx/weewx/wiki#skins

https://github.com/weewx/weewx/wiki#uploaders

veggiefrog, 2019-11-04:
The QRUQSP.org project uses rtl_433 to capture weather sensors to a Raspberry Pi and display them in a weather dashboard running on an old iPad 1. The data can also be sent over Amateur Radio as an APRS beacon. We're working on ways to sync this data to other Raspberry Pi's and the cloud.

https://qruqsp.org/

merbanan, 2020-02-06:
I think we should move this topic to the readme. I'd like to keep issues for stuff that can be closed. This is sort of open ended.

zuckschwerdt, 2020-02-07:
This list can be found in https://github.com/merbanan/rtl_433/blob/master/docs/LINKS.md
Closing as this isn't actionable, but feel free to add more suggestions and links here.

closed 2020-02-07 by zuckschwerdt