merbanan/rtl_433 #1035: prologue sensor, unexpected output

lasseo, 2019-04-03:
This was seen when sniffing:

time      : 2019-04-04 00:17:41
model     : Prologue sensor                        id        : 9
rid       : 84           Channel   : 1             Battery   : LOW           Button    : 0             Temperature: 5.40 C       Humidity  : 204 %
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 

- Looks wrong to have "Button"
- Humidity should be shown as 20.4%

I do not own this sensor so I can not confirm what device it is.

merbanan, 2019-04-04:
The prologue protocol is very simple and used by alot of devices. It also has no checksum so it is hard to verify the data except repeats.

Anyway this sensor reports 204 for humidity. That is CC in hex. Devices without humidity sensors has CC in the field where the humidity data should be. This is a bug that should be fixed as the check is easy to perform.

With regards to the button, I have a device which has a button on the back to manually send a transmission. When I do that the button bit is set. So this parameter is correct for my device.

So keep this issue open until I fixed the humidity==0xCC issue.

zuckschwerdt, 2019-04-04:
Can you confirm that the 5.4 C temperature reading is sensible for this device? Was this just a one off, or a series of similar readings?

lasseo, 2019-04-04:
Thanks for the information.
I am kind of new to this area, so did not know about the button option. With that explanation I guess it make sense to display it.

zuckschwerdt: I was getting this reading several times, during the hour I was playing with RTL-433. As I wrote in the issue report, I do not own the sensor ( must be one of my neighbours ). However my own temperature sensor displayed 5.7 C so for me the 5.4 C value should be considered correct.

Thanks for your support, and sorry if some of my reported issues are already known issues. I am trying to search in open and closed issues before posting.

zuckschwerdt, 2019-04-05:
Humidity 0xCC is fixed (and general clean up) with 579213f.

closed 2019-04-05 by zuckschwerdt