merbanan/rtl_433 #1038: Temperature processing in src/devices/wt450.c likely slightly incorrect

tobiasisenberg, 2019-04-06:
The decoding of the temperature in src/devices/wt450.c is likely slightly incorrect. According to the source code, the decoding instructions came from http://ala-paavola.fi/jaakko/doku.php?id=wt450h and there the temperature is decoded as one single number, not as temp_whole and temp_fraction separately. I independently implemented such a decoding for a personal project, also based on several sources including http://ala-paavola.fi/jaakko/doku.php?id=wt450h, and I largely followed the instructions of that website and get slightly different decoded temperatures. Since my sensor actually displays the measured temperature, at least for my sensor I can say that the decoding of http://ala-paavola.fi/jaakko/doku.php?id=wt450h is correct, the separate treatment of temp_whole and temp_fraction in src/devices/wt450.c is not. The difference is not huge, but it may be important to some.

Another issue is that the treatment of the temperature as 15 bits leads to unrealistically precise temperature values. I don't remember where I got it from, but in my personal project I do not use the last 4 bits of the code for the temperature, thus only use a 12 bit value. This is completely sufficient for the precision:
float temperatureConverted = ((float)temperature - 800.0) / 16.0;

Maybe this helps someone.

zuckschwerdt, 2019-04-06:
The code looks strange, agreed ;) And the difference is that you scale the "fraction" by 256 (by 16 and drop the bottom 4 bits) where our code scales by 100. Do you see the "fraction" byte going above 100 (or if you just look at the top 4 bits above 6? That would show that the scale of 100 isn't appropriate here.

zuckschwerdt, 2019-04-06:
On closer inspection, the fractional value is only 7 bit, i.e. 0-127. A scale of 100 seems equally likely as 128. Can you check which range of values you see in the last 7 bits?

tobiasisenberg, 2019-04-06:
For example, I just received the code

11000001 00110011 01010011 11101111 0000

where 0011 11101111 stands for the whole temperature according to my 12-bit scheme. I convert this to 12.9375C using my equation mentioned in the previous post.

Unfortunately, rtl_433 does not recognize the codes presently, so I cannot provide the log. I see a lot of messages of the sort:

wt450_callback: wrong size of bit per row 3
wt450_callback: wrong size of bit per row 22
wt450_callback: wrong size of bit per row 1
wt450_callback: wrong size of bit per row 2
wt450_callback: wrong size of bit per row 5
wt450_callback: wrong size of bit per row 23
wt450_callback: wrong size of bit per row 111

Yet if I would follow wt450.c, the same received code would be converted to:

temp_whole    = 00111110; //== 62 decimal;
temp_fraction = 01111000; //== 120 decimal;
temp               = 12 + 1.2; // == 13.2

And with my numbers I see a smooth temperature decline presently, whereas wt450.c would have created a "bump". Hope that this helps.

Btw, for the last two bits, I have the following in my records:
// b32 - b33  (2 bits): sequence number (0--2), sent in bursts of 3
// b34 - b34  (1 bit) : Parity (XOR of odd bits in message)
// b35 - b35  (1 bit) : Parity (XOR of even bits in message)
but, again, I don't remember where I got this from. Bits 32 and 33 seem to be 00 for me all the time, while the last two bits are changing.

zuckschwerdt, 2019-04-06:
Thanks, yes that looks like it fits. I'll test the code (looks like lots codes are received too short) and change that.

tobiasisenberg, 2019-04-06:
If you need a log with more values as recognized by rtl_433 let me know, I have a longer log. But these are all for above 13C (I had to wait a bit for the temperature to drop below; and then the problems with the unrecognized codes started). But maybe they help with verifying the last two bits. I could send you a PM if needed.

zuckschwerdt, 2019-04-06:
Fixed with 91dc977.
If you can record samples where the decoding fails please post them here (as zip).

tobiasisenberg, 2019-04-07:
Thanks a lot for the extremely fast fix. I will test again as soon as the fix is propagated to the Windows binaries. Btw, very nice project, very comprehensive list of supported protocols, and works out of the box!

zuckschwerdt, 2019-04-07:
Thanks! Glad to hear.
Sorry about the build, it's triggered now and should be ready in 5-10 minutes.

tobiasisenberg, 2019-04-07:
Build works. I now see the same temperatures as I computed and as are shown on the sensor, and the decoding works without flaws. As I said before, my specific device does not send a sequence number (always 0, despite the 3 bursts), but that's ok. Again, thanks for the extremely quick response and fix, this issue is resolved and can be closed.

closed 2019-04-07 by tobiasisenberg