merbanan/rtl_433 #1039: how to know the details of the cpatured signals

sidhu037, 2019-04-06:
Hi,

I am using RTL-SDR and I was able to capture few signals. Let me know how I can know the detail about the signal I am capturing. Is there any specific command for it.

merbanan, 2019-04-06:
Use -A to invoke the analyzer. It will tell you alot about the signal. And then you can use tests/sigrok-open.sh to get a good visualization.

closed 2019-04-06 by merbanan

sidhu037, 2019-04-07:
hi,
Could you please let me know how i can execute the sigrok-open.sh. while executing I am going through this error. I have edited the .sh file and added the destination of my captured signal.
[image].

merbanan, 2019-04-07:
You need to install sigrok also. Then you just use tests/sigrok-open.sh gfile.cu8 and it should be fine.

zuckschwerdt, 2019-04-07:
also no `sudo` needed.

@merbanan in current master the sigrok-*.sh moved to examples with all the other scripts.