merbanan/rtl_433 #1043: MQTT - Add ID in events topic

pbendersky, 2019-04-08:
See https://github.com/merbanan/rtl_433/pull/1016#issuecomment-480926807 for original comment.

I'd like to combine the MQTT `events` topic, that sends the raw JSON, but have the topic be prefixed with the device ID, so I can only subscribe to certain sensors. Having the raw JSON is useful as I need to combine the `state` and `open` values in some cases.

Basically, I'm looking to be able to do:

`rtl_433 -f 344975000 -M newmodel -F "mqtt://192.168.1.10,usechannel=beforeid,events=rtl/sensor" -M utc`

and have `rtl_433` publish in `rtl/sensor/<sensor_id>` the raw JSON.

doughecka, 2019-04-09:
From the discussion on #1016:
Just a thought, would having the ability to use 'tags' be useful, so the ID (or any other attribute) can be placed where-ever they want in the topic?
example:
rtl_433 -f 344975000 -M newmodel -M utc -F "mqtt://192.168.1.10,usechannel=beforeid,events=rtl/~ID~/events,devices=rtl/~ID~"
or
rtl_433 -f 344975000 -M newmodel -M utc -F "mqtt://192.168.1.10,usechannel=beforeid,events=rtl/~MODEL~/~ID~/events,devices=rtl/~MODEL~/~ID~"
or
rtl_433 -f 344975000 -M newmodel -M utc -F "mqtt://192.168.1.10,usechannel=beforeid,events=rtl/~MODEL~/~ID~~CHANNEL~/events,devices=rtl/~MODEL~/~ID~~CHANNEL~"
This would reduce the need to have a bunch of options to adjust little things here and there.

zuckschwerdt, 2019-06-23:
This will be possible if we work out #1079 format strings.

closed 2019-06-23 by zuckschwerdt