merbanan/rtl_433 #1044: Using two receivers with MQTT

sotatech, 2019-04-12:
I'm trying to capture two sensors - a Watchman Oil Gauge and a Current Cost electricity meter and post them using the new MQTT feature. Using a single receiver results in a lot of dropouts, so I want to use two - with one dedicated to each sensor.
Howerver, I don't see a way to specifiy the client id and my MQTT broker rejects the second connection with the error "Client rtl_433-7f24ffff already connected, closing old connection"
Is there any way around this?

zuckschwerdt, 2019-04-12:
This is possible, but you need to use different device queries for both instances. E.g. `rtl_433 -d 0 …` and `rtl_433 -d 1 …`.
Assuming you don't care about the order of receivers. If you have different antennas, e.g. for 433 MHz and 868 MHz you should write serial id strings to the receivers and select them with e.g. `rtl_433 :86800001 …`. See http://manpages.ubuntu.com/manpages/trusty/man1/rtl_eeprom.1.html

Which frequencies are those too sensors on?

RandomSporadicProjects, 2019-04-12:
Don't want to put you off using a new feature (I've not migrated yet), but I currently monitor my Oil Watchman and Current Cost Electricity meter (and many more) with 1 usb SDR dongle - both of mine are on 433 MHz.  The Watchman only sends data approx every 20-40 mins and the CC every 6 seconds.  I send all my RTL_433 data to one MQTT topic (RTL_433/Raw) and use a node-red function to separate the data out depending on need.

I've been using a bash script (got it from someones blog and then modified it) on a RPi to forward on the data

#!/bin/bash
pkill -f SDR
pkill mosquitto_pub
/home/pi/resetsdr.sh
#sleep 5;

export LANG=C
PATH="/usr/local/bin:/usr/local/sbin:/usr/bin:/usr/sbin:/bin:/sbin"
 
MQTT_HOST="ipaddressofhost"
 
#
# Start the listener and enter an endless loop
# -G -F json
/usr/local/bin/SDR -G -R -13 -R -10 -R -11 -R -40 -R -41 -R -123 -F json  |  while read line
do
# Log to file if file exists.
# Create file with touch /tmp/rtl_433.log if logging is needed
# [ -w /tmp/rtl_433.log ] && echo $line >> rtl_433.log
 
# Raw message to MQTT
 echo $line  | /usr/bin/mosquitto_pub -h $MQTT_HOST -i RTL_433 -l -t "RTL_433/Raw" -u "user" -P "password"
done

This has worked well for me for a year.  Occasionally the SDR will stop responding, and I've a cronjob that periodically runs this script and it kills any active processes and does a USB reset, which mostly allows it to carry on.  Mine still occasionally fails, but often runs for a couple of months without issue before requiring a manual reset.

sotatech, 2019-04-12:
@zuckschwerdt Thank you, that seems to have worked.
@RandomSporadicProjects Interesting approach, I'll give it a try. I have tried running RTL_433 as a service that pipes the output to MQTT, but it was just too unreliable. It would hang after a day or two requiring a reboot of the RPi. Also, it kept missing readings, leaving large gaps in my graphs, which is why I tried the two receiver approach.
I'll report back after a few days of testing.

sotatech, 2019-04-12:
@RandomSporadicProjects I'm no programmer, but your script refers to SDR and resetsdr.sh. Is SDR an alias for RTL_433 and resetsdr.sh the name of your script?
Thanks.

osborne82, 2019-04-12:
Following as tried running as a service and have to restart every other day

zuckschwerdt, 2019-04-12:
@osborne82 rtl_433 has a watchdog timer and will exit cleanly when the RTL-SDR stops working. This allows for a clean restart to hopefully get the receiver working again. Use something like Supervisord to automate this. See some notes here: http://triq.net/articles/rtl-433-networking

I my experience the RTL-SDR only becomes unreliable with insufficient heak sinking. The default 250k sample rate is no problem but at or above 1024k you will need to add a heat sink or use a receiver with a metal case like the Nooelec NESDR SMArt.

osborne82, 2019-04-12:
Brilliant I'll check that out,
Haven't checked out the respiratory in a while had all the mqtt stuff now
been added
Thanks

zuckschwerdt, 2019-04-12:
Yes, MQTT support has been merged. We really should have some kind of blog-like page to post news like this. Maybe a locked & pinned issue with updates?

osborne82, 2019-04-12:
Maybe add to small wiki like page to show latest news etc

RandomSporadicProjects, 2019-04-13:
> @RandomSporadicProjects I'm no programmer, but your script refers to SDR and resetsdr.sh. Is SDR an alias for RTL_433 and resetsdr.sh the name of your script?
> Thanks.

@sotatech yes, sorry for not being clearer.

I renamed the rtl program (in /usr/bin/local on my RPi) as I had other processes with rtl_433 in them that I didn't want to kill

I give the name runsdrstick.sh to the script I previously posted.

The usbreset saves me having to unplug and plug in the sdr dongle after it has hung which was my previous solution.

I cannot remember where I got it from (so cannot credit the author) but the c is below.  I compiled that, and then run the reset script to allow the main script to use the SDR again after a hang.

/* usbreset -- send a USB port reset to a USB device */
#include <stdio.h> 
#include <unistd.h> 
#include <fcntl.h> 
#include <errno.h> 
#include <sys/ioctl.h> 
#include <linux/usbdevice_fs.h>

int main(int argc,char **argv) {
  const char *filename;
  int fd;
  int rc;
  if(argc != 2)
  {
    fprintf(stderr,"Usage: usbreset device-filename\n");
    return 1;
  }
  filename = argv[1];
  fd = open(filename,O_WRONLY);
  if(fd < 0)
  {
    perror("Error opening output file");
    return 1;
  }
  printf("Resetting USB device %s\n",filename);
  rc = ioctl(fd,USBDEVFS_RESET,0);
  if(rc < 0)
  {
    perror("Error in ioctl");
    return 1;
  }
  printf("Reset successful\n");
  close(fd);
  return 0;
}

usbreset.sh

#!/bin/bash
USBNAME=RTL2838 
LSUSB=$(lsusb | grep --ignore-case $USBNAME) 
DEVICE="/dev/bus/usb/"$(echo $LSUSB | cut --delimiter=' ' --fields='2')"/"$(echo $LSUSB | cut --delimiter=' ' --fields='4' | tr --delete ":")
./usbreset $DEVICE 2>&1 >/dev/null

Now I'm the first to admit that this all seems sub-optimal and clunky, but it works well enough for me

My oil graph for the last year can be found below so you can see how it's worked for me.

[image]

sotatech, 2019-04-14:
Thanks for the clarification, so far your method seems to be working fine with one receiver. I got around the power issues by using a 5V PoE adaptor to supply the Pi from my PoE switch.
Also, I forgot to mention that the -G switch in your command line resulted in reception of various security and TPMS devices, so I took it out.
I'll try to adapt your script to use the built-in MQTT client and report back.

zuckschwerdt, 2019-04-14:
Are your sensors on different bands, e.g 433.92M and 868M? Otherwise the center frequency (and sample rate) can usually be choosen to cover multiple sensors, even if they are somewhat apart in frequency. E.g. sensors on 433.75 and 433.90: choose `-f 833.83M`.

sotatech, 2019-04-14:
No, they are both on 433.920MHz so it's a very simple setup. I have been running the shell script now for over 24 hours without any drop outs, so I think that RTL_433 does not like running as a service.

My script now looks like this:

MQTT_HOST="192.168.16.5"
while :
do
  rtl_433 -R 43 -R 44 -C si -F "mqtt://$MQTT_HOST:1883,user=<user>,pass=<password>"
done

zuckschwerdt, 2019-04-14:
Looks good. You can improve this by using `while sleep 10` instead to throttle (re-)start somewhat.

RandomSporadicProjects, 2019-04-14:
@sotatech glad you got it working.

- G I'm nosey and log all data I find, even TPMS :)

I've been playing with the built in MQTT today too, working well so far

Thanks to all developers

sotatech, 2019-04-14:
@zuckschwerdt Won't that risk missing readings, which is what I'm trying to avoid?

zuckschwerdt, 2019-04-14:
If you notice restarts more than say once or twice a day something is really broken. Usually even the cheapest RTL-SDRs will run weeks without trouble. If the receiver crashes however you likely need a short delay to get it working again.

sotatech, 2019-04-18:
As promised, here's an update on progress. I've converted my script to an rc.d file so that it autostarts on boot. While you would expect that I would encounter the same issues as running it as a service using systemd, the opposite is the case. It is running fine for the last few days on a Pi 2 withut any dropouts, whereas it was previousy on a Pi 3. Also, it is using much less CPU resources using this mehod.

Here's my /etc/init.d/rtl_433_mqtt file:

```
#! /bin/sh
# /etc/init.d/rtl_433_mqtt

### BEGIN INIT INFO
# Provides:          rtl_433_mqtt
# Required-Start:    $remote_fs $syslog
# Required-Stop:     $remote_fs $syslog
# Default-Start:     2 3 4 5
# Default-Stop:      0 1 6
# Short-Description: RTL_433 to MQTT Publisher for Watchman (R 43) and Current Cost CC128 (R 44)
# Description:       The tank depth ranges from 20 (1,100 litres) 86 - almost empty
### END INIT INFO

case "$1" in
  start)
    echo "Starting RTL_433 Script"
    MQTT_HOST="192.168.16.5"
    /usr/local/bin/rtl_433 -R 43 -R 44 -C si -F "mqtt://$MQTT_HOST:1883,user=mqtt-user,pass=mqtt-password"
    ;;

  stop)
    echo "Stopping RTL_433 Script"
    killall rtl_433
    ;;
  *)
    echo "Usage: /etc/init.d/rtl_433_mqtt.sh {start|stop}"
    exit 1
    ;;
esac

exit 0

```

zuckschwerdt, 2019-06-23:
Closing as it looks like startup is worked out now. Feel free to reopen if something needs to be done in rtl_433.

closed 2019-06-23 by zuckschwerdt