merbanan/rtl_433 #1051: Having trouble decoding a garage door remote

ofalvai, 2019-04-28:
I'm trying to decode a garage door remote with the general purpose decoder, and I would appreciate some help with the exact parameters.

Running pulse analyzer and pressing the main button for a few seconds gives the following output:

```
Pulse width distribution:
 [ 0] count:  605,  width:  524 us [444;536]    ( 131 S)
 [ 1] count:    1,  width: 2884 us [2884;2884]  ( 721 S)
 [ 2] count:    2,  width: 7616 us [7240;7992]  (1904 S)
 [ 3] count:    4,  width: 5296 us [4604;5796]  (1324 S)
 [ 4] count:   44,  width: 1572 us [1560;1920]  ( 393 S)
 [ 5] count:  529,  width: 1044 us [1036;1204]  ( 261 S)
 [ 6] count:    2,  width: 10956 us [10516;11396]       (2739 S)
 [ 7] count:    1,  width: 67352 us [67352;67352]       (16838 S)
 [ 8] count:    3,  width: 22036 us [19848;24240]       (5509 S)
 [ 9] count:    3,  width: 49612 us [43080;53604]       (12403 S)
 [10] count:    2,  width: 30948 us [29952;31944]       (7737 S)
 [11] count:    1,  width: 14844 us [14844;14844]       (3711 S)
 [12] count:    2,  width: 3968 us [3836;4100]  ( 992 S)
 [13] count:    1,  width:  164 us [164;164]    (  41 S)
Gap width distribution:
 [ 0] count:   16,  width:   48 us [44;56]      (  12 S)
 [ 1] count:    8,  width:   60 us [60;72]      (  15 S)
 [ 2] count:    1,  width:   80 us [80;80]      (  20 S)
 [ 3] count:    1,  width:  108 us [108;108]    (  27 S)
 [ 4] count:    1,  width: 94044 us [94044;94044]       (23511 S)
 [ 5] count:   22,  width: 1564 us [1560;1572]  ( 391 S)
 [ 6] count:  604,  width: 1048 us [1040;1056]  ( 262 S)
 [ 7] count:  525,  width:  528 us [512;540]    ( 132 S)
 [ 8] count:   21,  width: 18936 us [18932;18944]       (4734 S)
Pulse period distribution:
 [ 0] count:    1,  width:  496 us [496;496]    ( 124 S)
 [ 1] count:   23,  width: 3120 us [2932;3136]  ( 780 S)
 [ 2] count:    2,  width: 7692 us [7348;8036]  (1923 S)
 [ 3] count:    4,  width: 5348 us [4652;5852]  (1337 S)
 [ 4] count:    1,  width: 1980 us [1980;1980]  ( 495 S)
 [ 5] count:    3,  width: 1204 us [1144;1264]  ( 301 S)
 [ 6] count:    1,  width: 11452 us [11452;11452]       (2863 S)
 [ 7] count:    1,  width: 67412 us [67412;67412]       (16853 S)
 [ 8] count:   24,  width: 20696 us [19896;24296]       (5174 S)
 [ 9] count:    3,  width: 49664 us [43140;53648]       (12416 S)
 [10] count:    2,  width: 31008 us [30016;32004]       (7752 S)
 [11] count:    1,  width: 14896 us [14896;14896]       (3724 S)
 [12] count:    2,  width: 4040 us [3916;4164]  (1010 S)
 [13] count:    1,  width:  216 us [216;216]    (  54 S)
 [14] count:    1,  width: 104560 us [104560;104560]    (26140 S)
 [15] count: 1129,  width: 1576 us [1560;1584]  ( 394 S)
Level estimates [high, low]:  15969,      4
RSSI: -0.1 dB SNR: 35.0 dB Noise: -35.2 dB
Frequency offsets [F1, F2]:   19982,      0     (+76.2 kHz, +0.0 kHz)
Guessing modulation: No clue...

Detected OOK package    2019-04-28 19:00:15
Analyzing pulses...
Total count:   15,  width: 23.64 ms             ( 5910 S)
Pulse width distribution:
 [ 0] count:    9,  width:  524 us [524;532]    ( 131 S)
 [ 1] count:    5,  width: 1044 us [1044;1048]  ( 261 S)
 [ 2] count:    1,  width: 1568 us [1568;1568]  ( 392 S)
Gap width distribution:
 [ 0] count:    9,  width: 1048 us [1048;1056]  ( 262 S)
 [ 1] count:    5,  width:  528 us [516;532]    ( 132 S)
Pulse period distribution:
 [ 0] count:   14,  width: 1576 us [1560;1580]  ( 394 S)
Level estimates [high, low]:  15862,      4
RSSI: -0.1 dB SNR: 35.0 dB Noise: -35.2 dB
Frequency offsets [F1, F2]:   18806,      0     (+71.7 kHz, +0.0 kHz)
Guessing modulation: Pulse Width Modulation with sync/delimiter
Attempting demodulation... short_width: 524, long_width: 1044, reset_limit: 1060, sync_width: 1568
Use a flex decoder with -X 'n=name,m=OOK_PWM,s=524,l=1044,r=1060,g=0,t=0,y=1568'
pulse_demod_pwm(): Analyzer Device
bitbuffer:: Number of rows: 2
[00] {14} dd 58     : 11011101 010110
[01] { 0}           :

Detected OOK package    2019-04-28 19:00:15
Analyzing pulses...
Total count:  810,  width: 1568.98 ms           (392245 S)
Pulse width distribution:
 [ 0] count:    1,  width: 4808 us [4808;4808]  (1202 S)
 [ 1] count:  419,  width:  524 us [520;540]    ( 131 S)
 [ 2] count:  361,  width: 1044 us [1040;1056]  ( 261 S)
 [ 3] count:   29,  width: 1560 us [1556;1568]  ( 390 S)
Gap width distribution:
 [ 0] count:   15,  width: 1564 us [1564;1572]  ( 391 S)
 [ 1] count:  419,  width: 1048 us [1040;1056]  ( 262 S)
 [ 2] count:  361,  width:  528 us [516;540]    ( 132 S)
 [ 3] count:   14,  width: 18932 us [18928;18940]       (4733 S)
Pulse period distribution:
 [ 0] count:    1,  width: 6376 us [6376;6376]  (1594 S)
 [ 1] count:  780,  width: 1576 us [1560;1584]  ( 394 S)
 [ 2] count:   14,  width: 20496 us [20496;20504]       (5124 S)
 [ 3] count:   14,  width: 3128 us [3128;3132]  ( 782 S)
Level estimates [high, low]:  15883,      7
RSSI: -0.1 dB SNR: 33.0 dB Noise: -33.1 dB
Frequency offsets [F1, F2]:   19192,      0     (+73.2 kHz, +0.0 kHz)
Guessing modulation: Pulse Code Modulation (Not Return to Zero)
Attempting demodulation... short_width: 524, long_width: 524, reset_limit: 536576, sync_width: 0
Use a flex decoder with -X 'n=name,m=FSK_PCM,s=524,l=524,r=536576'
```

Based on a few measurements and some trial and error, I came up with the following parameters:

```-X 'n=name,m=OOK_PWM,s=500,l=1020,r=94044,g=1576,y=1546'```

The reset value is that large because I noticed that there's always exactly 1 pulse of this length in every analysis. I guess this is transmitted when I release the pressed button.

Sample output using the decoder:

```
time      : 2019-04-28 19:10:53
model     : remote       count     : 6             num_rows  : 6             rows      :
len       : 52           data      : e086000e702e3,
len       : 52           data      : e386000e702e3,
len       : 52           data      : e286000e702e3,
len       : 52           data      : e586000e702e3,
len       : 52           data      : e486000e702e3,
len       : 0            data      :
codes     : {52}e086000e702e3, {52}e386000e702e3, {52}e286000e702e3, {52}e586000e702e3, {52}e486000e702e3, {0}
Modulation: ASK          Freq      : 434.0 MHz
RSSI      : -0.1 dB      SNR       : 24.6 dB       Noise     : -24.7 dB

time      : 2019-04-28 19:12:02
model     : remote       count     : 6             num_rows  : 6             rows      :
len       : 52           data      : e0a40b1957091,
len       : 52           data      : e3a40b1957091,
len       : 52           data      : e2a40b1957091,
len       : 52           data      : e5a40b1957091,
len       : 52           data      : e4a40b1957091,
len       : 0            data      :
codes     : {52}e0a40b1957091, {52}e3a40b1957091, {52}e2a40b1957091, {52}e5a40b1957091, {52}e4a40b1957091, {0}
Modulation: ASK          Freq      : 434.0 MHz
RSSI      : -0.1 dB      SNR       : 33.0 dB       Noise     : -33.1 dB
```

Now my problem is the patterns in the decoded data. Forgive me if there's something really obvious here, I'm still learning the basics of SDR :) So as you can see, in each decoded message, the rows are the same *except* the second (hexadecimal) character. And there's always a sequence of characters `0, 3, 2, 5, 4` in every message (at least for the main button of the remote, the secondary button produces `3, 0, 1, 6, 7`). First of all, is this normal, or did I just mess up the decoder parameters? Second, is it possible to write a decoder that matches this strange pattern? My main goal is to filter out any other remote devices in my neighbourhood, and to differentiate between the primary and secondary buttons' messages.

Thanks for the help in advance!

zuckschwerdt, 2019-04-29:
Yes, looks like PWM with fixed bit length (see highest count fields in the three sections). But the signal seems awefully messy. You need to check the spectrogram (http://triq.net/iqs ) and the demodulation (sigrok-open.sh). The reset can't be that big, 20 times the long value is max, I think. You do not have sync symbols, drop the `y=`. Gap limit should likely be 1300, reset 19000. The result looks ok though.
Use e.g. `match=a40b1957091` to filter.

ofalvai, 2019-05-04:
Thanks for the suggestions! Yes, I know that the signal is messy, decoding other devices is unreliable most of the time. I guess I'll need to buy a better antenna.

closed 2019-05-04 by ofalvai