merbanan/rtl_433 #1056: Digoo DG-R8B no longer read humidity

Fredy72, 2019-05-09:
I have sensors Digoo DG-R8B (Nexus-T) and Digoo DG-R8H (Nexus-TH).
Digoo DG-R8B not read humidity.
Digoo DG-R8H work fine.

Using another way https://github.com/aquaticus/nexus433 they are both reading.

merbanan, 2019-05-09:
Hi, submit a sample signal from the Digoo DG-R8B sensor that is missing the humidity value. Then we can fix the issue.

Fredy72, 2019-05-10:
Hi, you mean files cu8?

[1_433.92M_250k.cu8.txt](https://github.com/merbanan/rtl_433/files/3166943/1_433.92M_250k.cu8.txt)
[2_433.92M_250k.cu8.txt](https://github.com/merbanan/rtl_433/files/3166944/2_433.92M_250k.cu8.txt)

Fredy72, 2019-05-10:
For example, I tried the git version e603dc5c63a0ef0f60747ecbe90a1b63a20c1acf and it works, temperature and humidity on Digoo DG-R8B and Digoo DG-R8H.

merbanan, 2019-05-10:
Can you post example output ?

Fredy72, 2019-05-10:
I have also tried the versions (changes for nexus.c) and the last one is functional 0a940ce0162c17fa844fb4f7cd3635c9187bf207 , the next 4d060eb1f337f8857143ad9656fe501e5261ea44 is not working.

Fredy72, 2019-05-10:
I have also tried the versions (changes for nexus.c) and the last one is functional d01b128042f017025f83be29e5fbae432b970ad5 , the next 0a940ce0162c17fa844fb4f7cd3635c9187bf207 is not working.

[rtl_433_0a940ce0162c17fa844fb4f7cd3635c9187bf207.txt](https://github.com/merbanan/rtl_433/files/3167137/rtl_433_0a940ce0162c17fa844fb4f7cd3635c9187bf207.txt)
[rtl_433_d01b128042f017025f83be29e5fbae432b970ad5.txt](https://github.com/merbanan/rtl_433/files/3167138/rtl_433_d01b128042f017025f83be29e5fbae432b970ad5.txt)

zuckschwerdt, 2019-05-10:
Thanks for the bisect. 0a940ce doesn't really change Nexus. And d01b128 is 3 years old. Can you find a more recent still working version? There were only [12 commits touching Nexus](https://github.com/merbanan/rtl_433/commits/master/src/devices/nexus.c), most those wouldn't change anything. I'd say c1537c1 is the only real change. But perhaps it's a general demod problem not specific to the decoder.

Fredy72, 2019-05-10:
I will try another version, it will take a while.

Fredy72, 2019-05-10:
Result:
d01b128042f017025f83be29e5fbae432b970ad5 it looks like it works
9e6d99cd6a50a77181ff11722669ba5eae14d1d4 not work (next in nexus.c, but more than year)

[rtl_433_9e6d99cd6a50a77181ff11722669ba5eae14d1d4.txt](https://github.com/merbanan/rtl_433/files/3167376/rtl_433_9e6d99cd6a50a77181ff11722669ba5eae14d1d4.txt)
[rtl_433_d01b128042f017025f83be29e5fbae432b970ad5.txt](https://github.com/merbanan/rtl_433/files/3167377/rtl_433_d01b128042f017025f83be29e5fbae432b970ad5.txt)

merbanan, 2019-05-10:
rtl_433 -vv -r ~/Downloads/2_433.92M_250k.cu8.txt ->
[00] {36} 70 80 da f0 00 : 01110000 10000000 11011010 11110000 0000

rtl_433 -vv -r ~/Downloads/1_433.92M_250k.cu8.txt ->
[00] {36} 70 80 dc f0 00 : 01110000 10000000 11011100 11110000 0000

As you see here the last 2 nibbles are 0s. That means there are not hygrometer present. When that happens the output doesn't contain the humidity member. So everything works as it should.

Fredy72, 2019-05-10:
But the 2 Digoo DG-R8Bs contain a hygrometer and will be displayed in older versions.

merbanan, 2019-05-10:
Well provide a sample signal that triggers not showing the hygrometer value and contains a non zero value for it.

https://github.com/merbanan/rtl_433/blob/master/src/devices/nexus.c#L77 contains the logic. If the value is 0 the hygrometer is counted as not present.

I think you have trouble in sorting out where the signal comes from. Your output is riddled with lots of other sensor. Try removing the antenna and placing the sensor close by, that should get rid of the "noise".

Fredy72, 2019-05-10:
I turned off the Digoo DG-R8H sensors and left only the Digoo DG-R8B and the report is correct. The ones that don't show up are not mine, but probably the neighbors, and they're probably not the Digoo DG-R8B. Sorry for reporting a bug. Thank you and I close. Have a nice weekend.

[rtl_433_aktual.txt](https://github.com/merbanan/rtl_433/files/3167475/rtl_433_aktual.txt)

closed 2019-05-10 by Fredy72