merbanan/rtl_433 #1059: Cheap Chinese TPMS

Tomazzzzz, 2019-05-16:
Hello guys, I want to build TPMS for my bike.
I have found some cheap chines TPMS for 8$
I have found FCCID for that model(s) https://fccid.io/2ANZATP620
I have opend one TPMS and there was a chip SP372 2315.
Using Universal Radio Hacking and RTL-SDR I managed to read the data packages.

Now the fun begins.
I have 3 sensors. I put them to pressure chamber all of them and read packages transmitted.
So far so good.
But the data seems to be encrypted :O
All starts with same 2 bytes. B3 5A
All ends with CRC width=16  poly=0x1021  init=0xd757  refin=false  refout=false  xorout=0x0000  check=0x3f81  residue=0x0000.

There are 5 byte of payload where is ID + pressure + temperature+ battery status.

In pressure chamber the ID + battery status should match and maybe temperature.
The only that should vary is pressure.

All 3 sensors transmit different payload at the same pressure. If pressure does not changes the same sensor transmits the same data but data are different between sensors.

It seems that here are some brilliant minds with a lot of practice in this area.
I would be glad if some one could help to sort out the protocol and these sensors to be added to rtl_433 program.
 
[TPMS 2 sensors.txt](https://github.com/merbanan/rtl_433/files/3188868/TPMS.2.sensors.txt)

THNX!!!

klohner, 2019-05-16:
Hi @Tomazzzzz - Would you be able to provide some .cu8 samples from these devices, each with a description identifying which of your 3 sensor devices generated the signal (is there an ID# on each device?), and the pressure/temperature reading that the receiver shows for each signal?  Also, can you provide a photo of the device and a URL you used to purchase the device?

Is the device something like this: https://www.ebay.com/itm/183813439094 ("Car TPMS Wireless Tire Pressure Monitoring System LCD + 4 External Sensors")

I tried putting the data in your text file into this [BitBench](http://triq.net/bitbench?c=b35a04299f695c0f63&c=b35a2e0feda396665f&c=b35a35dacebaab1b3f&c=b35a3b04f8a0a1e439&c=b35a53ece0a8992ad1&c=b35a54f90f99ac36b5&c=b35a5e5d095359a419&c=b35a759a0ebaeb0af8&c=b35a92ab21a7daf5d7&c=b35a94b94f99ec3ef8&c=b35aa9c6324ebff988&c=b35ac4695f691c3c74&c=b35ae986724effd315&c=%5B%5D&f=PREFIX%3A16h%208b%208b%208b%208b%208b%208b%20CRC%3A8h) but can't make too much sense of it.

zuckschwerdt, 2019-05-17:
@enavarro222 has some similar problems (suspected encryption) in #856

Tomazzzzz, 2019-05-17:
Hello @klohner, in BitBech you specify CRC as 8bit but it is 16bit.
The sensors looks like these [ebay](https://www.ebay.com/itm/1Pc-Black-Auto-TPMS-Tire-Parts-Pressure-External-Sensor-Monitor-System-for-TP620/192690421969?epid=4024819224&hash=item2cdd3e80d1:g:oPQAAOSwSc1bxDfv).
There are some clues, like:
Operating temperature: -30 - 80 degrees Celsius
Temperature resolution: plus or minus 2 degrees Celsius
Pressure range: 0 - 8.0Bar
Pressure resolution: plus or minus 0.1Bar

So it needs at least  5 bit for temperature and 7bit for pressure.
Plus some status for battery.
In any way out of 5 byte payload the 2 or 3 should be sensor values. 
Other 2-3 should have ID and encryption key or something.

Also rolling code is excluded. After many times removing and installing battery the package remains the same.

Tomazzzzz, 2019-09-01:
Hello guys,
I have captured a lot of packages for the same sensor.
There are 2 files attached. One with many different pressures. And one was recording with no pressure.
In the firs file there may be data from another sensor. It was away but I saw weak signal recorded.
One note, first column minus fifth always produce 0x?5.

[FSK_Same_Sensor_Diff_Presure.txt](https://github.com/merbanan/rtl_433/files/3563375/FSK_Same_Sensor_Diff_Presure.txt)
[FSK_Same_Sensor_0_Presure.txt](https://github.com/merbanan/rtl_433/files/3563376/FSK_Same_Sensor_0_Presure.txt)

zuckschwerdt, 2019-09-04:
Especially the 0-pressure should be interesting. Try things like taking the first byte and add, substract, or XOR it to all bytes (based on your 0x?5 observation), or other similar schemes. If you get it right a (mostly) fixed pressure field should be visible and also a temperature field that slowly changes.

sblakegit, 2019-09-16:
I'm interested in this too.  
I sourced https://www.amazon.co.uk/gp/product/B07MMZ9GRS/ref=ppx_yo_dt_b_asin_title_o00_s00?ie=UTF8&psc=1 
which looks pretty identical.
I had sourced them since it appeared that similar had been reported as working for home-brew pressure monitoring.  https://www.rtl-sdr.com/tag/home-brew/

I haven't attempted to capture any packets yet.  My airwaves are busy with WH2A sensors and i will need to go somewhere quiet.  My rig is working in that i am capturing data from the traffic going past the front of the house, just not my sensors.
Steve

zuckschwerdt, 2019-09-16:
For the Zeepin the checksum is still missing. Looks like a weird digest scheme but no luck so far.
Andreas supplied a lot of data points, but getting a checksum brute-force needs a really big draw of unique codes… and plenty of time.

To limit reception to a single sensor: remove the antenna and place the device right next to the receiver (10 cm).

Tomazzzzz, 2020-01-13:
Hi, I have some patter over here.
I have captured 13 different packages at 0 pressure. So only difference is temperature.
First 5 are bytes captured, next Bx1 are first nibble and Bx2 are the second.
As you can see B21 & B41 are stable 0xB & 0x9.
Lines (3,4), (5,6,7), (8,9),(12,13) have same second nibbles.
Lines (1,4),(2,9),(7,10) have same first nibbles.
So there is definitely a pattern. I just cannot see it.
My theory is that only 2 values are changed over time (pressure & temp).
So if you XOR these 2 values and then XOR the result with all bytes again, it should bring back one of values.

So, I believe that  B & 9 should be temperature value(since pressure not changed).
But, 0x9B is too big for 19 C.

We have 5 bytes the should be 1 pressure,1 temperature, 1 XOR and there are 2 for ID.
And ID is stable. 2 bytes for ID it is only 65k unique IDs :(

	B1	B2	B3	B4	B5		B11	B21	B31	B41	B51		B12	B22	B32	B42	B52
1	B4	B9	5F	99	FC		0B	0B	05	09	0F		04	09	0F	09	0C
2	C6	BD	61	9B	FA		0C	0B	06	09	0F		06	0D	01	0B	0A
3	97	BB	4E	9C	EB		09	0B	04	09	0E		07	0B	0E	0C	0B
4	B7	BB	5E	9C	FB		0B	0B	05	09	0F		07	0B	0E	0C	0B
5	A8	B1	33	9D	D8		0A	0B	03	09	0D		08	01	03	0D	08
6	A8	B1	53	9D	F8		0A	0B	05	09	0F		08	01	03	0D	08
7	C8	B1	63	9D	08		0C	0B	06	09	00		08	01	03	0D	08
8	A9	BF	50	9E	E9		0A	0B	05	09	0E		09	0F	00	0E	09
9	C9	BF	60	9E	F9		0C	0B	06	09	0F		09	0F	00	0E	09
10	CA	B5	65	9F	06		0C	0B	06	09	00		0A	05	05	0F	06
11	CB	B3	72	90	07		0C	0B	07	09	00		0B	03	02	00	07
12	AD	B7	44	92	D5		0A	0B	04	09	0D		0D	07	04	02	05
13	AD	B7	64	92	F5		0A	0B	06	09	0F		0D	07	04	02	05

Tomazzzzz, 2020-01-13:
Also I have captured many packages with different pressure and temperatures.
There some values missing in nibbles. Bellow are values seen and missing.
My pressure is low so may be I just did not give enough pressure...

```
B11 1,5-F (2,3,4 missing)
B12 0-F

B21 3,5-C (0,1,2,4,D,E,F missing)
B22 0-F

B31 0-C E-F (D)
B32 0-F

B41 0-1,4-B (2,3,C,D,E,F missing)
B42 0-F

B51 0-4,6,B-F  (5,7,8,9,A missing)
B51 0-F
```

zuckschwerdt, 2020-01-14:
Very interesting progress!
A [BitBench for this data](http://triq.net/bitbench?c=B4B95F99FC&c=C6BD619BFA&c=97BB4E9CEB&c=B7BB5E9CFB&c=A8B1339DD8&c=A8B1539DF8&c=C8B1639D08&c=A9BF509EE9&c=C9BF609EF9&c=CAB5659F06&c=CBB3729007&c=ADB74492D5&c=ADB76492F5&f=8h%20%0Ah4x.%20%0A.4xh%20%0A.), just to show how we can get the nibbles.
If you use `b` or `b` format you can see pairs with only a few bits difference. If it is XOR-based then xoring those pairs should reaveal something.
But if too much changes you can't really find it.

As you say, the info in the 5 bytes would be: ID, pressure, temperature, status and then maybe a key or counter. If the same pressure+temp always generates identical codes then there is likely no counter.
The pressure and temp can be encoded with a single byte, status can be 2 bits (alarm and battery), ID should really be more than one byte. That leaves just a nibble or byte for the key/counter, maybe.

If temperature has a resolution of 1°C you would ideally grab a single sensor at 0-pressure, cool the sensor in the freezer (say -18°C) and let it warm up slowly to room temperature (20°C) -- that should give you ~38 ordered values with probably only a single data byte incrementing.
The codes should be interesting.

Tomazzzzz, 2020-01-14:
Hello, the issue is that sensor transmit every 6 minutes. I need a way to control temperature.
I will try to capture more packages and keep eye on the temperature.

zuckschwerdt, 2020-01-14:
Maybe some thermal insulation of buffering. The important thing to watch out for is that you get each packet multiple times so you can know that each different packet is exactly one step in the temperature reading (likely 1°C).

Tomazzzzz, 2020-01-15:
Ok, today was reading sensor data over a desk lamp and results are bellow.
As u can see the last column steady grows till rich some temperature ant then stays there.

```
cf bb 76 94 03
cc b9 77 91 04
cd b7 74 92 05
ca b5 65 9f 06
cb b3 72 90 07
c8 b1 63 9d 08
c9 bf 60 9e f9
c6 bd 61 9b fa
c9 bf 60 9e f9
b7 bb 5e 9c fb
b5 b7 5c 9a fd

```
After I put my sensor in cup of hot water and got result bellow.
Pressure was 0 all time. Again last column goes down.

```
ba c5 55 8f e6
ba c5 55 8f e6
a7 cb 4e 8c db
a5 c7 4c 8a dd
a4 c9 4f 89 dc
b6 cd 51 8b da
b8 c1 53 8d e8
```

zuckschwerdt, 2020-01-15:
Great progress! Heating up one sensor it goes 03 ... 08 then f9 ...fd ?
I have see this pattern before (random +/- switches), it was Differential Manchester actually. But that might just be an accident of the encryption here. The CRC always works, right?

Tomazzzzz, 2020-01-15:
CRC is :
> `width=16  poly=0x1021  init=0xd757  refin=false  refout=false  xorout=0x0000  check=0x3f81  residue=0x0000  name=(none)`
and whole packages are:

```
b3 5a cf bb 76 94 03 0b f4
b3 5a cc b9 77 91 04 b0 6c
b3 5a cd b7 74 92 05 a4 45
b3 5a ca b5 65 9f 06 1c 95
b3 5a cb b3 72 90 07 57 b1
b3 5a c8 b1 63 9d 08 a7 eb
b3 5a c9 bf 60 9e f9 5c dd
b3 5a c6 bd 61 9b fa 2c ea
b3 5a c9 bf 60 9e f9 5c dd
b3 5a b7 bb 5e 9c fb dc 86
b3 5a b5 b7 5c 9a fd 73 37
```

Tomazzzzz, 2020-01-15:
Some more:
Second nibble of first byte (B12) and second nibble of fifth byte (B52) are related as:
9-9, 8-8, 7-B, B-7, 6-A, A-6, 5-D, D-5, 4-C, C-4.
c**d** b7 74 92 0**5**
b**5** b7 5c 9a f**d**
c**6** bd 61 9b f**a**
c**a** b5 65 9f 0**6**

First nibble of second byte (B21) and first nibble of forth byte (B41) in sum is 0x14.

B42 - B12 = 0x05 or 0xF5 if overflow.
B12 - B42 = 0x0B or 0xFB if overflow.

Tomazzzzz, 2020-01-23:
Hello, so good news.
I captured many packages and I have found pattern.

All packages are at 0 pressure and no alarm bit flags. I have with alarm bit flag and there are 2-3 nibbles that are changed. etc:
```
b3 5a a6 bd 51 9b ea e2 a2(21 C alarm)
b3 5a c6 bd 61 9b fa 2c ea

b3 5a b6 cd 61 8b ea 62 5b (alarm)
b3 5a b6 cd 51 8b da 91 ad
```

I have created a program where I can do some complex calculations.
In columns C15|C16=C17-0xA5=C18 is temperature C18 is in Celsius. This is confirmed with other sensor I have. 

C7 & C8 I cannot detect any pattern, except:
```
										2	2	.	2	2	2	2
7	7	7	6	7	6	6	6		5	5	5	5	5	5	5	5
6	6	6	5	6	5	5	5		4	4	4	4	4	4	4	4
1	1	1	.	.	0	.	.		F

										8	8	.	8	8	8	7
c	c	c	c	c	c	c	c		b	b	b	b	b	b	b	a
b	b	b	b	b	b	b	b		a	a	a	a	a	a	a	9
6	6	6	6	.	.	6	.		.	5
```
Also C19=C7-C8 and it will be 6 or 5.

C13&C12  or/and C13&C14 are counter. Also C11 & C9 are counters and C10 is also counter that increases each 16 values (hi nibble)

Bellow are packages (C1..C5)
C6..C19 are calculation columns:

```
C6=(C4 & 0xF) ^ 0x0E
C7=((C1 & 0xF0)>>4)
C8=((C3 & 0xF0)>>4)
C9=(C3 & 0xF) ^ 0x0E
C10=((C4 & 0xF0)>>4) ^ 0x0D
C11=(C1 & 0xF) ^ 0x0E
C12=(C6%2==0x01? ((C6 - 0x04)&0x0F) : C6)
C13=(C5 & 0xF0)>>4
C14=(C2& 0xF)
C15=(C2& 0xF0)>>4
C16=(C5 & 0xF)
C17=C15<<4 | C16
C18=C17-0xA5
C19=C7-C8
```

```
C1	C2	C2	C4	C5	C6	C7	C8	C9	C10	C11	C12	C13	C14	C15	C16	C17	C18	C19
84	A9	2F	E9	1C	07	08	02	01	03	0A	03	01	09	0A	0C	AC	07	06
85	A7	2C	EA	1D	04	08	02	02	03	0B	04	01	07	0A	0D	AD	08	06
83	A3	2A	E8	1F	06	08	02	04	03	0D	06	01	03	0A	0F	AF	0A	06
80	B1	2B	E5	10	0B	08	02	05	03	0E	07	01	01	0B	00	B0	0B	06
81	BF	28	E6	01	08	08	02	06	03	0F	08	00	0F	0B	01	B1	0C	06
7E	BD	29	E3	02	0D	07	02	07	03	00	09	00	0D	0B	02	B2	0D	05
CF	BB	76	94	03	0A	0C	07	08	04	01	0A	00	0B	0B	03	B3	0E	05
CC	B9	77	91	04	0F	0C	07	09	04	02	0B	00	09	0B	04	B4	0F	05
CD	B7	74	92	05	0C	0C	07	0A	04	03	0C	00	07	0B	05	B5	10	05
CA	B5	65	9F	06	01	0C	06	0B	04	04	0D	00	05	0B	06	B6	11	06
CB	B3	72	90	07	0E	0C	07	0C	04	05	0E	00	03	0B	07	B7	12	05
C8	B1	63	9D	08	03	0C	06	0D	04	06	0F	00	01	0B	08	B8	13	06
C9	BF	60	9E	F9	00	0C	06	0E	04	07	00	0F	0F	0B	09	B9	14	06
C6	BD	61	9B	FA	05	0C	06	0F	04	08	01	0F	0D	0B	0A	BA	15	06
B7	BB	5E	9C	FB	02	0B	05	00	04	09	02	0F	0B	0B	0B	BB	16	06
B4	B9	5F	99	FC	07	0B	05	01	04	0A	03	0F	09	0B	0C	BC	17	06
B5	B7	5C	9A	FD	04	0B	05	02	04	0B	04	0F	07	0B	0D	BD	18	06
B2	B5	5D	97	FE	09	0B	05	03	04	0C	05	0F	05	0B	0E	BE	19	06
B3	B3	5A	98	FF	06	0B	05	04	04	0D	06	0F	03	0B	0F	BF	1A	06
B0	C1	5B	95	F0	0B	0B	05	05	04	0E	07	0F	01	0C	00	C0	1B	06
B1	CF	58	96	E1	08	0B	05	06	04	0F	08	0E	0F	0C	01	C1	1C	06
AE	CD	59	93	E2	0D	0A	05	07	04	00	09	0E	0D	0C	02	C2	1D	05
BF	CB	66	84	E3	0A	0B	06	08	05	01	0A	0E	0B	0C	03	C3	1E	05
BC	C9	67	81	E4	0F	0B	06	09	05	02	0B	0E	09	0C	04	C4	1F	05
BD	C7	64	82	E5	0C	0B	06	0A	05	03	0C	0E	07	0C	05	C5	20	05
BA	C5	55	8F	E6	01	0B	05	0B	05	04	0D	0E	05	0C	06	C6	21	06
BB	C3	62	80	E7	0E	0B	06	0C	05	05	0E	0E	03	0C	07	C7	22	05
B8	C1	53	8D	E8	03	0B	05	0D	05	06	0F	0E	01	0C	08	C8	23	06
B9	CF	50	8E	D9	00	0B	05	0E	05	07	00	0D	0F	0C	09	C9	24	06
B6	CD	51	8B	DA	05	0B	05	0F	05	08	01	0D	0D	0C	0A	CA	25	06
A7	CB	4E	8C	DB	02	0A	04	00	05	09	02	0D	0B	0C	0B	CB	26	06
A4	C9	4F	89	DC	07	0A	04	01	05	0A	03	0D	09	0C	0C	CC	27	06
A5	C7	4C	8A	DD	04	0A	04	02	05	0B	04	0D	07	0C	0D	CD	28	06
A2	C5	4D	87	DE	09	0A	04	03	05	0C	05	0D	05	0C	0E	CE	29	06
A3	C3	4A	88	DF	06	0A	04	04	05	0D	06	0D	03	0C	0F	CF	2A	06
A0	D1	4B	85	D0	0B	0A	04	05	05	0E	07	0D	01	0D	00	D0	2B	06
A1	DF	48	86	C1	08	0A	04	06	05	0F	08	0C	0F	0D	01	D1	2C	06
9E	DD	49	83	C2	0D	09	04	07	05	00	09	0C	0D	0D	02	D2	2D	05
6F	DB	16	B4	C3	0A	06	01	08	06	01	0A	0C	0B	0D	03	D3	2E	05
6C	D9	17	B1	C4	0F	06	01	09	06	02	0B	0C	09	0D	04	D4	2F	05
6D	D7	14	B2	C5	0C	06	01	0A	06	03	0C	0C	07	0D	05	D5	30	05
68	D1	03	BD	C8	03	06	00	0D	06	06	0F	0C	01	0D	08	D8	33	06
57	DB	FE	BC	BB	02	05	0F	00	06	09	02	0B	0B	0D	0B	DB	36	F6
```

merbanan, 2020-01-24:
I have seen other protocols permuting bits so C17 being the temperature sounds ok. And having a bias also makes sense, but the bias being 0xA5 sounds totally fabricated. If that is the case then other parameters might also have this pattern.

gdt, 2023-09-30:
@Tomazzzzz What's the status of this and path forward?

gdt, 2025-07-04:
feedback timeout

closed 2025-07-04 by gdt