merbanan/rtl_433 #1060: ecowitt wh51 soil moisture sensor

magnusboman, 2019-05-17:
I recently bought a couple of ecowitt wh51 soil moisture sensors and was hoping to get their values via rtl_433 but it didn't out of the box.
I'm attaching 4 traces. 2 of them when dry and two when immersed in water. The device ID of the sensor is 00C4F2
Specification states moisture range 0-100%, 1% resolution.
I'm new to this so not sure on how to go about to figure the traces out so would be awesome if someone could help out

[wh51.zip](https://github.com/merbanan/rtl_433/files/3189479/wh51.zip)

zuckschwerdt, 2019-05-17:
You shouldn't remove the metadata about frequency and sample rate from the filename. I'm guessing you used the defaults?
The signal is 2-ASK PCM. ASK is not well supported currently.
@merbanan a good test case for the new demod?

magnusboman, 2019-05-17:
Whops, sorry, didn't realise. Yeah, I used defaults and filenames were g00x_433M_250k.cu8.

merbanan, 2019-05-17:
@zuckschwerdt well I didn't really think of adding ASK support but now with another sensor that is a 100% increase of sensors that currently can't be decoded. Doing a min/max tracker decision on a filtered I^2+Q^2 envelope should be able to handle both OOK and ASK. Need to prototype it to see if it works well.

magnusboman, 2019-05-18:
I've attached some more recordings using 3 different devices. Each directory indicates the DeviceID.
The first recording is when I insert the battery and the rest are "normal". All recordings are when dry.
[wh51.zip](https://github.com/merbanan/rtl_433/files/3193477/wh51.zip)

merbanan, 2019-05-24:
I had a look at the samples and they use both ASK and FSK modulation. Are the devices different models? Or do you have other device in the proximity?

magnusboman, 2019-05-24:
I removed the antenna from the receiver and put the devices 10 cm from it to try to isolate everything but maybe it still picked some other device up. I'll try to turn off all other devices around the place and redo the samples.

magnusboman, 2019-05-24:
I turned off all other radio devices and then did a spectrum analyse for a couple of minutes and nothing else is coming through.

```
rtl_433 -f 433872000 -S all
rtl_433 version 18.12-207-g82d0f55 branch master at 201905061642 inputs file rtl_tcp RTL-SDR
Trying conf file at "rtl_433.conf"...
Trying conf file at "/home/xyz/.config/rtl_433/rtl_433.conf"...
Trying conf file at "/usr/local/etc/rtl_433/rtl_433.conf"...
Trying conf file at "/etc/rtl_433/rtl_433.conf"...

	Consider using "-M newmodel" to transition to new model keys. This will become the default someday.
	A table of changes and discussion is at https://github.com/merbanan/rtl_433/pull/986.

Registered 98 out of 125 device decoding protocols [ 1-4 8 11-12 15-17 19-21 23 25-26 29-36 38-60 62-63 67-71 73-100 102-103 108-116 119 121 124-125 ]
Detached kernel driver
Found Rafael Micro R820T tuner
Exact sample rate is: 250000.000414 Hz
[R82XX] PLL not locked!
Sample rate set to 250000 S/s.
Tuner gain set to Auto.
Tuned to 433.872MHz.
pulse_FSK_detect(): Maximum number of pulses reached!
*** Saving signal to file g001_433.872M_250k.cu8 (773642 samples, 1572864 bytes)
*** Saving signal to file g002_433.872M_250k.cu8 (45192 samples, 131072 bytes)
*** Saving signal to file g003_433.872M_250k.cu8 (35934 samples, 131072 bytes)
*** Saving signal to file g004_433.872M_250k.cu8 (45191 samples, 131072 bytes)
*** Saving signal to file g005_433.872M_250k.cu8 (35937 samples, 131072 bytes)
*** Saving signal to file g006_433.872M_250k.cu8 (45192 samples, 131072 bytes)
*** Saving signal to file g007_433.872M_250k.cu8 (35936 samples, 131072 bytes)
*** Saving signal to file g008_433.872M_250k.cu8 (45189 samples, 131072 bytes)
*** Saving signal to file g009_433.872M_250k.cu8 (35935 samples, 131072 bytes)
*** Saving signal to file g010_433.872M_250k.cu8 (45190 samples, 131072 bytes)
*** Saving signal to file g011_433.872M_250k.cu8 (35934 samples, 131072 bytes)
*** Saving signal to file g012_433.872M_250k.cu8 (45190 samples, 131072 bytes)

```
Each time the LED on the device flash corresponds to when it saves data to a new file.

[00C4F2.zip](https://github.com/merbanan/rtl_433/files/3218750/00C4F2.zip)

merbanan, 2019-05-25:
All these samples are FSK modulated. I am guessing they are using PCM. With that I can use the following flex syntax:

rtl_433 -X "n=soil_sensor,m=FSK_PCM,s=56,l=56,r=5000,g=4000" -r  wh51/00C4F2/00C4F2/g004_433.872M_250k.cu8

The samples are clipping though so that might be an issue.

Anyway running this on a few samples I get:
45ba8a30189e49efe00f819
45ba8a20189e45efe00f819
45ba8a20199e41efe00f819
45ba8a20189e41efe01f033
45ba8a20189e42f7f007c0c

merbanan, 2019-05-25:
This is made by Fine Offset. The WH25 uses FSK PCM encoding with 58us bits (so they use the same transmission logic and most likely the same encoding logic).

aa2ddc5100c4f20f7f08f819 is also a bitstream found

So looking at the Fine Offset code I see this:

Data layout:
aa 2d dx xx .....

So we are on the right track. With some samples that are not clipping together with official readings we can start matching the CRC and values.

merbanan, 2019-05-25:
Use examples/sigrok-open.sh wh51/00C4F0/g008_433.872M_250k.cu8 and look at the I and Q channel. They should not clip. Clipping will cause decoding artifacts. Use the -g option and set a sufficiently low gain. Then post new samples that has corresponding readings. After that the fun stuff with figuring out the payload starts.

magnusboman, 2019-05-25:
Samples using `-g 50`

[gain50.zip](https://github.com/merbanan/rtl_433/files/3219773/gain50.zip)

roger-, 2019-11-02:
Any progress with this?

0x4d0x44, 2019-12-28:
I bought a couple of compatible sensors (available on Aliexpress/Ebay/Amazon as misol/1 Soil Moisture Sensor) and discovered that rtl_433 had no support for them. After I found this discussion, I started analyzing the protocol and (almost) fully decoded it.

The sensors send the following data:
ID: 3 bytes serial number
Moisture level: percentage 0% to 100%
Raw moisture (AD): value varies from 12-15 to 400-500. Sensor manual explains relationship with moisture percentage
Battery voltage: fresh battery reads 1.5-1.6V. Sensor still works with 0.7V
Transmission boost: sensor usually transmits every 70 sec. After a moisture level change the sensor transmits 7 times the updated value every 10 seconds. This value is set to 7 following a sudden moisture change and decremented at each transmission until reaches 0

I added support for the sensor and currently they are sending data to my MQTT server via rtl_433. I'm not an expert with rtl_433 so I adapted some code in fineoffset.c, added the entry in rtl_433_devices.h and created pull request #1242 for it. The code seems stable and is working without any issue in my environment.

merbanan, 2020-01-07:
PR merged.

closed 2020-01-07 by merbanan

hackery, 2020-01-27:
Nice timing, as I just bought a few of these sensors and was going to be working on a decoder :)
Bug in https://github.com/merbanan/rtl_433/commit/b96d3463a8f4dfe0a62fe962a84365edf3dfe1d0 passed the wrong type for `battery_mv` into `data_make` though -

```
Tuner gain set to Auto.
Tuned to 433.920MHz.

pulse_FSK_detect(): Maximum number of pulses reached!
vdata_make() bad data type (680935712)
```

Fix in #1274
```
time      : 2020-01-27 15:05:27
model     : Fineoffset-WH51                        ID        : ffffff
Battery level: 0.000     Battery   : 1400 mV       Moisture  : 0 %           Transmission boost: 6
AD raw    : 22           Integrity : CRC
```

zuckschwerdt, 2020-01-27:
…and battery_level was also bad, fixed with 4cab9ff :)

hackery, 2020-01-28:
I found mention of calibration in a PDF [manual](https://fccid.io/WA5WH51/User-Manual/Users-Manual-3849955.pdf) for the device, changing the AD values considered 0% and 100%, but the default curve calculation doesn't seem to align with what I see in `ad_raw` vs `moisture`. I'm going to try adding a `moisture_derived` output locally to examine that. What's your understanding of the relationship between these fields?

Is there any way to pass in parameters for a device, to provide internal calculation changes? (and then how for multiple devices?) I suspect it would need to be calculated post-hoc in e.g. Grafana or some intermediate.

merbanan, 2020-01-28:
The output should be as close to the raw signal as possible but still usable. And all decoders are supposed to be stateless. So you have to do the adaption at the consumer stage.

algorni, 2023-04-14:
So the WH51 is suported generally? also the 868MHz version?

zuckschwerdt, 2023-04-15:
Is there an 868M version? I know of 433.92M and 915M. It should work as the sensor is already FSK, which is used on 868M usually.

gdt, 2023-04-15:
I would guess, emphasis on guess, that the 868 is the same as 915 except for tx frequency, in which case it should work.  But there is no way to know what it does until somebody tries one.