merbanan/rtl_433 #1064: Duplicate Messages (Acurite-Tower & Acurite-Lightning)

regulatre, 2019-05-23:
Below is a sample of a single "Chirp" of readings from my Acurite Lightning detector (Awesome device btw!)

*Expected behavior*: Single JSON reading

*Actual behavior*: A total of six messages, all having the same timestamp, three of which have model name `Acurite Lightning 6045M` and the other three have model name `Acurite-Lightning`.  I have a hunch that the "triplet" of messages *may* be because the device is chirping the same reading three times per burst, but I haven't found a way to confirm this. The triple readings may also be a bug. 

I reviewed the configuration file, where I have most protocols enabled, but I only see one associated with the Acurite Lightning detector (`protocol 40  # Acurite 592TXR Temp/Humidity, 5n1 Weather Station, 6045 Lightning`). I have this protocol enabled.  

``` json
{"time" : "2019-05-23 13:04:25.666273", "protocol" : 40, "model" : "Acurite Lightning 6045M", "id" : 112, "channel" : "C", "temperature_C" : 20.167, "humidity" : 54, "strike_count" : 13, "storm_dist" : 22, "active" : 0, "rfi" : 0, "ussb1" : 0, "battery_ok" : 1, "exception" : 0, "raw_msg" : "00706f3611878d96d0", "mod" : "ASK", "freq" : 433.923, "rssi" : -5.610, "snr" : 31.763, "noise" : -37.373}
{"time" : "2019-05-23 13:04:25.666273", "protocol" : 40, "model" : "Acurite Lightning 6045M", "id" : 112, "channel" : "C", "temperature_C" : 20.167, "humidity" : 54, "strike_count" : 13, "storm_dist" : 22, "active" : 0, "rfi" : 0, "ussb1" : 0, "battery_ok" : 1, "exception" : 0, "raw_msg" : "00706f3611878d96d0", "mod" : "ASK", "freq" : 433.923, "rssi" : -5.610, "snr" : 31.763, "noise" : -37.373}
{"time" : "2019-05-23 13:04:25.666273", "protocol" : 40, "model" : "Acurite Lightning 6045M", "id" : 112, "channel" : "C", "temperature_C" : 20.167, "humidity" : 54, "strike_count" : 13, "storm_dist" : 22, "active" : 0, "rfi" : 0, "ussb1" : 0, "battery_ok" : 1, "exception" : 0, "raw_msg" : "00706f3611878d96d0", "mod" : "ASK", "freq" : 433.923, "rssi" : -5.610, "snr" : 31.763, "noise" : -37.373}
{"time" : "2019-05-23 13:04:25.666273", "protocol" : 40, "model" : "Acurite-Lightning", "id" : 112, "channel" : "C", "temperature_C" : 20.167, "humidity" : 54, "strike_count" : 13, "storm_dist" : 22, "active" : 0, "rfi" : 0, "ussb1" : 0, "battery_ok" : 1, "exception" : 0, "raw_msg" : "00706f3611878d96d0", "mod" : "ASK", "freq" : 433.923, "rssi" : -5.610, "snr" : 31.763, "noise" : -37.373}
{"time" : "2019-05-23 13:04:25.666273", "protocol" : 40, "model" : "Acurite-Lightning", "id" : 112, "channel" : "C", "temperature_C" : 20.167, "humidity" : 54, "strike_count" : 13, "storm_dist" : 22, "active" : 0, "rfi" : 0, "ussb1" : 0, "battery_ok" : 1, "exception" : 0, "raw_msg" : "00706f3611878d96d0", "mod" : "ASK", "freq" : 433.923, "rssi" : -5.610, "snr" : 31.763, "noise" : -37.373}
{"time" : "2019-05-23 13:04:25.666273", "protocol" : 40, "model" : "Acurite-Lightning", "id" : 112, "channel" : "C", "temperature_C" : 20.167, "humidity" : 54, "strike_count" : 13, "storm_dist" : 22, "active" : 0, "rfi" : 0, "ussb1" : 0, "battery_ok" : 1, "exception" : 0, "raw_msg" : "00706f3611878d96d0", "mod" : "ASK", "freq" : 433.923, "rssi" : -5.610, "snr" : 31.763, "noise" : -37.373}
```

regulatre, 2019-05-23:
Similar to above, the Acurite-Tower sensor also shows up with duplicates, half of them have the model name `Acurite tower sensor` and the other half have model name `Acurite-Tower`.

``` json
{"temperature_F":74.84,"sensor_id":"14304","model":"Acurite tower sensor","port":33056,"snr":37.239,"time":"2019-05-23 02:59:54.320264","@version":"1","host":"decibel.k1gto.com","channel":"A","humidity":44,"temperature_C":23.8,"protocol":40,"battery_low":0,"mod":"ASK","noise":-37.373,"rx_timestamp":"2019-05-23T02:59:54.320Z","freq":433.945,"@timestamp":"2019-05-23T02:59:54.320Z","rssi":-0.134}
{"temperature_F":74.84,"sensor_id":"14304","model":"Acurite tower sensor","port":33056,"snr":37.239,"time":"2019-05-23 02:59:54.320264","@version":"1","host":"decibel.k1gto.com","channel":"A","humidity":44,"temperature_C":23.8,"protocol":40,"battery_low":0,"mod":"ASK","noise":-37.373,"rx_timestamp":"2019-05-23T02:59:54.320Z","freq":433.945,"@timestamp":"2019-05-23T02:59:54.320Z","rssi":-0.134}
{"temperature_F":74.84,"sensor_id":"14304","model":"Acurite tower sensor","port":33056,"snr":37.239,"time":"2019-05-23 02:59:54.320264","@version":"1","host":"decibel.k1gto.com","channel":"A","humidity":44,"temperature_C":23.8,"protocol":40,"battery_low":0,"mod":"ASK","noise":-37.373,"rx_timestamp":"2019-05-23T02:59:54.320Z","freq":433.945,"@timestamp":"2019-05-23T02:59:54.320Z","rssi":-0.134}
{"temperature_F":74.84,"sensor_id":"14304","model":"Acurite-Tower","port":33056,"snr":37.239,"time":"2019-05-23 02:59:54.320264","@version":"1","host":"decibel.k1gto.com","channel":"A","humidity":44,"temperature_C":23.8,"protocol":40,"mod":"ASK","noise":-37.373,"battery_ok":1,"rx_timestamp":"2019-05-23T02:59:54.320Z","freq":433.945,"@timestamp":"2019-05-23T02:59:54.320Z","rssi":-0.134}
{"temperature_F":74.84,"sensor_id":"14304","model":"Acurite-Tower","port":33056,"snr":37.239,"time":"2019-05-23 02:59:54.320264","@version":"1","host":"decibel.k1gto.com","channel":"A","humidity":44,"temperature_C":23.8,"protocol":40,"mod":"ASK","noise":-37.373,"battery_ok":1,"rx_timestamp":"2019-05-23T02:59:54.320Z","freq":433.945,"@timestamp":"2019-05-23T02:59:54.320Z","rssi":-0.134}
{"temperature_F":74.84,"sensor_id":"14304","model":"Acurite-Tower","port":33056,"snr":37.239,"time":"2019-05-23 02:59:54.320264","@version":"1","host":"decibel.k1gto.com","channel":"A","humidity":44,"temperature_C":23.8,"protocol":40,"mod":"ASK","noise":-37.373,"battery_ok":1,"rx_timestamp":"2019-05-23T02:59:54.320Z","freq":433.945,"@timestamp":"2019-05-23T02:59:54.320Z","rssi":-0.134}
```

zuckschwerdt, 2019-05-23:
The output looks processed in some way, not copyied directly from `-F json`?
The model names you are seeing are both the old and new model names. Are you using two outputs, one with `-M newmodel` and one without?

regulatre, 2019-05-23:
I'm invoking it like this
```/usr/local/bin/rtl_433 -G -g 18.6 -c /usr/local/etc/rtl_433/rtl_433.conf```

In the configuration, I have a single output enabled, for JSON:

```
output json:/root/433sensors.log
```

regulatre, 2019-05-23:
the `-G` enables "all the things". The question is, should the app be decoding a single transmission by multiple decoders? (Is there a way to tell it to stop at the first successful protocol decode?)

regulatre, 2019-05-23:
Put another way, from what I can tell I don't know that I can solve this by disabling one or more protocols in the configuration.

zuckschwerdt, 2019-05-23:
You should not use `-G`. There is a reason we disable decoders.

I guess in this case enabling all decoders, enabling default decoders and adding decoders from a conf file leads to duplicate decoders. I'll look into that.

regulatre, 2019-05-23:
I removed -G and left it to the default enabled protocols in the config file. I no longer see multiple model names for the same reports. 

Being that messages show up in "triplets" does this suggest the sensor is sending the message three times with each chirp? Curious the timestamp is precisely the same for all three. 

``` json 
{"time" : "2019-05-23 19:05:05.733021", "protocol" : 40, "model" : "Acurite-Lightning", "id" : 112, "channel" : "C", "temperature_C" : 20.500, "humidity" : 54, "strike_count" : 15, "storm_dist" : 15, "active" : 0, "rfi" : 0, "ussb1" : 0, "battery_ok" : 1, "exception" : 0, "raw_msg" : "00706f36118d0f0fd1", "mod" : "ASK", "freq" : 433.922, "rssi" : -2.303, "snr" : 36.830, "noise" : -39.134}
{"time" : "2019-05-23 19:05:05.733021", "protocol" : 40, "model" : "Acurite-Lightning", "id" : 112, "channel" : "C", "temperature_C" : 20.500, "humidity" : 54, "strike_count" : 15, "storm_dist" : 15, "active" : 0, "rfi" : 0, "ussb1" : 0, "battery_ok" : 1, "exception" : 0, "raw_msg" : "00706f36118d0f0fd1", "mod" : "ASK", "freq" : 433.922, "rssi" : -2.303, "snr" : 36.830, "noise" : -39.134}
{"time" : "2019-05-23 19:05:05.733021", "protocol" : 40, "model" : "Acurite-Lightning", "id" : 112, "channel" : "C", "temperature_C" : 20.500, "humidity" : 54, "strike_count" : 15, "storm_dist" : 15, "active" : 0, "rfi" : 0, "ussb1" : 0, "battery_ok" : 1, "exception" : 0, "raw_msg" : "00706f36118d0f0fd1", "mod" : "ASK", "freq" : 433.922, "rssi" : -2.303, "snr" : 36.830, "noise" : -39.134}

```

merbanan, 2019-05-23:
@regulatre yes, most likely the sensor sends the messages 3 times with so much time apart that rtl_433 consider it to be 3 separate transmissions.

closed 2019-05-23 by regulatre