merbanan/rtl_433 #1065: Using -F mqtt://localhost:1883 results in "MQTT connect error: Connection refused"

DeadEnded, 2019-05-24:
I'm sure I will need to supply more information, but here is what I have so far.

Mosquitto broker in a docker container - no credentials needed.
RTL433 in a docker container being executed in a bash script:
`    /usr/local/bin/rtl_433 -M newmodel -F "mqtt://172.17.0.4:1883,user=,pass=,usechannel=afterid,events=rtl433/events,devices=rtl433/devices"`

I want to remove the IP address as when I restart containers their IP can change.  I have accomplished this on all other mappings in my system, except this.  If I replace the IP address with "localhost" I get:

`
Trying conf file at "rtl_433.conf"...,
Trying conf file at "/root/.config/rtl_433/rtl_433.conf"...,
Trying conf file at "/usr/local/etc/rtl_433/rtl_433.conf"...,
Trying conf file at "/etc/rtl_433/rtl_433.conf"...,
Publishing MQTT UDP datagrams to localhost port 1883,
Registered 98 out of 125 device decoding protocols [ 1-4 8 11-12 15-17 19-21 23 25-26 29-36 38-60 62-63 67-71 73-100 102-103 108-116 119 121 124-125 ],
Found Rafael Micro R820T tuner,
Exact sample rate is: 250000.000414 Hz,
[R82XX] PLL not locked!,
Sample rate set to 250000 S/s.,
Tuner gain set to Auto.,
Tuned to 433.920MHz.
MQTT connect error: Connection refused
`

Same goes for 127.0.0.1, and leaving it blank.
I don't know if this is something I am doing wrong on mosquitto, or rtl433... but is seems like it should would as the instructions outline using localhost:1883.

Any help is appreciated.
Thanks!
DeadEnd

zuckschwerdt, 2019-05-24:
This is very likely a problem with container networking. You should set up networking to link containers or address container by name (i.e. not by localhost or ip).

0BuRner, 2019-05-24:
This link should help you : https://stackoverflow.com/questions/31149501/how-to-reach-docker-containers-by-name-instead-of-ip-address

DeadEnded, 2019-05-24:
Fantastic!
I had already created the user-defined network so after a couple of tests, it appears I have this working!
Thanks for the help.

closed 2019-05-24 by DeadEnded