merbanan/rtl_433 #1067: JSON output as MQTT payload.

kukabu, 2019-05-27:
Hello

Is it possible publish JSON output as MQTT payload?

closed 2019-05-27 by kukabu

petterreinholdtsen, 2019-09-03:
A simple way to do this is using the following shell oneliner:

```
rtl_433 -F json | while read line; do echo $line | mosquitto_pub -h 192.168.0.37 -i RTL_433 -l -t "RTL_433/Raw"; done
```

A related question is if it is possible to get the mqtt output format of rtl_433 to work with the autodiscovery of Home Assistant, described in https://www.home-assistant.io/docs/mqtt/discovery/.  I have not yet cracked that code.

zuckschwerdt, 2019-09-04:
Pipes are fun to get some result fast. But for stable operation you'd better use sockets. MQTT is available with `-F` and the JSON payload can be selected as `events`.

There is a MQTT Home Assistant auto discovery for rtl_433 events in the examples: https://github.com/merbanan/rtl_433/blob/master/examples/rtl_433_mqtt_hass.py

petterreinholdtsen, 2019-09-04:
[Christian W. Zuckschwerdt]
> Pipes are fun to get some result fast. But for stable operation you'd
> better use sockets. MQTT is available with `-F` and the JSON payload
> can be selected as `events`.

Thank you for the tip.  I am not quite sure what you mean by "the JSON
payload can be selected as `events`".  Can you explan a bit more?

> There is a MQTT Home Assistant auto discovery for rtl_433 events in
> the examples:
> https://github.com/merbanan/rtl_433/blob/master/examples/rtl_433_mqtt_hass.py

I tried this, but fail to get it working.  The problem seem to be the
way the mqtt data is sent from rtl_433 when I do like this:

  rtl_433 -F "mqtt://10.11.12.13:1883,retain=0,devices=rtl_433/events"

Perhaps the example can get a comment at the top explaining how to run
rtl_433 to get the correct mqtt messages?

-- 
Happy hacking
Petter Reinholdtsen

zuckschwerdt, 2019-09-04:
Use `rtl_433 -F mqtt` to get the JSON string at the topic `rtl_433/<hostname>/events`. Or use `rtl_433 -F mqtt,events=my-topic` to get the JSON string at some other topic.

For the HASS script you should add `-C si -M newmodel`. And for your example adjust the script setting:
```
MQTT_HOST = "10.11.12.13"
MQTT_PORT = 1883
MQTT_TOPIC = "rtl_433/events"
```