merbanan/rtl_433 #1070: Add platform and language to the ReadMe file

PeterMortensen, 2019-05-28:
For overview when evaluating, I suggest adding the programming language it is written in and the platform(s) it runs on to the ReadMe file. E.g. "It is written in C++ and runs on Linux systems." (if that is the case).

(I found this project from the show notes for a new video by "the guy with the Swiss accent", Andreas Spiess, "Safely Monitor and Alarm with Supervisord and Telegram", www DOT youtube DOT com/watch?v=z1y6j8-V7J0 . Andreas Spiess was recently interviewed by podcast The Amp Hour - theamphour DOT com/435-an-interview-with-andreas-spiess/)

zuckschwerdt, 2019-05-28:
Yes, and I might have a draft somewhere. Key points are:
- Written in portable C
- Older compilers and toolchain supported
- Low resources
- Linux, MacOS, Windows and embedded Linux

zuckschwerdt, 2019-08-28:
I've added the note with 39f0cfe now, thanks!

closed 2019-08-28 by zuckschwerdt