merbanan/rtl_433 #1071: cu8 files vs rtl_sdr

MeFisto94, 2019-05-30:
Hello Guys,
In what terms are cu8 files different than the recordings rtl_sdr itself can do?
I was trying to replay cu8 files today and even though rpitx supports `complex unsigned byte (u8)` as datatype, the replay was unsuccessful as compared to when using rtl_sdr.

I am now trying to figure out why and where the problem lies, as I was expecting rtl_433 to use the same format as rtl_sdr but only cutted to be shorter (where the actual signal is).
Is rtl_433 different and if so, would it be easy to add support for it?

merbanan, 2019-05-30:
There is no difference. Maybe the file is to short for the other tools.

zuckschwerdt, 2019-05-30:
Use e.g. http://triq.net/iqs to inspect cu8 files (and other types).
Also this kind of file has no header, you can just concat (and cut) them as needed.

MeFisto94, 2019-06-02:
[image]
Probably not something you guys can help with, I have to inspect the tool further and what is probably the best: Invest in building an antenna and not sending wave files but directly FSK Bytestreams (to not transmit the noise I recorded).

The first one (47) is transmitting the record from rtl_sdr, where you can see both pulses are transmitted good (the device always sends everything twice) and there rtl_433 is also able to decode it properly, but not for the second one (as there the second pulse is at fault).

But I need a better antenna since the device doesn't even care about rtl_sdr samples

closed 2019-06-02 by MeFisto94

MeFisto94, 2019-08-06:
Hey, sorry for abusing this issue, but the topic is quite related:
I have creating a 2-FSK Synthesizer, which aims to write cu8 files.
This works quite well so far with one exception: The Q Component isn't continuous where frequency change occurs. This is because I am just passing -90°/+90° as phase shift to the sine() method.

Now I thought about it and even when sampling the I component and using this as source, I'd have a 180° jump at the border of a frequency (from -90° to +90°), so is the following "expected"?

How could I solve it if not? I guess in reality those curves get softened by physics, but can I pass them to rtl_433 like that?

[image]

zuckschwerdt, 2019-08-06:
Do you really want that phase change on Q? You could use an NCO with stepped frequency change or some filter. Or even easier, fit the frequency to the bit rate, i.e. the bit length should be an integral multiple of the lower frequency (and the upper frequency should be double the lower frequency). Or just use -f and +f instead of two different frequencies.

You can also try `code_gen` form [tx_tools](https://github.com/zuckschwerdt/tx_tools/). The [PMV107J](https://github.com/zuckschwerdt/tx_tools/blob/master/examples/pmv107j.txt) example is 2-FSK. It offers a grammar for creating signals, gain, noise, bandwidth and other stuff. I'm currently expanding the tool with a UI. Expect major enhancements in the next weeks.

merbanan, 2019-08-06:
You could use a hilbert transform on the Q-channel. Or just synthesize the Q channel and then lowpass filter it. Synthetic signals will most likely trip the demodulator code some where.

MeFisto94, 2019-08-06:
Hey Guys, thanks for the quick replies!
The code basically looks like this (with a more clever method calling add_sample and thinking about "padding"):

```java
    // -90°, so Q is always before I, for a positive freq.
    public static final double q_offset = -0.5d * Math.PI;
    public void add_sample(double freq, double time) throws IOException {
        int i = double_to_byte(Math.sin(2d * Math.PI * phi));
        int q = double_to_byte(Math.sin(2d * Math.PI * phi + (freq < 0 ? -1d : 1d) * q_offset));

        outFile.write(i);
        outFile.write(q);

        movePhi(freq, time);
    }

    public void movePhi(double freq, double time) {
        phi += Math.abs(freq) * time;
        phi %= 1;
    }
```

>Do you really want that phase change on Q?

Yes and no, that happens because here is a change from 2e3 Hz to -1e3 Hz (The Phase Change occurs due to the negative frequency, without it looks fine, I just confirmed that). So how would it look in reality? Isn't that kind of expected mathematically?

>Or even easier, fit the frequency to the bit rate, i.e. the bit length should be an integral multiple of the lower frequency (and the upper frequency should be double the lower frequency).

That is what I actually already have. Every Bit is a multiple of the period duration (currently 100/200 times, so FSK is "hearable")

>Or just use -f and +f instead of two different frequencies.

Exactly the same problem. What I could do is instead of -f and f use f and 2f, but that requires a higher sample rate, where the "sign" of the frequency is essentially "free". Also I'd need to offset the base frequency, but it would work I suspect.

>You can also try code_gen form tx_tools. The PMV107J example is 2-FSK. 

I should've asked instead of crunching my brain the whole monday. But at least it strengthend my understanding of IQ.

>Or just synthesize the Q channel and then lowpass filter it.
You mean the lowpass would weaken the jump?
Hilbert Transform looks exactly what one needs, however I am not sure that it works, because I don't have samples in between the peaks.

merbanan, 2019-08-06:
> >Or just synthesize the Q channel and then lowpass filter it.
>You mean the lowpass would weaken the jump?

Yes, use a fir filter and sample shift the signal the amount of taps you use.

>Hilbert Transform looks exactly what one needs, however I am not sure that it works, because I don't have samples in between the peaks.

It is a sinc filter and quite annoying to implement.