merbanan/rtl_433 #1072: RTL_433 loses device

slvrscoobie, 2019-06-06:
Hi All, 
Im struggling with a problem with my SDR. I have it set up and running well, for most of my devices. Im using it to capture weather data from a couple of Ambient weather devices. 
For most of them, i can tune anywhere from 914.87xxmhz to 914.93xx mhz and still pick up everything inside and around the house. However, my outdoor array seems to get lost after a while. 
[image]
[image]
nothing changes, but it seems like overnight most times, i wake up and theres no report from my outdoor array for a while (hours). Ive tried to then tune the frequency and antenna position and sometimes itll come back for a couple days. but it usually eventually falls off. 
Today I spent about 30 minutes trying various frequencies and i get no hits on it at all. 
Now I do have 2 AW consoles that are picking it up fine, so i know it isnt that its not transmitting. 
Any advice on how to get better stability?

zuckschwerdt, 2019-06-06:
Try the option `-M level` to get an idea what center frequency the devices are on. Better even have a look with CubicSDR or Gqrx. Try to pick a frequency in the middle to cover all, the preset 250k sample rate covers +/- 125kHz. Raise the sample rate with e.g. `-s 1024k` to cover a wider frequency range.

slvrscoobie, 2019-06-06:
I tried CubicSDR but that was on another SDR so i think the specific frequency might be slightly off. I did try Gqrx, but I dont know if I understand what im looking for. I see red dots around 914.89mhz but when i tune there i dont get any better data.. 
[image]

slvrscoobie, 2019-06-06:
and heres from -M level
{"time" : "2019-06-06 15:44:48", "model" : "AmbientWeather-WH31E", "id" : 192, "channel" : 6, "battery" : "OK", "temperature_C" : -17.000, "humidity" : 63, "data" : "b600000000", "mic" : "CRC", "mod" : "FSK", "freq1" : 914.953, "freq2" : 914.877, "rssi" : -3.689, "snr" : 28.913, "noise" : -32.602}
{"time" : "2019-06-06 15:44:48", "model" : "AmbientWeather-WH31E", "id" : 192, "channel" : 6, "battery" : "OK", "temperature_C" : -17.000, "humidity" : 63, "data" : "b600000000", "mic" : "CRC", "mod" : "FSK", "freq1" : 914.953, "freq2" : 914.877, "rssi" : -3.689, "snr" : 28.913, "noise" : -32.602}
{"time" : "2019-06-06 15:44:56", "model" : "AmbientWeather-WH31E", "id" : 40, "channel" : 4, "battery" : "OK", "temperature_C" : 0.900, "humidity" : 32, "data" : "7000000000", "mic" : "CRC", "mod" : "FSK", "freq1" : 914.967, "freq2" : 914.896, "rssi" : -0.105, "snr" : 34.258, "noise" : -34.363}
{"time" : "2019-06-06 15:44:56", "model" : "AmbientWeather-WH31E", "id" : 40, "channel" : 4, "battery" : "OK", "temperature_C" : 0.900, "humidity" : 32, "data" : "7000000000", "mic" : "CRC", "mod" : "FSK", "freq1" : 914.967, "freq2" : 914.896, "rssi" : -0.105, "snr" : 34.258, "noise" : -34.363}
{"time" : "2019-06-06 15:44:58", "model" : "AmbientWeather-WH31E", "id" : 211, "channel" : 1, "battery" : "OK", "temperature_C" : 26.100, "humidity" : 63, "data" : "1200000000", "mic" : "CRC", "mod" : "FSK", "freq1" : 914.957, "freq2" : 914.888, "rssi" : -2.384, "snr" : 32.770, "noise" : -35.154}
{"time" : "2019-06-06 15:44:58", "model" : "AmbientWeather-WH31E", "id" : 211, "channel" : 1, "battery" : "OK", "temperature_C" : 26.100, "humidity" : 63, "data" : "1200000000", "mic" : "CRC", "mod" : "FSK", "freq1" : 914.957, "freq2" : 914.888, "rssi" : -2.384, "snr" : 32.770, "noise" : -35.154}
{"time" : "2019-06-06 15:45:14", "model" : "AmbientWeather-WH31E", "id" : 116, "channel" : 2, "battery" : "OK", "temperature_C" : 33.200, "humidity" : 44, "data" : "9200000000", "mic" : "CRC", "mod" : "FSK", "freq1" : 914.959, "freq2" : 914.883, "rssi" : -0.409, "snr" : 36.964, "noise" : -37.373}
{"time" : "2019-06-06 15:45:14", "model" : "AmbientWeather-WH31E", "id" : 116, "channel" : 2, "battery" : "OK", "temperature_C" : 33.200, "humidity" : 44, "data" : "9200000000", "mic" : "CRC", "mod" : "FSK", "freq1" : 914.959, "freq2" : 914.883, "rssi" : -0.409, "snr" : 36.964, "noise" : -37.373}
{"time" : "2019-06-06 15:45:15", "model" : "Fine Offset Electronics, WH25", "id" : 224, "temperature_C" : 24.800, "humidity" : 68, "pressure_hPa" : 996.100, "battery" : "OK", "mic" : "CHECKSUM", "mod" : "FSK", "freq1" : 914.961, "freq2" : 914.843, "rssi" : -0.131, "snr" : 31.221, "noise" : -31.352}
{"time" : "2019-06-06 15:45:31", "model" : "AmbientWeather-WH31E", "id" : 160, "channel" : 7, "battery" : "OK", "temperature_C" : 21.200, "humidity" : 67, "data" : "c800000000", "mic" : "CRC", "mod" : "FSK", "freq1" : 914.954, "freq2" : 914.881, "rssi" : -0.136, "snr" : 32.466, "noise" : -32.602}
{"time" : "2019-06-06 15:45:31", "model" : "AmbientWeather-WH31E", "id" : 160, "channel" : 7, "battery" : "OK", "temperature_C" : 21.200, "humidity" : 67, "data" : "c800000000", "mic" : "CRC", "mod" : "FSK", "freq1" : 914.954, "freq2" : 914.881, "rssi" : -0.136, "snr" : 32.466, "noise" : -32.602}
{"time" : "2019-06-06 15:45:34", "model" : "AmbientWeather-WH31E", "id" : 47, "channel" : 5, "battery" : "OK", "temperature_C" : 26.000, "humidity" : 60, "data" : "9800000000", "mic" : "CRC", "mod" : "FSK", "freq1" : 914.958, "freq2" : 914.874, "rssi" : -11.392, "snr" : 21.721, "noise" : -33.113}
{"time" : "2019-06-06 15:45:34", "model" : "AmbientWeather-WH31E", "id" : 47, "channel" : 5, "battery" : "OK", "temperature_C" : 26.000, "humidity" : 60, "data" : "9800000000", "mic" : "CRC", "mod" : "FSK", "freq1" : 914.958, "freq2" : 914.874, "rssi" : -11.392, "snr" : 21.721, "noise" : -33.113}
{"time" : "2019-06-06 15:45:42", "model" : "AmbientWeather-WH31E", "id" : 94, "channel" : 8, "battery" : "OK", "temperature_C" : -14.900, "humidity" : 69, "data" : "3700000000", "mic" : "CRC", "mod" : "FSK", "freq1" : 914.968, "freq2" : 914.892, "rssi" : -0.136, "snr" : 32.008, "noise" : -32.144}
{"time" : "2019-06-06 15:45:42", "model" : "AmbientWeather-WH31E", "id" : 94, "channel" : 8, "battery" : "OK", "temperature_C" : -14.900, "humidity" : 69, "data" : "3700000000", "mic" : "CRC", "mod" : "FSK", "freq1" : 914.968, "freq2" : 914.892, "rssi" : -0.136, "snr" : 32.008, "noise" : -32.144}
{"time" : "2019-06-06 15:45:54", "model" : "AmbientWeather-WH31E", "id" : 192, "channel" : 6, "battery" : "OK", "temperature_C" : -16.900, "humidity" : 63, "data" : "0b00000000", "mic" : "CRC", "mod" : "FSK", "freq1" : 914.955, "freq2" : 914.884, "rssi" : -3.509, "snr" : 29.093, "noise" : -32.602}
looks like its well above my 914.88 but if i tune to 914.95 it seems like i get less hits than at 914.8?

slvrscoobie, 2019-06-06:
I tried unplugging the USB device and then re-plugging in, and it started to find the WH65 again. 
What would cause that?? It would easily find the W31s but the WH65 took a reset before it would see them.

slvrscoobie, 2019-06-06:
Found Rafael Micro R820T tuner
Exact sample rate is: 250000.000414 Hz
[R82XX] PLL not locked!
Sample rate set to 250000 S/s.
Tuner gain set to Auto.
Tuned to 914.900MHz.
{"time" : "2019-06-06 18:38:54", "model" : "AmbientWeather-WH31E", "id" : 47, "channel" : 5, "battery" : "OK", "temperature_C" : 27.000, "humidity" : 53, "data" : "b500000000", "mic" : "CRC", "mod" : "FSK", "freq1" : 914.956, "freq2" : 914.886, "rssi" : -0.106, "snr" : 34.257, "noise" : -34.363}
{"time" : "2019-06-06 18:38:54", "model" : "AmbientWeather-WH31E", "id" : 47, "channel" : 5, "battery" : "OK", "temperature_C" : 27.000, "humidity" : 53, "data" : "b500000000", "mic" : "CRC", "mod" : "FSK", "freq1" : 914.956, "freq2" : 914.886, "rssi" : -0.106, "snr" : 34.257, "noise" : -34.363}
{"time" : "2019-06-06 18:39:07", "model" : "AmbientWeather-WH31E", "id" : 94, "channel" : 8, "battery" : "OK", "temperature_C" : -15.300, "humidity" : 69, "data" : "8700000000", "mic" : "CRC", "mod" : "FSK", "freq1" : 914.969, "freq2" : 914.886, "rssi" : -5.777, "snr" : 26.824, "noise" : -32.602}
{"time" : "2019-06-06 18:39:07", "model" : "AmbientWeather-WH31E", "id" : 94, "channel" : 8, "battery" : "OK", "temperature_C" : -15.300, "humidity" : 69, "data" : "8700000000", "mic" : "CRC", "mod" : "FSK", "freq1" : 914.969, "freq2" : 914.886, "rssi" : -5.777, "snr" : 26.824, "noise" : -32.602}
{"time" : "2019-06-06 18:39:15", "model" : "Fine Offset Electronics, WH25", "id" : 224, "temperature_C" : 25.400, "humidity" : 65, "pressure_hPa" : 996.500, "battery" : "OK", "mic" : "CHECKSUM", "mod" : "FSK", "freq1" : 914.956, "freq2" : 914.837, "rssi" : -0.129, "snr" : 35.025, "noise" : -35.154}
{"time" : "2019-06-06 18:39:29", "model" : "Fine Offset WH65B", "id" : 10, "temperature_C" : 28.200, "humidity" : 51, "wind_dir_deg" : 296, "wind_speed_ms" : 0.319, "gust_speed_ms" : 0.510, "rainfall_mm" : 717.804, "uv" : 3110, "uvi" : 8, "light_lux" : 103323.000, "battery" : "OK", "mic" : "CRC", "mod" : "FSK", "freq1" : 914.949, "freq2" : 914.884, "rssi" : -12.144, "snr" : 14.685, "noise" : -26.829}
So whats the Freq1 and Freq2? should I set it between the two? set it at Freq1, with bandwidth to cover Freq2?

zuckschwerdt, 2019-06-10:
Replugging the device could maybe cool it down a little and counter drift, not sure though.
For `OOK` the `freq1` is the estimated center frequency, For `FSK` both `freq1` and `freq2` are the estimated used frequencies. With CubicSDR or Gqrx you need to "eyeball" the frequencies with the cursor. Note the frequencies of your devices and then tune rtl_433 so that they lay safely withing the receiver band (center frequency `-f` plus/minus half the sample rate `-s`).
The frequencies here look ok, but it would be best to try with a higher sample rate (`-s 1024k`).

slvrscoobie, 2019-06-10:
Each time i try with the higher sample rate, i get nothing back, not the WH31 nor the WH65 - its like it doesnt hear anything (or its just picking up too much noise to decode them?)
it was working well the past 2 days @ 914.9mhz but today it started to rain and about an hour later it wasnt picking up the signal anymore.

zuckschwerdt, 2019-06-10:
You should record the signals and inspect visually (http://triq.net/iqs). As long as some signal is detected (even if it's not decoding properly) you can use `-S all` otherwise you would need to use `-w grab_914.9M_250k.cu8` (careful this gets large quickly, maybe use `-T 60`).
If you can grab a signal with 250k and also 1024k we can compare and see what is wrong with the higher sample rate.

slvrscoobie, 2019-06-10:
I recorded a couple of messages and saved them. Here is a link https://drive.google.com/file/d/1J5Vp8azxk8_Nb7VPkEyZEoggDNXy5suu/view?usp=sharing

slvrscoobie, 2019-06-12:
I redid my antenna setup, and got better readings almost immediately (SNR values higher). So its been 36 hours and no drops so far, so maybe the Antenna was just mistuned and not close enough to 915 to pick it up reliable, even though it was better than a 1/4 wave dipole. Ill close this for now, but it is weird itll be reading the signal for 3 days and then just stop reading it.

closed 2019-06-12 by slvrscoobie

zuckschwerdt, 2019-06-12:
Glad to hear. The signals you recorded are right up to the lower (g004-g006) and higher (g003) frequency bound (relevant parts close to the top or bottom in the spectrogram). Even a small amount of drift could reduce or eliminate one of the signals. (g007 looks ok.)
Maybe the better antenna allows for good reception even with bandfilter losses at the edges.