merbanan/rtl_433 #1073: Windowing to limit the frequency range

MeFisto94, 2019-06-09:
What possibilities do I have to limit the receiving bandwidth of rtl_433?
Would this be a useful feature or is there some easy "workaround" I am missing?

My problem is I'm trying to analyze an upconverted signal, which means the "ham it up!" adds 125MHz onto everything, but when trying to decode an originally 127kHz Signal, that means I have a very big peak at 125.000MHz (DC Offset), which ruins decoding the signal and even when watching the file with Audacity (recorded with rtl_sdr, as 433 didn't detect anything), I mainly see the Sine Wave of the 125 MHz (I guess).

Now I guess turning down the samplerate to 2 * f_bandwidth could work, like tuning into 125.100Mhz and then having roughly 60k as samplerate should cut everything above 125.130 and below 125.070 MHz? Is that the expected "solution"/approach? Can I even safely do that without loosing information? Furthermore, could I "resample" the already taken capture to rtl_433 to recognize the signals?

zuckschwerdt, 2019-06-10:
For now, if you really need a filter on the received data, you could
- receive to stdout (`rtl_433 -w -`)
- pipe through a filter (`| my_filter`)
- read from stdin and process (`| rtl_433 -r -`)

You can also just tune away from the upconverter carrier, but you'd need to test or read up how narrow the receivers band filter is. I'd use e.g. `-f 125.200M` with the default sample rate to get the 125.075M – 125.325M band.

MeFisto94, 2019-06-10:
I tried to leverage sox as a bandpass filter which showed partial success:
`sox -r 100k -e unsigned -b 8 -c 1 infile.raw outfile.raw band -n 1k 10k`
The Problem was that I directly tuned into 125.127M so the center frequency would've been 0. I don't know how well sox handles negative frequencies though, so maybe I would have to tune up higher and use the filter.

Anyway I can already see some ripples in Audacity, but their peak isn't large compared to the noise level. Maybe I could improve the filter some way? (I'll try to tune in higher later this day)

zuckschwerdt, 2019-06-10:
You need `-c 2` there as I/Q data is two distinct "channels". Also the default rate is 250k -- it doesn't matter much but you need to scale your filter parameters appropriately.

zuckschwerdt, 2019-06-10:
Regarding negative frequencies: Audacity and sox will read and process real valued channel data for each I and Q. Only if you view both together as complex valued data you can assign a negative frequency.
A DC component will always be trouble, some operation or filter might remove that component (i.e. center the signal), always try to have at least 10k offset (a single OOK or FSK pulse should show a decent number of oscillations).

MeFisto94, 2019-06-10:
Yeah the 100k were because of my lousy attempts to cut the frequencies.
And yeah, now I remember that the sign of the frequency just depends on the phase shift between I and Q. 
[image]
That's what Audacity shows after applying sox decently, unfortunately the peaks are very low, with offset tuning I might have a better result, because AGC works better when I don't have the strong DC part in it.

The signal looks strange to me though, not the typical FSK but also no AM/OOK. I wonder if the signals purpose is just to "be there". I'll report back when rtl_433 detects something though.

zuckschwerdt, 2019-06-10:
If you want to view the A or F component in Audacity, maybe try [plugins for that](http://triq.net/articles/sdr-nyquist-plugin-am-decode-iq-signals).

MeFisto94, 2019-06-10:
I used the plugin and what I saw were alternating phases (positive, negative) and amplitudes looking like either PWM or PCM. In the meantime I discovered that rtl_433 is already able to detect the pulses, which leads to the following:
```
 Detected OOK package    @15.600744s
Analyzing pulses...
Total count:  162,  width: 106.40 ms            (26599 S)
Pulse width distribution:
 [ 0] count:  124,  width:  120 us [104;148]    (  30 S)
 [ 1] count:    4,  width:  368 us [364;380]    (  92 S)
 [ 2] count:   30,  width:  244 us [232;256]    (  61 S)
 [ 3] count:    3,  width: 3952 us [3948;3960]  ( 988 S)
 [ 4] count:    1,  width:   96 us [96;96]      (  24 S)
Gap width distribution:
 [ 0] count:  124,  width:  132 us [116;152]    (  33 S)
 [ 1] count:    5,  width:  396 us [384;404]    (  99 S)
 [ 2] count:   30,  width:  264 us [256;276]    (  66 S)
 [ 3] count:    1,  width: 18380 us [18380;18380]       (4595 S)
 [ 4] count:    1,  width: 25432 us [25432;25432]       (6358 S)
Pulse period distribution:
 [ 0] count:  111,  width:  252 us [236;276]    (  63 S)
 [ 1] count:    4,  width:  764 us [764;768]    ( 191 S)
 [ 2] count:   18,  width:  512 us [504;540]    ( 128 S)
 [ 3] count:   26,  width:  384 us [368;396]    (  96 S)
 [ 4] count:    1,  width: 22340 us [22340;22340]       (5585 S)
 [ 5] count:    1,  width: 29384 us [29384;29384]       (7346 S)
Level estimates [high, low]:   2685,    133
RSSI: -7.9 dB SNR: 13.0 dB Noise: -20.9 dB
Frequency offsets [F1, F2]:    -113,      0     (-0.4 kHz, +0.0 kHz)
Guessing modulation: No clue...
```
So that looks like a clear 12 short, 240 long. 
Now when assuming `-X "n=flex,m=OOK_PWM,s=120,l=240,r=1000,g=10000"`, I get a halfway sane output as follows, there is something wrong with the length again though. Looks a bit like a problem with the preamble though. PCM shows nothing but Interlogix decoding with always the same code, but I don't know if it decodes all the pulses properly.

```
time      : @1.458212s
model     : flex         count     : 1             num_rows  : 1             rows      : 
len       : 47           data      : ffff4f6babb8
codes     : {47}ffff4f6babb8

time      : @1.458212s
model     : flex         count     : 1             num_rows  : 1             rows      : 
len       : 68           data      : ffff5ffff4f6bf5dc
codes     : {68}ffff5ffff4f6bf5dc

time      : @1.458212s
model     : flex         count     : 1             num_rows  : 1             rows      : 
len       : 47           data      : ffff4f6b6bb8
codes     : {47}ffff4f6b6bb8

time      : @3.657012s
model     : flex         count     : 1             num_rows  : 1             rows      : 
len       : 48           data      : ffff4f6bf5dc
codes     : {48}ffff4f6bf5dc

time      : @3.657012s
model     : flex         count     : 1             num_rows  : 1             rows      : 
len       : 67           data      : ffff5ffff4f6b6bb8
codes     : {67}ffff5ffff4f6b6bb8

time      : @3.657012s
model     : flex         count     : 1             num_rows  : 1             rows      : 
len       : 47           data      : ffff4f6babb8
codes     : {47}ffff4f6babb8
```
It goes on like this, difference seems to be only the number of f's and occasionally a 5.

zuckschwerdt, 2019-06-10:
Can you confirm that the symbols are short/short and long/long or is it a coincidence the pulse and gap counts match up? (Either use the Pulseview converter or `-w OOK:-`) I would have guessed PCM but the counts matching is strange. On the other hand PWM usually has short/long and long/short as symbols.

MeFisto94, 2019-06-10:
I am uncertain, if you look at the following picture, the first pulses look like PCM, but then there are phases where a pulse is twice as wide (hence my guess for PWM), but after that short period it goes back to what looks like PCM. So maybe it's just a bad transmitter loosing the gap. And what you can see in the above image is the clear sine, which is a continous pulse. 
[image]
That's a short exempt:
```
;ook 162 pulses
;freq1 -1175
152 104
148 112
140 112
144 132
124 136
116 136
124 132
120 132
124 132
120 132
128 128
128 128
128 124
136 120
132 144
112 136
372 400
112 140
244 268
244 264
120 268
120 136
116 144
116 132
248 264
124 128
128 124
256 140
116 272
240 136
120 132
128 128
128 124
132 124
132 260
124 132
248 144
112 260
252 276
108 144
116 140
116 140
240 268
120 136
120 132
124 132
252 144
3948 18396
112 152
108 148
108 144
116 136
116 144
112 140
116 136
124 132
124 132
124 128
128 128
128 124
132 128
128 128
124 132
128 136
372 400
116 136
244 264
88 388
```

zuckschwerdt, 2019-06-10:
If you quantize this to 125 µs then you get some long/short or short/long pairs in there. Might be PCM with a boring payload (runs of 0's or 1's).
```
1 1
1 1
1 1
1 1
1 1
1 1
1 1
1 1
1 1
1 1
1 1
1 1
1 1
1 1
1 1
1 1
3 3
1 1
2 2
2 2
1 2
1 1
1 1
1 1
2 2
1 1
1 1
2 1
1 2
2 1
1 1
1 1
1 1
1 1
1 2
1 1
2 1
1 2
2 2
1 1
1 1
1 1
2 2
1 1
1 1
1 1
2 1
32 147
1 1
1 1
1 1
1 1
1 1
1 1
1 1
1 1
1 1
1 1
1 1
1 1
1 1
1 1
1 1
1 1
3 3
1 1
2 2
1 3
```

MeFisto94, 2019-06-11:
What values would I have to put into the flex decoder then for that? It could well be that the payload is boring, as it's purpose is to trigger a sender to wake up and communicate on 433 MHz

zuckschwerdt, 2019-06-13:
Assuming the bit width is 120 µs, use e.g.
```
-X "n=mypcm,m=OOK_PCM,s=120,l=120,r=1000"
```
i.e. in 120 µs slices (long width) expect 120 µs pulses (short width). This is NRZ, for RZ short would be shorter than long. Reset width limits the maximum run of bits, 2 or 3 seems to be the max here so you could go with 400 µs maybe.

zuckschwerdt, 2019-06-23:
Closing as this is likely not an issue we need to track. Feel free to continue this issue or reopen when something needs to be done in rtl_433.

closed 2019-06-23 by zuckschwerdt