merbanan/rtl_433 #1075: -y option not working as expected

Mindavi, 2019-06-15:
Hey, I was looking at issue #1062 and thought I'd try out the `-y` option for my hideki-type temperature sensor.
However, I can't seem to get it to work.

[g150_433.92M_250k.cu8.zip](https://github.com/merbanan/rtl_433/files/3293052/g150_433.92M_250k.cu8.zip)
(rename the file to *.cu8 instead of .zip).

running that file through `rtl_433 -r file.cu8 -A` gives me an output that says
```
bitbuffer:: Number of rows: 1 
[00] {72} 54 14 88 4a 80 22 04 90 42
```

Then running `rtl_433 -y '{72} 54 14 88 4a 80 22 04 90 42'` gives me
```
Verifying test data with device HIDEKI TS04 Temperature, Humidity, Wind and Rain Sensor.
hideki_ts04_callback: Parity error at 2
```

I can't seem to figure out why it's not working. Does that have something to do with encoding? Am I using the `-y` flag wrong?

zuckschwerdt, 2019-06-15:
The actual bitbuffers seen by the Hideki decoder (using [these parameters](https://github.com/merbanan/rtl_433/blob/master/src/devices/hideki.c#L205)) are:
```
{72} 06 c4 ab 20 bf b3 e2 9d 69
{72} 06 c4 ab 20 df b3 e2 91 0f
{72} 06 c4 ab 20 8f b3 e2 9b 4b
```

Mindavi, 2019-06-17:
Ah so instead of auto-analyzing(`-A`) I should use those parameters in a decoder? Should I use the `-X` option for that, or can I see what bitbuffers are sent to specific devices somehow?

zuckschwerdt, 2019-06-17:
The bitbuffers are specific to the decoder parameters (the guessed `-A` parameters will usually differ). The best way to see the actual buffers is to insert `bitbuffer_print(bitbuffer);` early in the decoder. Otherwise you can replicate the decoder parameters with a flex spec, yes.

Mindavi, 2019-06-17:
Thank you, I'll use the former option. Maybe it would be good when those are printed by default by trace decoding? Anyway, my question is answered so closing now.

closed 2019-06-17 by Mindavi