merbanan/rtl_433 #1076: Water Leakage Sensor - sometimes detected, sometimes not

europacafe, 2019-06-16:
Hi,
I got one outdoor weather sensors (wh24), one indoor (wh25), and another indoor (la crosse tx-440-th). All detected without problem with my cheap tv stick by the following command
rtl_433 -F json -M utc | mosquitto_pub -t home/rtl_433 -l -h 192.168.2.14

I just got a 433mhz water leakage sensor. Even I put it near my rtl-sdr stick, sometimes the signals are picked up, but mostly not. The sensor lights up every time I short the sensor legs, though.

I'm not sure whether it is due to the sensor weak transmission, or it could be fine tuned somehow.

zuckschwerdt, 2019-06-16:
Could be the frequency. Use CubicSDR or Gqrx to ensure the frequency is not to much at the edges of the window. Could also be the modulation. Grab a sample, check with http://triq.ne/iqs and post it here.

europacafe, 2019-06-16:
Thanks. I am quite new to this. I don't know how to grab the sample. I just grab a screenshot of the waterfall. You can see small vertical green line was coming from one of my weather sensor. The two signals  with quite wide signals came from my water leak detector
[image]
. I'm not sure this could help.

zuckschwerdt, 2019-06-16:
Do you use some R820T(2) dongle? That's quite a DC spike. Or is that a signal? Tune to e.g. 433.85M and see if the fat line in the middle also moves.
Did you use the default sample rate (of 250k) with rtl_433? It would only cover 433.92+/-0.125.
The OOK signal at 433.79 and the FSK at 433.98 thus are very much at the edges of the window. Perhaps try `-s 1024k` for `rtl_433`.

europacafe, 2019-06-16:
The long green vertical lines are not signals. I don't know what they are.

Data signals came on left and right of frequency 434. When I run using default frequency and sampling rate of 250k, I can still pick up my other sensors data.
The -s 1024k or 433.85 can't capture data at all.
One thing I noticed, the data signals which I can capture look narrow bases and sharp spikes; but my water leakage signal looks much a wider-base signal, as seen in the above screenshot.

zuckschwerdt, 2019-06-16:
The narrow signals are OOK (a single frequency on/off), the wider ones are FSK (two frequencies alternating).
Does the spike in the middle move with the tuned frequency? If not then you have something jamming there and need to tune away from that, e.g. to 434.08 M, with the default 250 k that should still capture your sensors.

europacafe, 2019-06-19:
Thank you. 434.08 did not give any data at all. I think the signal produced by the sensor is too fuzzy. I may try another sensor when I get it.

Thanks.

zuckschwerdt, 2019-06-19:
You can set Gqrx to 250k sample rate and tune somewhere around 434.08 to see if you can get rid of the spike at 433.920 and get good reception. It very much trial and error to get it right.

zuckschwerdt, 2019-06-23:
Closing as this is likely not an issue we need to track. Feel free to continue this issue, we just don't need to track progress ;)

closed 2019-06-23 by zuckschwerdt