merbanan/rtl_433 #1077: TFA 30.3222.02 (wind/huminity/temp.)

Peterxxxy, 2019-06-16:
Hi, I'm trying to decode the TFA 30.3222.02 sensor, but I'm stuck with reading the values.
Sorry if there is something very obvious here that I don't see, it's the first time I try to decode such data.
The both other existing TFA-sensor device protocols seem not to work.
I'm receiving a signal and think I have the correct or at least working parameters (from -A and pulseview):
`-X 'n=name,m=OOK_PWM,s=290,l=560,r=1160,g=0,t=0,y=835'`

I did some recordings in different environments and looked for the changing digits.
I get two different hex codes with each transmission. I started to focus on the fist one, because it seems like that's the part with the temperature and humidity and I have data to compare with. (if I get how to read the data there its hopefully the same with the wind speed in the second hex code)
The data could be encoded like this (I=identification, H=humidity, T=temperature):
`?????IIIIIIIIIIIIIIIIIII????????????? HHHHHHH ????? TTTTTTTTTTTTTTT ????????`
`(2nd [# = maybe wind speed]: ?????IIIIIIIIIIIIIIIIIII??????????? ######### ???????????? ######## ????)`

Here are some recorded codes:
```
1111010101111000000010101111111011010 1001100 11111 010110001100100 10000000 (f5 78 0a fe d4 cf ac 64 80)
1111010101111000000010101111111011010 1100010 11111 011101001111011 10000000 (f5 78 0a fe d6 2f ba 7b 80)
1111010101111000000010101111111011010 0001100 11111 100010001001111 10000000 (f5 78 0a fe d0 cf c4 4f 80)
```

To find out how to read the data out of the parts I put the sensor in the fridge for some minutes and then took recordings while it was warming up back to room temperature (that day 26°).
I tried to invert and/or reverse the temperature/humidity bits. When they are just inverted I got some nice values:
```
temp. humid.
17796  29
18937  29
20146  31
21282  41
23634  54
15076  107
```
But than at about 24° the values got wired...
If I try to apply that inverting to my previous recordings I also get this odd behavior: the colder outside temperatures (20-22°) are matching, but for the inside ones I got too low values (15280, 17327, 14036, ...).

Now I'm stuck. Maybe I'm wrong with the parameters or I guessed wrong with the inversion? I would appreciate some help or tips what to do next.
Here are some recordings: 
[recordings.zip](https://github.com/merbanan/rtl_433/files/3294161/recordings.zip)

Thanks for your help!

zuckschwerdt, 2019-06-16:
`g=0,t=0` are the defaults (i.e. not used) -- you can remove that.
ID is usually an 8-bit value, but there also may be a preamble or sync.
A good way to get a grip on the encoding and ask for help is to paste codes in [BitBench](http://triq.net/bitbench), try a format string, then post a link here.

Peterxxxy, 2019-06-16:
Thanks for that lightning-fast answer.
The BitBench is very helpfull. It does exactly that stuff I did manually the last days.

I put some data in BitBench and figured out how to get the results I described above (and found a possible id with 8 bit).
Here's the [BitBench](http://triq.net/bitbench?c=f5%2078%200a%20fe%20d0%20cf%20c4%204f%2080%20%5B26%C2%B0%2C%2057%25%5D&c=f5%2078%200a%20fe%20d4%20cf%20ac%2064%2080%20%5B20%C2%B0%2C%2076%25%5D&c=f2%203b%20c5%20fe%20d0%205f%20c9%202b%2080%20%5B26%C2%B0%2C%2053%25%5D&c=f5%2078%200a%20fe%20d6%202f%20ba%207b%2080%20%5Bfridge%201%5D&c=f5%2078%200a%20fe%20d6%202f%20b6%2006%2080%20%5Bfridge%202%5D&c=f5%2078%200a%20fe%20d6%200f%20b1%204d%2080%20%5Bfridge%203%5D&c=f5%2078%200a%20fe%20d5%206f%20ac%20dd%2080%20%5Bfridge%204%5D&c=f5%2078%200a%20fe%20d4%209f%20a3%20ad%2080%20%5Bfridge%205%5D&c=f5%2078%200a%20fe%20d1%204f%20c5%201b%2080%20%5Bfridge%206%5D&c=F2%203B%20C5%20FE%20D5%20EF%20B1%2094%2080&c=F2%203B%20C5%20FE%20D3%209F%20B5%20CE%2080&c=F3%2081%208D%20FE%20D7%207F%20B6%20F2%2080&c=F3%2081%208D%20FE%20D5%20BF%20B9%20FA%2080&c=F3%2081%208D%20FE%20D4%206F%20C2%206A%2080&c=F3%2081%208D%20FE%20D3%20AF%20C8%20F8%2080&c=F3%2081%208D%20FE%20D2%209F%20CB%205F%2080&f=5x%20ID%3F%3Ahh%20b%20ID%2FCH%3F%3Ah%20bb%20ID%2FCH%3F%3Ah%2013x%20HUM%3A~7d%20xxxxx%20TEMP_C%3A~15d%208x)

Peterxxxy, 2019-07-02:
I couldn't solve this problem, but now I'm using another sensor which works great out of the box.
So I think there is no need to keep this open.

closed 2019-07-02 by Peterxxxy

Peterxxxy, 2021-10-09:
Hello @Domifry, I think it is a VENTUS W132 (back then it was about 23€).
My program uses device-protocol 16 (`[16]  AlectoV1 Weather Sensor (Alecto WS3500 WS4500 Ventus W155/W044 Oregon)`) to read the sensor.

Domifry, 2021-10-09:
Hello @Peterxxxy - thanks! I will order it and test it!

Domifry, 2021-10-09:
Update: I succeeded with the TFA 30.3222.02 (wind/huminity/temp.) - so no testing of the other one! See here: [https://github.com/merbanan/rtl_433/issues/1827#issuecomment-939356981](https://github.com/merbanan/rtl_433/issues/1827#issuecomment-939356981)