merbanan/rtl_433 #1078: MQTT: Set channel to "0" when "usechannel=replaceid" and protocol does not have channels

curlyel, 2019-06-17:
I assume, that `usechannel=replaceid` is intended to _get rid of random ID's in MQTT topics_ (e.g. changing during battery replacement).

This working pretty well so far, _as long as a protocol does have channels_ ;-)

But: If a protocol _does not have channels_, then the random ID is used in generating the MQTT topic again, e.g. Alecto WS-1200 (a.k.a. Fineoffset-...) rain gauges:

```
sensors/rtl_433/Fineoffset-WH0530/236/id 236
sensors/rtl_433/Fineoffset-WH0530/236/temperature_C 23.900000
sensors/rtl_433/Fineoffset-WH0530/236/rain_mm 6.300000
sensors/rtl_433/Fineoffset-WH0530/236/battery_ok 1
sensors/rtl_433/Fineoffset-WH0530/236/mic CRC
sensors/rtl_433/Fineoffset-WH0530/147/id 147
sensors/rtl_433/Fineoffset-WH0530/147/temperature_C 32.500000
sensors/rtl_433/Fineoffset-WH0530/147/rain_mm 82.199997
sensors/rtl_433/Fineoffset-WH0530/147/battery_ok 1
sensors/rtl_433/Fineoffset-WH0530/147/mic CRC
```
It's likely more useful, if the channel value is set to "0" in that case or is omitted completely.

Btw.: An Oregon Scientific rain gauges have none selectable channel too. This one transmits a `channel=0`, see:

```
sensors/rtl_433/OS/Oregon-PCR800/0/id 142
sensors/rtl_433/OS/Oregon-PCR800/0/channel 0
sensors/rtl_433/OS/Oregon-PCR800/0/battery_ok 1
sensors/rtl_433/OS/Oregon-PCR800/0/rain_rate_mm_h 0.000000
sensors/rtl_433/OS/Oregon-PCR800/0/rain_mm 1739.979614
```

zuckschwerdt, 2019-06-18:
Proposed with #1079. But this leads to a rabbit hole of options, when/where do we want more defaults? Needs more thought.

zuckschwerdt, 2019-06-23:
Thanks for the idea. Tracked in the PR now.

closed 2019-06-23 by zuckschwerdt