merbanan/rtl_433 #1080: Lidl AURIO Funk-Wetterstation

vpeter4, 2019-06-18:
I have this device from LIDL bought in 12/2018 and would like to use it with rtl_433. But I can't get it working. Anyone got in touch with it?

https://www.lidl.de/de/auriol-funk-wetterstation/p286524

zuckschwerdt, 2019-06-18:
Another Auriol branded sensor is supported by the Alecto protocol. But this sensor might be quite different as "Auriol" is just a rebranding.
Best to grab a few samples (`-S all`), check them (http://triq.net/iqs), and upload here.

vpeter4, 2019-06-19:
After testing and testing I finally got some useful response decoded as GT-WT-02 sensor. This sensor is from ALDI and my sensors are from LIDL (and both have few things in common).
Sadly I only got one response from one sensor (I have 3 sensors on different channels) and nothing more. Not sure the reason. RTL receiver seems to be working fine because I see some (undecoded) response from my car remote control.
It looks like data from sensors are not even received (saved data with -S is just a junk and they don't happen when sensor transmits). Could it be that frequency is different? Any other tip to check? Or maybe better antenna? I'm using small antenna like https://snag.gy/VcMIQx.jpg

zuckschwerdt, 2019-06-19:
Best bet for a clean receive without interference is to remove the antenna and place the sensor at about 10 cm from the receiver.
Frequencies are traditionally 433.92 M, but 868 M is also commonly found in newer products.
It's easiest to use CubicSDR or Gqrx to look at live data transmissions, just to spot them and determine the frequency.

vpeter4, 2019-06-24:
So here is one sample: http://s000.tinyupload.com/index.php?file_id=35044905851481169890
Looks ok to me but can't decode it. Does it even make sense?

zuckschwerdt, 2019-06-24:
Yes, that's a very clean sample. It's PPM with 650 µs pulses and
- 2000 µs short pulse
- 4000 µs long pulse
- 9000 µs sync pulse

Use a flex decoder with `-X 'n=something,m=OOK_PPM,s=2000,l=4000,y=9000,g=4500,r=9500,t=500'`

Run this, log the output along with what you see on the head unit. Then we can figure out the decoding.

Are there any markings on the sensor, maybe in the battery compartment? We would need the type/product code to properly name things.

vpeter4, 2019-06-24:
The only marking on a PCB is _YJ-T12 V02_ which is not useful at all.
Still wondering how I got decoded data as _GT-WT-02 sensor_. And the decoded data was really from my sensor (I'm 99.99% sure). But it happened only once :(
From console at that time
```
time      : 2019-06-19 09:30:49
model     : GT-WT02      ID Code   : 166
Channel   : 3            Battery   : OK            Button    : 0             Temperature: 30.2 C       Humidity  : 48 %
Integrity : CHECKSUM
pulse_demod_ppm(): GT-WT-02 Sensor
bitbuffer:: Number of rows: 11
[00] {37} a6 21 2e 61 48 : 10100110 00100001 00101110 01100001 01001
[01] { 1} 00             : 0
[02] {39} a9 88 4b 98 52 : 10101001 10001000 01001011 10011000 0101001
[03] { 1} 00             : 0
[04] {39} a9 88 4b 98 52 : 10101001 10001000 01001011 10011000 0101001
[05] { 1} 00             : 0
[06] {39} a9 88 4b 98 52 : 10101001 10001000 01001011 10011000 0101001
[07] { 1} 00             : 0
[08] {39} a9 88 4b 98 52 : 10101001 10001000 01001011 10011000 0101001
[09] { 0}                :
[10] { 0}                :
```

In next weeks I will try to decode it using GPIO on Raspberry Pi.

zuckschwerdt, 2019-06-24:
Are those readings plausible? The GT-WT-02 requires 37 bits, here we have 39 bits -- but that's just sync bits. Maybe they are the same with different sync bits.
If you want to test, just change the `37` in https://github.com/merbanan/rtl_433/blob/master/src/devices/gt_wt_02.c#L57
But chances are the sync bits are at the front and things won't work using this hack.

vpeter4, 2019-06-24:
Success at last!
For **GT-WT-02 Sensor** I add on top of function _gt_wt_02_process_row()_ shifting left for 2 bits (2=39-37):
```
		// shift data left for 2 bits
		b[0] = (b[0] << 2) + ((b[1] & 0xC0) >> 6);
		b[1] = (b[1] << 2) + ((b[2] & 0xC0) >> 6);
		b[2] = (b[2] << 2) + ((b[3] & 0xC0) >> 6);
		b[3] = (b[3] << 2) + ((b[4] & 0xC0) >> 6);
		b[4] = (b[4] << 2);
		
		bitbuffer->bits_per_row[row] -= 2;
```
Is it possible to skip this 2 sync bits with some better way? To make this decoder universal.

And thanks for all the help. I really appreciate it.

zuckschwerdt, 2019-06-24:
Sure, if this hack works out to produce reliable results let me know in a few days and I'll add something suitable.

vpeter4, 2019-06-29:
I'm using build with a small change posted in my previous post (skipping 2 sync bits). It works perfectly with my sensors.
I was thinking is this approach would be fine and I think it would be. If I understand correctly all decoders are called with same data and only one continues and produce result. This **GT-WT-02 Sensor** is using 37 bits. If there is more data like 39 bits for my sensor we just cut it out and continue processing. If all goes well we get result.

zuckschwerdt, 2019-07-09:
Supported with 804eafe now. Thanks for reporting and testing this. Reopen if something doesn't work.

closed 2019-07-09 by zuckschwerdt

vpeter4, 2019-07-09:
Tested new version and it is working. Thank you again for the help.