merbanan/rtl_433 #1082: Why we're adding the new model keys option

richardh9935, 2019-06-20:
I've been struggling to understand what -M newmodel was about, until I read:
" `model` is just a short unique key for the protocol. Not a description of any kind."
_Originally posted by @zuckschwerdt in https://github.com/merbanan/rtl_433/pull/986#issuecomment-475974458_

So am I right in thinking that I would change my usage of RTL-433 from
rtl_433 -R 40 -T 400 -F csv:log1.csv. to
rtl_433 -R -M acurite5n1 -T 400 -F csv:log1.csv

ie the "40" was the old model, and "-M acurite5n1" is the new way.

zuckschwerdt, 2019-06-20:
The `-R` option is unchanged. Use `-M newmodel` to get _better output_. E.g. the output of `S3318P Temperature & Humidity Sensor` will be replaced by `Conrad-S3318P`, which is quicker to read and easier to log into a database e.g.

zuckschwerdt, 2019-06-23:
Closing as this is likely not an issue we need to track. Feel free to continue this issue, we just don't need to track progress ;)

closed 2019-06-23 by zuckschwerdt