merbanan/rtl_433 #1083: Sample Python Process

lesreaper, 2019-06-21:
I'm looking to test out the Python pipeline for streaming MQTT, and the defaults don't seem to be working. Am I missing something obvious?

I've tried both of these two to kick off the broker:
```
rtl_433 -f 345M -G -a -A -F "mqtt://localhost:1883,user=username,pass=password,retain=0,usechannel=replaceid, devices=test"
```
The above I included `test` as a topic I believe.
```
rtl_433 -f 345M -G -A -F syslog:localhost:1433
```

Then, in another terminal tab I ran the relay:
```
cd examples
python3 rtl_433_mqtt_relay.py
```

Then, in another tab, I ran the test client:
```
cd examples
python3 mqtt_rtl_433_test_client.py 
```

When I use the standard Honeywell sensor, it has problems reading it at the moment, but it does go through a bunch of attempts to demodulate. However, I would expect there to be some kind of result in the client, but it prints nothing extra beyond the starting text of:
```
[INFO/MainProcess] MQTT RTL_433 Test Client
[INFO/MainProcess] MQTT Connection: Connection Accepted.
```

Is there anything I should be doing different in this kind of setup to get actionable data in the client?

zuckschwerdt, 2019-06-21:
Those are three different ways to send MQTT. Choose one, don't use all.
- `rtl_433 -F mqtt` **or**
- `rtl_433 -F syslog` and `rtl_433_mqtt_relay.py` **or**
- `rtl_433 -F json` and `mqtt_rtl_433_test_client.py`

Never use `-G`. Really, I mean that: never.

zuckschwerdt, 2019-06-23:
Closing as this is likely not an issue we need to track. Feel free to continue this issue, we just don't need to track progress ;)

closed 2019-06-23 by zuckschwerdt

lesreaper, 2019-06-25:
Ok, so I'm back to this part of the code.nI'm still not understanding the process, or at least, it's not working properly. 

I do the JSON as you say, sounds great, JSON is easier to parse. I run this from the command line as per the example doc:
```
rtl_433 -F json -M utc | mosquitto_pub -t home/rtl_433 -l
```
However, it gives me a not authorized, not sure where the username and password for the MQTT broker goes. Usually there's a broker, publisher, and client. I'm assuming using the JSON here turns this into a broker and publisher. Is this not the case? Do I have to run the broker separately?

zuckschwerdt, 2019-06-25:
You need a broker already running somewhere. All three methods above use JSON and all are simple MQTT clients (clients are pub and sub capable).
Use [`man mosquitto_pub`](https://mosquitto.org/man/mosquitto_pub-1.html)  to learn about passwords.
The first method (prefered) would take a password this way: `rtl_433 -F "mqtt,password=pA55w0rD"`

lesreaper, 2019-06-27:
I don't know, I'm just at a loss why this doesn't work. 

I have the broker running, I can see port 1883 open using `netstat -at`.

I copy the code straight from `mqtt_rtl_433_test_client.py`, only changing the following:

```
MQTT_SERVER = "localhost"
# Setup MQTT client
mqtt_client.username_pw_set('username', 'password')
```

I start the Python script and it connects and is authorized.

In another window, I run `rtl_433 -R 23 -a -A -F "mqtt,user=username,pass=password,retain=0,devices=home/rtl_433"`

It connects and is authorized. 

The opening door sensor gets picked up. The DSC-4950 is 23, but sometimes it doesn't get the nice headers and I don't get the pretty output. Not sure what's happening there, but I'm assuming that's another issue.. 

However, nothing in the client runs when the door is activated. I added checks in the function for `on_message`, and it never even fires when the door is opened.

zuckschwerdt, 2019-06-27:
You are using two different MQTT clients here. `mqtt_rtl_433_test_client.py` is a client with no input, thus no output. You would have to use pipes to make it actually do something. But read on.

`rtl_433 -R 23 -a -A -F "mqtt,user=username,pass=password,retain=0,devices=home/rtl_433"` is also a MQTT client and will send every received data structure to the broker on it's own.
Add a `-F json` or `-F kv` to see what goes to the broker. (kv is colorful and easier to read, json releates directly to the MQTT output.

lesreaper, 2019-06-27:
Ok,so now I understand you can add two flags of the same type to this setup. So the command is:
`rtl_433 -R 23 -a -A -F "mqtt,user=username,pass=password,retain=0,devices=home/rtl_433" -F json` to see that json data passed to the broker. 

That's awesome, and it looks good, especially with `-F kv` instead of `-F json`

But how do I then read it in my Python app if the `mqtt_rtl_433_test_client.py` doesn't read these messages from the broker. Isn't that the point of an MQTT client? Should I be using another one of the sample files?

zuckschwerdt, 2019-06-27:
I see where the confusion is now. `mqtt_rtl_433_test_client.py` is a bridge to publish JSON **to** MQTT. What you want is something like mosquitto_sub to read **from** MQTT (subscribe). Try some of the Paho MQTT examples to receive topic from MQTT.

lesreaper, 2019-06-28:
I wish I could say I solved this problem after a few hours of work, but I didn't. I'm a little closer. I'm using `rtl_433 -R 23 -a -A -F "mqtt,user=username,pass=password,retain=0,devices=rtl_433" -F json`
and a basic Paho client:

```
# The callback for when the client receives a CONNACK response from the server.
def on_connect(client, userdata, flags, rc):
    print("Connected with result code "+str(rc))

    # Subscribing in on_connect() means that if we lose the connection and
    # reconnect then subscriptions will be renewed.
    # client.subscribe("$SYS/#")
    client.subscribe("rtl_433")

# The callback for when a PUBLISH message is received from the server.
def on_message(client, userdata, msg):
    print(msg.topic + " " + str(msg.payload))

# Setup MQTT client
client = mqtt.Client()
client.username_pw_set('username', 'password')
client.on_connect = on_connect
client.on_message = on_message
client.connect("localhost", 1883, 60)
client.loop_start()
```

If I subscribe to the `$SYS` topic, I get something like this: https://i.imgur.com/9fhxs05.png

However, I get nothing on the `rtl_433` topic, which is under `devices=/rtl_433` in the linux command line. 

I tried this fork: https://github.com/mverleun/RTL433-to-mqtt
I looked at other options: https://www.hagensieker.com/wordpress/2019/03/06/how-to-keep-rtl_433-alive-for-your-home-automation-using-supervisor/

Honestly, I don't know what I'm missing. Have you found a tut or example where someone is able to get a MQTT message from a door sensor and read it in Python? Am I doing it right and my topics aren't being read for some reason?

zuckschwerdt, 2019-06-28:
Looks good. Careful with topic names, `rtl_433` isn't the same as `/rtl_433` (usually topics don't start with a slash, but that looks ok in your snippet).
Perhaps try `mosquitto_sub` and `mosquitto_pub` first to confirm that both the publish from rtl_433 and receive in python work.

lesreaper, 2019-06-30:
Well, I at last got it to work, and it doesn't make any sense to me. However, here it is:

Install the RTL antenna via the instructions from here: https://github.com/osmocom/rtl-sdr

Install RTL_433 from this site

Install MQTT for Linux via these instructions: http://www.steves-internet-guide.com/install-mosquitto-linux/

In one terminal, activate rtl_433 with this command: 
`rtl_433 -F "mqtt://localhost:1883,user=username,pass=password,retain=0" -a -A -R [integer of whatever device you need] -F json -M utc | mosquitto_pub -t rtl_433 -l -u username -P password`

Put this in your python3 file and run it:
```
import paho.mqtt.client as mqtt

# The callback for when the client receives a CONNACK response from the server.
def on_connect(client, userdata, flags, rc):
    print("Connected with result code "+str(rc))

    # Subscribing in on_connect() means that if we lose the connection and
    # reconnect then subscriptions will be renewed.
    # client.subscribe("$SYS/#")
    client.subscribe("rtl_433")

# The callback for when a PUBLISH message is received from the server.
def on_message(client, userdata, msg):
    print(msg.topic + " " + str(msg.payload))

# Setup MQTT client
client = mqtt.Client()
client.username_pw_set('username', 'password')
client.on_connect = on_connect
client.on_message = on_message
client.connect("localhost", 1883, 60)
client.loop_start()
```

After all of this, it will send JSON to the Python client with the data readable as a JSON file. 

Thank you for all the help @zuckschwerdt ! Hopefully this will stay here for awhile in case anyone has any issues moving forward.

That was exciting, wasn't it?

zuckschwerdt, 2019-06-30:
Great to hear! Thanks for documenting this. Some notes:
- You don't need `-a` or `-A` if you are not debugging/looking for a new device.
- You are sending the data twice:
  `rtl_433 -F "mqtt://localhost:1883,user=username,pass=password,retain=0" -R [integer] -F json`is the new way,
  `rtl_433 -R [integer] -F json -M utc | mosquitto_pub -t rtl_433 -l -u username -P password` is the old way.