merbanan/rtl_433 #1088: rtl_433 report "No supported devices found."

zeddski, 2019-06-27:
RFXcom (RFXtrx433) on Raspberry Pi (2), running domoticz.
rtl_433 reports No supported devices found:

> rtl_433
> rtl_433 version 18.12-219-g657b4b0 branch master at 201906271241 inputs file rtl_tcp RTL-SDR SoapySDR
> Trying conf file at "rtl_433.conf"...
> Trying conf file at "/root/.config/rtl_433/rtl_433.conf"...
> Trying conf file at "/usr/local/etc/rtl_433/rtl_433.conf"...
> Reading conf from "/usr/local/etc/rtl_433/rtl_433.conf".
> Protocol [31] "TFA-Twin-Plus-30.3049, Conrad KW9010, Ea2 BL999" does not take arguments ", Conrad KW9010, Ea2 BL999"!
> Protocol [33] "WT450, WT260H, WT405H" does not take arguments ", WT260H, WT405H"!
> Protocol [40] "Acurite 592TXR Temp/Humidity, 5n1 Weather Station, 6045 Lightning" does not take arguments ", 5n1 Weather Station, 6045 Lightning"!
> Protocol [42] "HIDEKI TS04 Temperature, Humidity, Wind and Rain Sensor" does not take arguments ", Humidity, Wind and Rain Sensor"!
> Protocol [74] "Acurite 00275rm,00276rm Temp/Humidity with optional probe" does not take arguments ",00276rm Temp/Humidity with optional probe"!
> Protocol [75] "LaCrosse TX35DTH-IT, TFA Dostmann 30.3155 Temperature/Humidity sensor" does not take arguments ", TFA Dostmann 30.3155 Temperature/Humidity sensor"!
> Protocol [78] "Fine Offset Electronics, WH25, WH32B, WH24, WH65B, HP1000 Temperature/Humidity/Pressure Sensor" does not take arguments ", WH25, WH24, WH65B, HP1000 Temperature/Humidity/Pressure Sensor"!
> Protocol [79] "Fine Offset Electronics, WH0530 Temperature/Rain Sensor" does not take arguments ", WH0530 Temperature/Rain Sensor"!
> Registered 101 out of 128 device decoding protocols [ 1-4 8 11-12 15-17 19-21 23 25-26 29-36 38-60 62-63 67-71 73-100 102-103 108-116 119 121 124-128 ]
> No supported devices found.

I've got no idea how to troubleshoot and google is no help this time.
Any hints what could be wrong?

zuckschwerdt, 2019-06-27:
What does `lsusb` report? Is any other program blocking the the receiver?

zeddski, 2019-07-02:
Thanks for replying. 
I checked 'lsusb':
> Bus 001 Device 004: ID 0403:6001 Future Technology Devices International, Ltd FT232 USB-Serial (UART) IC
> Bus 001 Device 003: ID 0424:ec00 Standard Microsystems Corp. SMSC9512/9514 Fast Ethernet Adapter
> Bus 001 Device 002: ID 0424:9514 Standard Microsystems Corp. SMC9514 Hub
> Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

and just to be sure, checked the RFXcom device:
>lsusb -v
>Bus 001 Device 004: ID 0403:6001 Future Technology Devices International, Ltd FT232 USB-Serial 

> (UART) IC
> Device Descriptor:
>   bLength                18
>   bDescriptorType         1
>   bcdUSB               2.00
>   bDeviceClass            0 (Defined at Interface level)
>   bDeviceSubClass         0
>   bDeviceProtocol         0
>   bMaxPacketSize0         8
>   idVendor           0x0403 Future Technology Devices International, Ltd
>   idProduct          0x6001 FT232 USB-Serial (UART) IC
>   bcdDevice            6.00
>   iManufacturer           1 RFXCOM
>   iProduct                2 RFXtrx433
>   iSerial                 3 A1BNI6L
>   bNumConfigurations      1
>   Configuration Descriptor:
>     bLength                 9
>     bDescriptorType         2
>     wTotalLength           32
>     bNumInterfaces          1
>     bConfigurationValue     1
>     iConfiguration          0
>     bmAttributes         0xa0
>       (Bus Powered)
>       Remote Wakeup
>     MaxPower               90mA
>     Interface Descriptor:
>       bLength                 9
>       bDescriptorType         4
>       bInterfaceNumber        0
>       bAlternateSetting       0
>       bNumEndpoints           2
>       bInterfaceClass       255 Vendor Specific Class
>       bInterfaceSubClass    255 Vendor Specific Subclass
>       bInterfaceProtocol    255 Vendor Specific Protocol
>       iInterface              2 RFXtrx433
>       Endpoint Descriptor:
>         bLength                 7
>         bDescriptorType         5
>         bEndpointAddress     0x81  EP 1 IN
>         bmAttributes            2
>           Transfer Type            Bulk
>           Synch Type               None
>           Usage Type               Data
>         wMaxPacketSize     0x0040  1x 64 bytes
>         bInterval               0
>       Endpoint Descriptor:
>         bLength                 7
>         bDescriptorType         5
>         bEndpointAddress     0x02  EP 2 OUT
>         bmAttributes            2
>           Transfer Type            Bulk
>           Synch Type               None
>           Usage Type               Data
>         wMaxPacketSize     0x0040  1x 64 bytes
>         bInterval               0
> Device Status:     0x0000
>   (Bus Powered)

As far as I can tell, no module or process has blocked the receiver but how can I tell for sure?

zeddski, 2019-07-02:
... or maybe, does rtl_433 not support RFXcom devices (which would be my mistake I guess)?

zuckschwerdt, 2019-07-03:
We only support RTL-SDR (clones) and devices with a working SoapySDR driver. I'll have to look into RFXcom support. Will take some time.

zuckschwerdt, 2019-07-03:
Wouldn't a RFXcom receiver already output decoded data? Can you give more detail on what you already see from the RFXcom (and it's software) and what rtl_433 can add?

zeddski, 2019-07-03:
Thanks Christian, my bad.

zeddski, 2019-07-03:
> Wouldn't a RFXcom receiver already output decoded data? Can you give more detail on what you already see from the RFXcom (and it's software) and what rtl_433 can add?

Yes, RFXcom does provide a (large) number of supported protocols but there are always more. I was looking for a need way to reverse engineer these unknown protocols through rtl_433.
Maybe I'm looking in the wrong direction, I'll continue my search!

closed 2019-07-03 by zeddski