merbanan/rtl_433 #1089: Recomendation : add simple mqtt topic example

ashtez, 2019-06-27:
Hello,

First of all, great work ! 
But right now, the -F documentation is really hard to understand ! Especially on how to add an mqtt topic. So maybe could you change it to : 

```
Option -F:
[-F kv|json|csv|mqtt|syslog|null] Produce decoded output in given format.
	Without this option the default is KV output. Use "-F null" to remove the default.
	Append output to file with :<filename> (e.g. -F csv:log.csv), defaults to stdout.
	Specify MQTT server with e.g. -F mqtt://localhost:1883
	Add MQTT options with e.g. -F "mqtt://host:1883,opt=arg"
	MQTT options are: user=foo, pass=bar, retain[=0|1],
		 usechannel=replaceid|afterid|beforeid|no, <format>[=topic]
	Supported MQTT formats: (default is all)
	  events: posts JSON event data
	  states: posts JSON state data
	  devices: posts device and sensor info in nested topics
	E.g. -F "mqtt://localhost:1883,user=USERNAME,pass=PASSWORD,retain=0,devices=/rtl_433"
	or for a very simple push: -F "mqtt://localhost:1883,events=topic"
	Specify host/port for syslog with e.g. -F syslog:127.0.0.1:1514
```

What changed is the **or for a very simple push: -F "mqtt://localhost:1883,events=topic"** line.

Thank you, Ashtez

zuckschwerdt, 2019-06-28:
Thanks, yes more documentation is always nice. I don't want to have a wall of text with every option, perhaps we need wiki links for each option and a nice formatted page with e.g. discussion and tips?

The MQTT format topic is just about to change, details in #1079. I guess we need even more explanation for that.

gabrielklein, 2019-06-30:
Hi, don't hesitate to modify https://github.com/merbanan/rtl_433/wiki/Usefull-commands .
I've tried to describe a few useful commands. I have no specific relation to this project :)

zuckschwerdt, 2019-08-29:
Some documentation add with ed9bdc4. Specifically at https://triq.org/rtl_433/OPERATION.html#mqtt-output
More examples to follow. Post your examples here and I'll include them.

closed 2019-08-29 by zuckschwerdt