merbanan/rtl_433 #1091: With newmodel, battery ok/low is difficult to use

kbahey, 2019-06-27:
Previously, Interlogix devices sent a JSON field like this:

`"battery":"OK"`
Or
`"battery":"LOW"`

That made consuming that field over MQTT, in Home Assitant, straightforward:

```
  - platform: mqtt
    name: Door_Battery
    state_topic: 'home/alarm/abcdef'
    device_class: battery
    payload_on: 'LOW'
    payload_off: 'OK'
    value_template: '{{ value_json.battery }}'
```
In other words, one field tells us whether the battery is ok or low.

With the -M newmodel, the Interlogix device sends the following instead:

`"battery_ok":1`
And looking at the code, it seems low battery has a new field name:
`"battery_low":1`

So, now, we have two fields, and there is no obvious way to consume that in MQTT/Home Assistant. 

The only way around it is changing the Python/paho-mqtt program that I use, to check these values. But that is added complexity. Right now I just pass the JSON message as is to Home Assistant, using just the device ID as a topic.

First question: are my assumptions above correct, re: the newmodel format?

Second question: if it is, then this is a functionality issue, and need to be rethought/redesigned.

zuckschwerdt, 2019-06-27:
There shouldn't be any key `battery_low`. It's only `battery_ok` now. The value is 0-1, rational: 0% to 100%. Should make graphing the value easier. Also some sensors report intermediate values.

kbahey, 2019-06-27:
Great to hear.

closed 2019-06-27 by kbahey