merbanan/rtl_433 #1095: Missing LICENCE file?

Funcan, 2019-07-02:
Hi

I'm working on a portapack style interface for a raspberry pi + limeSDR-mini combination, and the rtl_433 project offers frankly amazing functionality. I can't see a license declared anywhere in the repo, though the google group suggests the devs are amenable to reuse. Can you clarify the license please?

If the licence is declared and I'm just missing it, feel very free to include mockery and/or insults in your answer, I don't mind.

Many thanks

zuckschwerdt, 2019-07-03:
It should be GPL 2+, but we still need to scan all past contributions carefully to nail that down.

alexmyczko, 2019-08-05:
I think I already did that: http://phd-sid.ethz.ch/debian/rtl-433/rtl-433-18.12/debian/copyright

zuckschwerdt, 2019-08-05:
Good starting point, thanks! We recently included some source-libs (like Mongoose) and I'd like to scan all decoders (and maybe recover the copyright on some or prompt authors to add one) also.
I guess I need to make a table of all files an then group them.

alexmyczko, 2019-08-14:
Meanwhile I've got a more complete one: http://phd-sid.ethz.ch/debian/rtl-433/rtl-433-18.05%2Bgit2018017/debian/copyright

merbanan, 2019-08-14:
Feel free to send a PR and change the email to banan@ludd.ltu.se and add Christian W. Zuckschwerdt to the authors.

alexmyczko, 2019-08-14:
Not sure there's already COPYING and AUTHORS... is a LICENSE file needed when there's the other two?

merbanan, 2019-08-14:
Well you are free to maintain the package related files in the rtl_433 repo.

alexmyczko, 2020-01-06:
Is there a plan to test/merge many of the open PR? and make a new release before 26th Feb 2020? https://wiki.ubuntu.com/FocalFossa/ReleaseSchedule

That'd allow a lot of Ubuntu LTS users to be able install it with one command: apt install rtl-433 (and have a recent version)

zuckschwerdt, 2020-01-06:
As far as I know there are no open PR's that are just held up. Very much dependent on the device owners and hard to predict.
Note that the current master many still need tweaks to the new demod to work for more devices.
And the next release may very well default to `-M newmodel`. Lots of somewhat noticeable changes.

merbanan, 2020-01-06:
I plan to slightly rework the minmax demod. Some small changes to the output are also planned. But nothing that would hold up a 20.02 release (released in early February). I maybe want to add some more wmbus stuff also. But I think the current code is in an ok release state.

alexmyczko, 2020-10-30:
do you think it's time for a new release? debian freeze/new stable release is nearing:
https://release.debian.org/bullseye/freeze_policy.html

zuckschwerdt, 2020-10-30:
Yes, I planned a release for some weeks now, but there is always more to merge ;) I'll try to identify if something really needs merging and otherwise compile a changelog and release.

zuckschwerdt, 2020-11-13:
@alexmyczko a `20.11` release is now tagged. As always we are still very active with many ideas, but the version is well tested and stable with no loose ends.

merbanan, 2020-11-13:
\o/

alexmyczko, 2020-11-17:
For the impatient https://mentors.debian.net/package/rtl-433/

gdt, 2023-09-25:
This issue says LICENCE file is missing, but COPYING is present, so I think the issue is invalid.  If there is a specific defect about license labeling please open an issue and describe it clearly enough that it is fixable.

gdt, 2023-09-25:
As a side comment, there are many packaging systems and debian is just one.  So having debian files in a debian dir is fine, but content used by many should be assessed on a broad basis vs one system's norms.  That said, I suspect there are few if any actual conflicts.

closed 2023-09-25 by gdt