merbanan/rtl_433 #1099: openSUSE leap 15.1 no metadata found

hemna, 2019-07-06:
Trying to follow the instructions for installing the nightly build from obs

Added the repo:
```
└─[$] <git:(master)> sudo zypper addrepo -f obs://home:mnhauke:rtl_433:nightly/rtl_433 rtl_433-nightly
Adding repository 'rtl_433-nightly' .........................................................................................[done]
Repository 'rtl_433-nightly' successfully added

URI         : https://download.opensuse.org/repositories/home:/mnhauke:/rtl_433:/nightly/rtl_433
Enabled     : Yes                                                                               
GPG Check   : Yes                                                                               
Autorefresh : Yes                                                                               
Priority    : 99 (default priority)                                                             

Repository priorities are without effect. All enabled repositories share the same priority.
```
Then tried to zypper refresh

```
Retrieving repository 'rtl_433-nightly' metadata ...........................................................................[error]
Repository 'rtl_433-nightly' is invalid.
[rtl_433-nightly|https://download.opensuse.org/repositories/home:/mnhauke:/rtl_433:/nightly/rtl_433] Valid metadata not found at specified URL
Please check if the URIs defined for this repository are pointing to a valid repository.
Skipping repository 'rtl_433-nightly' because of the above error.
Some of the repositories have not been refreshed because of an error.
```

zuckschwerdt, 2019-07-06:
Looks like the paths might have changed. Can somebody look into that and post updated instructions?

mnhauke, 2019-07-06:
I maintain the daily OBS builds for rtl_433 ...
https://build.opensuse.org/package/show/home:mnhauke:rtl_433:nightly/rtl_433

Thanks for the report.

There was a caching issue on the machine from where I automatically trigger the OBS builds and one of the fallouts was that recently added repository targets did not have properly published binaries. As far as I have seen only Leap 15.1 was affected.

The issue should be fixed by now.

merbanan, 2019-07-07:
Ok issue closed then.

closed 2019-07-07 by merbanan